fwupd/fwupd #1145: 8BitDo M30 support

cbxbiker61, 2019-04-08:
Initial support for the 8BitDo M30.  I was able to successfully upgrade firmware from 1.00 to 1.10 by making a CAB file containing M30_Firmware_V1.10.dat.  Note that the "official" 8BitDo M30 firmware packages has mis-spelled M30_Fimware_V1.10.dat.

***************
Patch
***************

--- ./plugins/ebitdo/ebitdo.quirk.orig	2019-03-26 04:51:49.000000000 -0600
+++ ./plugins/ebitdo/ebitdo.quirk	2019-04-07 23:22:00.302090539 -0600
@@ -79,3 +79,13 @@ Flags = none
 Plugin = ebitdo
 Flags = none
 InstallDuration = 120
+
+# M30
+[DeviceInstanceId=USB\VID_1235&PID_5006]
+Plugin = ebitdo
+Flags = none
+[DeviceInstanceId=USB\VID_2DC8&PID_5006]
+Plugin = ebitdo
+Flags = none
+InstallDuration = 120

********************
m30.metainfo.xml
********************
<?xml version="1.0" encoding="UTF-8"?>

<component type="firmware">
  <id>com.8bitdo.m30.firmware</id>
  <name>M30</name>
  <summary>Firmware for the 8Bitdo M30 Game Controller</summary>

  <description>
    <p>
      Updating the firmware on your M30 device improves performance and adds
      new features.
    </p>
  </description>
  <provides>
    
    <firmware type="flashed">b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e</firmware>
  </provides>
  <url type="homepage">http://www.8bitdo.com/m30/</url>
  <metadata_license>CC0-1.0</metadata_license>
  <project_license>Proprietary</project_license>
  <developer_name>8Bitdo</developer_name>
  <releases>
    <release urgency="medium" version="1.10" date="2019-04-07">
      <checksum filename="M30_Firmware_V1.10.dat" target="content"/>
      <description>
        <p>
          Enhanced the stability of the Bluetooth pairing.
        </p>
      </description>
    </release>
  </releases>
  <screenshots>
    <screenshot type="default">
      <image type="source">https://raw.githubusercontent.com/hughsie/8bitdo-firmware/master/screenshots/M30.png</image>
      <caption>Unplug the controller, hold down L+R+START for 3 seconds until both LEDs are flashing then reconnect the controller.</caption>
    </screenshot>
  </screenshots>

  
  <requires>
    <id compare="ge" version="0.9.3">org.freedesktop.fwupd</id>
  </requires>
  <custom>
    <value key="LVFS::UpdateProtocol">com.8bitdo</value>
  </custom>

</component>

hughsie, 2019-04-08:
Can you do a pull request for fwupd and also https://github.com/hughsie/8bitdo-firmware please? Thanks.

cbxbiker61, 2019-04-08:
Yep, I'll do it.

hughsie, 2019-04-11:
Much appreciated, thanks.

closed 2019-04-11 by hughsie