fwupd/fwupd #1152: DFU plugin fails to probe Dell D6000 dock

superm1, 2019-04-15:
When fwupd runs the following errors are displayed:

```
09:13:40:0141 FuPlugin             performing usb_device_added() on dfu
09:13:40:0142 FuPluginDfu          failed to parse interface data for 17e9:6006: interface found, but not the correct length for functional data: 33 bytes: 09 21 01 c8 00 00 40 01 01 10 40 0a 00 36 09 02 21 52 69 64 67 65 44 6f 63 08 0b 02 03 01 00 20 06
09:13:40:0142 FuEngine             failed to add USB device 17e9:6006: failed to add device using on dfu: 17e9:6006 is not supported: no DFU interfaces
```

USB output:
```

Bus 002 Device 009: ID 17e9:6006 DisplayLink 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x17e9 DisplayLink
  idProduct          0x6006 
  bcdDevice           30.00
  iManufacturer           1 DisplayLink
  iProduct                2 Dell Universal Dock D6000
  iSerial                 3 1710250587
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          618
    bNumInterfaces          7
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           6
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      3 
      iInterface              0 
      ** UNRECOGNIZED:  0c 5f 01 00 0a 00 04 04 01 00 04 00
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
[... 494 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               3
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           70
  bNumDeviceCaps          4
Device Status:     0x000d
  Self Powered
  U1 Enabled
  U2 Enabled
```

superm1, 2019-04-15:
At least looking at the code to me it seems like there needs to be either:
1. a case that it's `> sizeof(DfuFuncDescriptor)` and then only copy `sizeof(DfuFuncDescriptor)`
2. Adjust
```
if (sz == sizeof(DfuFuncDescriptor)) {
		memcpy (&desc, buf, sz);
```

To be:
```
if (sz >= sizeof(DfuFuncDescriptor)) {
		memcpy (&desc, buf, sizeof(DfuFuncDescriptor));
```

hughsie, 2019-04-15:
Do you think the Dell device actually has a valid USB DFU descriptor? Did you expect DFU to work?

superm1, 2019-04-16:
I do expect DFU to be used for flashing some components, the Windows updater does use DFU.  You can rip it apart with 7z and see a dfu/ folder with various ROM images.

https://downloads.dell.com/FOLDER05084853M/1/Dock_D6000_FW_Updater_A02_73F77.exe

I don't have a lot of insight beyond this right now though.  Maybe for now a quirk that allows non-standard DFU descriptor size to not let fwupd freak out with this VID/PID?  If we discover later that's wrong can adjust it.

superm1, 2019-04-16:
btw, this device is developed differently than other docks, it's actually a displaylink based product, so getting more answers and development will require some hoops.

superm1, 2019-04-16:
I double checked DFU 1.1 spec and to me it looks like all the fields are properly filled out.  That other data at the end, should just be vendor specific data.

superm1, 2019-04-17:
With that PR see this now, thanks.

```
Dell Universal Dock D6000
  DeviceId:             8ad56c3d40864e26aecb5621725c2ff0ad4ec52a
  Guid:                 46bbee30-ca19-5fb9-b81f-5a8e3a8c3c8a <- USB\VID_17E9&PID_6006&REV_3000
  Guid:                 1b3b6e05-bd4d-5880-b68d-c7c00b6681f8 <- USB\VID_17E9&PID_6006
  Guid:                 a9a9b566-6fd9-5165-9969-f56f5d3390f7 <- USB\VID_17E9
  Guid:                 9b2e35bd-5287-5084-afe6-ed4e600cea8b <- USB\CLASS_FF&SUBCLASS_00&PROT_03
  Guid:                 faae61e5-c90f-5283-9278-4bc68419996b <- USB\CLASS_FF&SUBCLASS_00
  Guid:                 4bd986bc-4aaf-5baf-8bc2-9f7928182387 <- USB\CLASS_FF
  Guid:                 073430ea-f120-5ba5-ab70-cc4fe2fe1fd9 <- USB\CLASS_FE&SUBCLASS_01&PROT_01
  Guid:                 ab182806-032b-5b7b-994f-b0713031d2b6 <- USB\CLASS_FE&SUBCLASS_01
  Guid:                 c418fefd-0db6-5152-a4da-7a7527183cd1 <- USB\CLASS_FE
  Guid:                 60ce568a-d691-5046-b704-a3a0d938900d <- USB\CLASS_01&SUBCLASS_01&PROT_20
  Guid:                 b2a98bf6-dca5-5c9f-b5e8-f0aae82f7c90 <- USB\CLASS_01&SUBCLASS_01
  Guid:                 cc878932-9d8b-53f6-8760-2708ef53c61f <- USB\CLASS_01
  Guid:                 115a4195-c614-59c8-999e-57e0b89505da <- USB\CLASS_01&SUBCLASS_02&PROT_20
  Guid:                 1c123739-37d1-5e32-9b6a-bdbed964851f <- USB\CLASS_01&SUBCLASS_02
  Guid:                 9d80e9c0-2635-5155-aefe-5caa5bca6b55 <- USB\CLASS_02&SUBCLASS_0D&PROT_00
  Guid:                 89b53976-2657-55b2-aaa8-70d2943ad9f1 <- USB\CLASS_02&SUBCLASS_0D
  Guid:                 9da74da6-bfa4-53cc-8cc1-7559a7419a15 <- USB\CLASS_02
  Guid:                 0e01ba64-2147-5caf-8cab-74f9c72fa96d <- USB\CLASS_0A&SUBCLASS_00&PROT_01
  Guid:                 07386dae-0684-56f7-aa73-1c54347ef354 <- USB\CLASS_0A&SUBCLASS_00
  Guid:                 b520fa99-b299-5e6c-8054-dae4d65a5cf4 <- USB\CLASS_0A
  Serial:               1710250587
  Plugin:               dfu
  Flags:                updatable|registered
  Vendor:               DisplayLink
  VendorId:             USB:0x17E9
  Version:              30.0
  Icon:                 drive-harddisk-usb
  Created:              2019-04-17
```

closed 2019-04-17 by superm1