fwupd/fwupd #1156: Firmware version formats were different, device was 'quad' and release is 'plain'

lubo, 2019-04-16:
When trying to update XPS 13 9360 System Firmware from version 0.2.10.0 to 0.2.11.0 using
`$ fwupdmgr update`, I get:

```
No upgrades for XPS13 9360 Thunderbolt Controller, current is 21.00: 21.00=same
Downloading 0.2.11.0 for XPS 13 9360 System Firmware...
Decompressing…           [***************************************]
Firmware version formats were different, device was 'quad' and release is 'plain'
```

`$ fwupdmgr --version` output:

```
client version:	1.2.7
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.7
```

`$ fwupdmgr get-devices` output:

```
XPS13 9360 Thunderbolt Controller
  DeviceId:             f316028454ee963aeddc15ba7a94b81f1303ef82
  Guid:                 dfd51125-338f-56ff-b721-fa3bea8e534e
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              21.00
  Icon:                 computer
  Created:              2019-04-16

XPS 13 9360 System Firmware
  DeviceId:             3f0ecc7823ee99c718693e8ecca38f54e8738dc5
  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(6e5b9ba5c7f0b7db73c550c578944a5ca7530e6a)
  Version:              0.2.10.0
  VersionLowest:        0.2.10.0
  Icon:                 computer
  Created:              2019-04-16

XPS 13 9360 TPM 1.2
  DeviceId:             6f9095025a636a5185bfce08ed8e9fc845bc4b8a
  Guid:                 f9bdd338-b410-5e73-902d-7b6e4694bb56
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|require-ac|supported|registered
  Vendor:               Dell Inc.
  Version:              5.81.2.1
  Icon:                 computer
  Created:              2019-04-16

PC300 NVMe SK hynix 512GB
  DeviceId:             e11623b2caa18fee292058a5c09ca4e6152f7ecf
  Guid:                 66cc0d83-cfdc-5602-9581-3637989f0c64
  Guid:                 8c1c105b-8aad-5ad2-9d67-68d143164128
  Guid:                 c93201aa-cbec-55d4-bafb-f7c488646839
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               SK hynix
  VendorId:             NVME:0x1C5C
  Version:              20004A00
  Icon:                 drive-harddisk
  Created:              2019-04-16

```

`$ efibootmgr -v` output:

```
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001
Boot0000* Windows Boot Manager	HD(1,GPT,fe7b857d-5afb-4c90-bae8-ce57a56971e0,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* grub	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,AC-E4-2E-61-90-02-79-99)/HD(1,GPT,220c0573-da92-4a3e-863b-9e239dc72bd2,0x100,0x6400)/File(\EFI\grub\grubx64.efi)
Boot0002* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,220c0573-da92-4a3e-863b-9e239dc72bd2,0x100,0x6400)/File(\EFI\arch\fwupx64.efi)
```

`$ efivar -l | grep fw` output is empty.

`$ tree /boot/efo` output:

```
/boot/efi
└── EFI
    ├── arch
    │   ├── fw
    │   ├── fwupdx64.efi
    │   └── fwupx64.efi
    ├── Dell
    │   ├── Bios
    │   │   └── Recovery
    │   │       ├── BIOS_CUR.RCV
    │   │       └── BIOS_PRE.rcv
    │   └── logs
    │       ├── diags_current.xml
    │       └── diags_previous.xml
    └── grub
        └── grubx64.efi

8 directories, 7 files
```

I'm using Arch Linux with `fwupd` installed using pacman. I've tried rebooting and secure boot is disabled.

superm1, 2019-04-17:
This is an unfortunate side effect of https://github.com/hughsie/fwupd/commit/ce3cea8cb49bee10078d72b835792a18b85f6be5.  For Dell UEFI firmware on LVFS the version is just a number (like it gets from ESRT table).  The CAB file doesn't have a format encoded since it wasn't needed.

For now as a workaround you can apply the flag `--force` to make it install the firmware.

@hughsie I can push for adding this flag in some future platform CAB files, but I don't believe we can change the build tools for existing ones.  Would you consider:
* A mass correction to LVFS release metadata for the Dell UEFI capsules to set this VersionFormat on LVFS - to quad.
* As a vendor specific LVFS account setting automatically associate VersionFormat with what the vendor is nominally using when setting protocol?  Eg for Dell firmware when protocol is set to UEFI capsule set it to Quad.

lubo, 2019-04-17:
@superm1 Using `--force` helped and the update was successful. Thanks.

d-vogel, 2019-04-17:
Same here with Latitude E7470. Thanks for the tip @superm1 !

hughsie, 2019-04-17:
> A mass correction to LVFS release metadata for the Dell UEFI capsules to set this VersionFormat on LVFS - to quad.

I don't think that works -- we'd have the correct value in the `firmware.xml.gz` but not in the `.metainfo.xml` -- the latter being used when doing `install`.

> As a vendor specific LVFS account setting automatically associate VersionFormat with what the vendor is nominally using when setting protocol

I'll investigate this a bit in an hour or so, and if that fails I'll just revert my commit.

closed 2019-04-17 by hughsie

AdamWill, 2019-04-30:
I'm still seeing this error with fwupd 1.2.8.

AdamWill, 2019-04-30:
[root@xps13k adamw]# fwupdmgr update
    No upgrades for XPS13 9360 Thunderbolt Controller, current is 26.01: 26.01=same, 21.00=older
    Downloading 0.2.11.0 for XPS 13 9360 System Firmware...
    Fetching firmware https://fwupd.org/downloads/91bdcaa2a93372e83bd39181320af20c955f21ac-XPS_9360_2.11.0.cab
    Downloading…             [*                                      ] Less than oneDownloading…             [*                                      ] Less than oneDownloading…             [**                                     ] Less than oneDownloading…             [**                                     ] Less than oneDownloading…             [***                                    ] Less than oneDownloading…             [***                                    ] Less than oneDownloading…             [****                                   ] Less than oneDownloading…             [****                                   ] Less than oneDownloading…             [*****                                  ] Less than oneDownloading…             [*****                                  ] Less than oneDownloading…             [*****                                  ] Less than oneDownloading…             [******                                 ] Less than oneDownloading…             [*******                                ] Less than oneDownloading…             [*******                                ] Less than oneDownloading…             [*******                                ] Less than oneDownloading…             [***************************************]
    Decompressing…           [***************************************]
    Firmware version formats were different, device was 'quad' and release is 'plain'
    [root@xps13k adamw]# rpm -q fwupd
    fwupd-1.2.8-1.fc30.x86_64
    [root@xps13k adamw]#

hughsie, 2019-04-30:
@AdamWill we're working on the proper fix in https://github.com/hughsie/fwupd/pull/1175 -- in the meantime using `--force ` will allow you to continue.

superm1, 2019-04-30:
Has the daemon been restarted since upgrading to 1.2.8?

Can you share fwupdmgr --version output?

[EXTERNAL EMAIL]

AdamWill, 2019-04-30:
@hughsie With `--force`, the operation proceeded but the installation of the firmware failed on reboot.

superm1, 2019-04-30:
Can you elaborate on what you mean with "failed"?  Was it something related to battery being too low?
Can you upload a failure report to LVFS?

AdamWill, 2019-04-30:
"Can you elaborate on what you mean with "failed"?"

The firmware displayed a message saying the update had failed. It was only shown briefly and then the system rebooted.

superm1, 2019-04-30:
Was it with a Dell logo?

Would you be able to capture a picture of the error?

[EXTERNAL EMAIL]

AdamWill, 2019-04-30:
I don't remember. It was only shown briefly. I can try it again in a bit, I'm working on three other things at the moment.

superm1, 2019-05-01:
If it had a Dell logo:
* Was your battery adequately charged?
* Was you AC adapter plugged in?
* Can you check `fwupdmgr get-results` for more information on the error?

If it didn't have a Dell logo:
* Can you please check the logging visible through `/usr/libexec/fwupdate` for more information?

If you don't know if it had a Dell logo, please re-try it.

AdamWill, 2019-05-01:
OK, a retry with `--force` worked fine.

superm1, 2019-05-01:
OK good, it was probably just low battery or lack of AC then.

superm1, 2019-05-01:
I'm still a little perplexed why you had this behavior on 1.2.8.

`Firmware version formats were different, device was 'quad' and release is 'plain'`

https://github.com/hughsie/fwupd/commit/21ab7f588c71f80adc0e8e62c456960521998d00 should have worked on 1.2.8 on your machine.

Can you please share the output of:
```
/usr/libexec/fwupd/fwupdtool hwids
```

AdamWill, 2019-05-01:
Computer Information
    --------------------
    BiosVendor: Dell Inc.
    BiosVersion: 2.11.0
    BiosMajorRelease: 2
    BiosMinorRelease: 11
    Manufacturer: Dell Inc.
    Family: XPS
    ProductName: XPS 13 9360
    ProductSku: 75B
    EnclosureKind: 9
    BaseboardManufacturer: Dell Inc.
    BaseboardProduct: 839Y6

    Hardware IDs
    ------------
    {925d0c50-8b85-55f3-ada4-569068be39fd}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
    {37cf97bb-a085-54b9-b2ad-5f52754a40a6}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
    {394fd645-6208-592c-adc7-1520fda1bb66}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
    {fae55358-25d5-54fd-81d7-c6ccfde33a73}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
    {94e8fe4a-d368-520f-bb81-7e73f1513933}   <- Manufacturer + Family + ProductName + ProductSku
    {5bd169c3-df04-560d-b1e0-4e32e62ed68a}   <- Manufacturer + Family + ProductName
    {ff2ea282-d362-5c4e-b84f-d5a00404e2cf}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
    {31fe1e66-a857-51d9-8531-4d85ea705b66}   <- Manufacturer + ProductSku
    {0a3382e9-4157-5bc3-86e1-0884d918b515}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
    {01e59c76-43cc-5384-82c3-8235fe966152}   <- Manufacturer + ProductName
    {6d57fad8-e4fe-52ad-95e9-43edabfd283d}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
    {61178075-a8fd-563c-9045-44227d8c121f}   <- Manufacturer + Family
    {29ca4080-4d68-5395-afe2-35fa9a4f13bf}   <- Manufacturer + EnclosureKind
    {3cd0217f-3040-5945-a31b-2e40b8648db4}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
    {85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Manufacturer

abusch, 2019-05-02:
I've also had the same issue with version 1.2.8 (`--force` worked). Here's the output of `fwupdtool hwids` in case it helps:
```
Computer Information
--------------------
BiosVendor: Dell Inc.
BiosVersion: 1.12.1
BiosMajorRelease: 1
BiosMinorRelease: 12
Manufacturer: Dell Inc.
Family: Precision
ProductName: Precision 5520
ProductSku: 7BF
EnclosureKind: a
BaseboardManufacturer: Dell Inc.
BaseboardProduct: R6JFH

Hardware IDs
------------
{6ef9d8ff-95ef-52e0-ae85-450a54fe2527}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{e0216ab2-fe2a-5ef5-891e-3b5b15b6b425}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{723cb1fc-092d-5e0f-95ed-e283ce871480}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{1ffedf60-dde4-5896-98fa-0fccfb283bca}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{33589868-421e-5302-baf8-d12b8c5ebdb2}   <- Manufacturer + Family + ProductName + ProductSku
{ff197f1b-a45e-511b-8689-4e549d5dfa57}   <- Manufacturer + Family + ProductName
{6dcea9d2-39ae-58ba-ae3e-9092fc52dbb1}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{1aa04a6f-352f-5ca8-b751-8b8f9fafa92e}   <- Manufacturer + ProductSku
{b61b598d-2fa4-52e5-9b07-eb34b554d799}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{0c57c850-c007-5780-853f-b1b81a6bea83}   <- Manufacturer + ProductName
{1b9eb02e-428b-537d-b5f5-1ad44c02c92c}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{9bf5227f-7ae3-5421-9a57-a440b2cc99ed}   <- Manufacturer + Family
{5c163113-9296-5c8d-a92d-ae04eb59a0f7}   <- Manufacturer + EnclosureKind
{2b8e8f95-0b6b-5171-8845-7aaecb9200f1}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Manufacturer
```

superm1, 2019-05-02:
Would you mind checking on that referenced issue above with master? #1181 want to make sure it's done properly.

Thanks!

[EXTERNAL EMAIL]

ford-prefect, 2019-05-11:
I still see this 1.2.8 on my Precision 5530:

```
$ fwupdmgr update
Downloading 0.1.10.1 for Precision 5530 System Firmware...
Decompressing…           [***************************************]
Firmware version formats were different, device was 'quad' and release is 'plain'
```

superm1, 2019-05-11:
It should be fixed in master.
Could you please try master and confirm it is fixed there for you?

ford-prefect, 2019-05-13:
It might be tough for me to compile and try this out at the moment, will let you know if/when I am able.

superm1, 2019-05-13:
If it helps there are snaps that track master in the edge channel you can install.

BookGin, 2019-05-18:
I can also confirm that updating to 0.2.11.0 with `--force` option works fine for me on my XPS 13 9360 System Firmware.

```sh
$ sudo pacman -S fwupd
$ fwupdmgr --version  
client version:	1.2.8
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.8
```