fwupd/fwupd #1161: "fwupdmgr update" installs 8BitDo sf30_pro_sn_30_pro on m30

cbxbiker61, 2019-04-18:
fwupdmgr --version                                                                                              
client version: 1.2.7
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.7

Linux 5.0.7

If the m30 is put into firmware update mode prior to running "fwupdmgr update" it installs the sf30_pro_sn_30_pro firmware rather than correct firmware.  This did no real damage as I was able to reflash to the correct version with a .cab file.

Not sure how fwupdmgr is going to be able to tell these devices apart, since when they're in firmware update mode they look alike on the USB bus as far as I can tell.

cbxbiker61, 2019-04-18:
I think it must've been pulling in "failed" attempts.  After I removed /var/lib/fwupd and recreated it the device does not show up as updateable.

I just looked in /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz.  The M30 is not in there yet.

hughsie, 2019-04-18:
I've just moved M30 to testing, it'll appear in the metadata in a few minutes.

Felixoid, 2019-06-20:
Unfortunately, it's still an issue. Today it did try to upgrade me to sf30_pro_sn_30_pro as well

```
Devices that were not updated correctly:

 • 8Bitdo   (1.29 → 1.25)
 • 8Bitdo   (1.25 → 1.10)
 • 8Bitdo   (1.00 → 1.29)
```

Thanks, Chaos, it's easy to downgrade. I've sent the report in fwupd

closed 2019-09-30 by stale[bot]