fwupd/fwupd #1162: Device conflict between iFi nano iDSD Black Label and 8BitDo M30 controller

pcm720, 2019-04-22:
For some reason, fwupd can't add 8BitDo M30 controller to supported devices while iFi nano iDSD Black Label USB DAC is connected.
If I unplug nano iDSD and reconnect M30, fwupd successfully detects the gamepad.

Here's `# /usr/lib/fwupd/fwupd --verbose` output:
```
14:31:49:0714 FuPlugin             performing usb_device_added() on dfu
14:31:49:0714 FuPluginDfu          DFU v1.1
14:31:49:0714 FuPluginDfu          using DFU transfer size 0x0040 bytes
14:31:49:0714 FuDevice             using 51a5f0521bddd3e9d93388f831b2b5ffdf225d9a for usb:00:09
14:31:49:0715 FuPluginDfu          no UM0424 sector description in iFi (by AMR) DFU
14:31:49:0715 FuPlugin             emit added from dfu: 51a5f0521bddd3e9d93388f831b2b5ffdf225d9a
14:31:49:0715 FuPlugin             performing fu_plugin_device_registered() on uefi
14:31:49:0715 FuDeviceList         ::added 51a5f0521bddd3e9d93388f831b2b5ffdf225d9a
14:31:49:0716 XbSilo               1,69ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='21985dfd-1213-5d9d-a54a-71e2422e917a']/../..|components/component/provides/firmware[@type='flashed'][text()='4bd15d9f-4710-5711-a415-07e4a980747a']/../..|components/component/provides/firmware[@type='flashed'][text()='be1b4605-827a-587f-a71d-2f22e6f5ebe3']/../..|components/component/provides/firmware[@type='flashed'][text()='60ce568a-d691-5046-b704-a3a0d938900d']/../..|components/component/provides/firmware[@type='flashed'][text()='b2a98bf6-dca5-5c9f-b5e8-f0aae82f7c90']/../..|components/component/provides/firmware[@type='flashed'][text()='cc878932-9d8b-53f6-8760-2708ef53c61f']/../..|components/component/provides/firmware[@type='flashed'][text()='115a4195-c614-59c8-999e-57e0b89505da']/../..|components/component/provides/firmware[@type='flashed'][text()='1c123739-37d1-5e32-9b6a-bdbed964851f']/../..|components/component/provides/firmware[@type='flashed'][text()='073430ea-f120-5ba5-ab70-cc4fe2fe1fd9']/../..|components/component/provides/firmware[@type='flashed'][text()='ab182806-032b-5b7b-994f-b0713031d2b6']/../..|components/component/provides/firmware[@type='flashed'][text()='c418fefd-0db6-5152-a4da-7a7527183cd1']/../..|components/component/provides/firmware[@type='flashed'][text()='afa5518d-7c55-51cb-8e1c-bf3b11b67dfa']/../..|components/component/provides/firmware[@type='flashed'][text()='cee9c5bc-aa12-51eb-a220-d0ef588d7cb2']/../..|components/component/provides/firmware[@type='flashed'][text()='571229d5-caed-5cbc-91c1-5c216601a194']/../..` limit=1 -> 0 results
14:31:49:0716 FuHistory            get device
14:31:49:0717 FuEngine             no plugin specified for USB device 046d:c539
14:31:49:0717 FuEngine             no plugin specified for USB device 046d:c53d
14:31:49:0717 FuPlugin             performing usb_device_added() on ebitdo
14:31:49:0717 FuDevice             using 71413822e87deae707da948bd4b7e537af463c19 for usb:00:02
14:31:49:0726 FuPlugin             emit added from ebitdo: 71413822e87deae707da948bd4b7e537af463c19
14:31:49:0726 FuPlugin             performing fu_plugin_device_registered() on uefi
14:31:49:0726 FuDeviceList         ignoring device 71413822e87deae707da948bd4b7e537af463c19 [ebitdo] existing device 51a5f0521bddd3e9d93388f831b2b5ffdf225d9a [dfu] already exists
14:31:49:0726 XbSilo               0,30ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='043c7673-bd45-5e2e-b8b9-c22df2dfec2a']/../..|components/component/provides/firmware[@type='flashed'][text()='b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e']/../..|components/component/provides/firmware[@type='flashed'][text()='cc93af80-221d-50fe-acb2-de0b278379dd']/../..|components/component/provides/firmware[@type='flashed'][text()='49c74d94-86d7-5587-b48f-aa36decb1568']/../..|components/component/provides/firmware[@type='flashed'][text()='5bc60e62-f461-58bd-91b6-7de3392f8bd6']/../..|components/component/provides/firmware[@type='flashed'][text()='afa5518d-7c55-51cb-8e1c-bf3b11b67dfa']/../..|components/component/provides/firmware[@type='flashed'][text()='cee9c5bc-aa12-51eb-a220-d0ef588d7cb2']/../..|components/component/provides/firmware[@type='flashed'][text()='571229d5-caed-5cbc-91c1-5c216601a194']/../..` limit=1 -> 1 results
```
--------------
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.7
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.7
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
iFi (by AMR) HD USB Audio
  DeviceId:             51a5f0521bddd3e9d93388f831b2b5ffdf225d9a
  Guid:                 21985dfd-1213-5d9d-a54a-71e2422e917a
  Guid:                 4bd15d9f-4710-5711-a415-07e4a980747a
  Guid:                 be1b4605-827a-587f-a71d-2f22e6f5ebe3
  Guid:                 60ce568a-d691-5046-b704-a3a0d938900d
  Guid:                 b2a98bf6-dca5-5c9f-b5e8-f0aae82f7c90
  Guid:                 cc878932-9d8b-53f6-8760-2708ef53c61f
  Guid:                 115a4195-c614-59c8-999e-57e0b89505da
  Guid:                 1c123739-37d1-5e32-9b6a-bdbed964851f
  Guid:                 073430ea-f120-5ba5-ab70-cc4fe2fe1fd9
  Guid:                 ab182806-032b-5b7b-994f-b0713031d2b6
  Guid:                 c418fefd-0db6-5152-a4da-7a7527183cd1
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194
  Plugin:               dfu
  Flags:                updatable|registered
  Vendor:               iFi (by AMR)
  VendorId:             USB:0x20B1
  Version:              5.30
  Icon:                 drive-harddisk-usb
  Created:              2019-04-22

WDC WD30EZRZ-00GXCB0
  DeviceId:             602b0a6cc821d155208724f0e22f8d111542b74c
  Guid:                 aaebb49f-5ebd-5492-bdec-f3e386466879
  Guid:                 ef46686e-82df-582a-a948-e919a81541d4
  Guid:                 c217d516-e636-5757-8387-072817ae30a9
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              80.00A80
  Icon:                 drive-harddisk
  Created:              2019-04-22

Samsung SSD 970 EVO 500GB
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Guid:                 5fc1d3d7-f6cf-5f1c-ba7e-18a56f00f262
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              2B2QEXE7
  Icon:                 drive-harddisk
  Created:              2019-04-22
```
Please answer the following questions:

- Operating system and version: Arch Linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes

hughsie, 2019-04-23:
> fwupdmgr get-devices

Can you do this as root please. I think both devices are using the same DFU VID, but I don't know why the iFi device is returning the generic bootloader vid/pid in runtime mode.

pcm720, 2019-04-23:
Sure, here's the output of `# fwupdmgr get-devices`:
```
iFi (by AMR) HD USB Audio
  DeviceId:             51a5f0521bddd3e9d93388f831b2b5ffdf225d9a
  Guid:                 21985dfd-1213-5d9d-a54a-71e2422e917a <- USB\VID_20B1&PID_3008&REV_0530
  Guid:                 4bd15d9f-4710-5711-a415-07e4a980747a <- USB\VID_20B1&PID_3008
  Guid:                 be1b4605-827a-587f-a71d-2f22e6f5ebe3 <- USB\VID_20B1
  Guid:                 60ce568a-d691-5046-b704-a3a0d938900d <- USB\CLASS_01&SUBCLASS_01&PROT_20
  Guid:                 b2a98bf6-dca5-5c9f-b5e8-f0aae82f7c90 <- USB\CLASS_01&SUBCLASS_01
  Guid:                 cc878932-9d8b-53f6-8760-2708ef53c61f <- USB\CLASS_01
  Guid:                 115a4195-c614-59c8-999e-57e0b89505da <- USB\CLASS_01&SUBCLASS_02&PROT_20
  Guid:                 1c123739-37d1-5e32-9b6a-bdbed964851f <- USB\CLASS_01&SUBCLASS_02
  Guid:                 073430ea-f120-5ba5-ab70-cc4fe2fe1fd9 <- USB\CLASS_FE&SUBCLASS_01&PROT_01
  Guid:                 ab182806-032b-5b7b-994f-b0713031d2b6 <- USB\CLASS_FE&SUBCLASS_01
  Guid:                 c418fefd-0db6-5152-a4da-7a7527183cd1 <- USB\CLASS_FE
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa <- USB\CLASS_03&SUBCLASS_00&PROT_00
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2 <- USB\CLASS_03&SUBCLASS_00
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194 <- USB\CLASS_03
  Serial:               00000000
  Plugin:               dfu
  Flags:                updatable|registered
  Vendor:               iFi (by AMR)
  VendorId:             USB:0x20B1
  Version:              5.30
  Icon:                 drive-harddisk-usb
  Created:              2019-04-23

WDC WD30EZRZ-00GXCB0
  DeviceId:             602b0a6cc821d155208724f0e22f8d111542b74c
  Guid:                 aaebb49f-5ebd-5492-bdec-f3e386466879 <- IDE\WDC_WD30EZRZ-00GXCB0____________________80.00A80
  Guid:                 ef46686e-82df-582a-a948-e919a81541d4 <- IDE\0WDC_WD30EZRZ-00GXCB0____________________
  Guid:                 c217d516-e636-5757-8387-072817ae30a9 <- WDC WD30EZRZ-00GXCB0
  Serial:               WD-WCC7K5ADK37N
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              80.00A80
  Icon:                 drive-harddisk
  Created:              2019-04-23

Samsung SSD 970 EVO 500GB
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 47335265-a509-51f7-841e-1c94911af66b <- NVME\VEN_144D&DEV_A808
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf <- NVME\VEN_144D
  Guid:                 5fc1d3d7-f6cf-5f1c-ba7e-18a56f00f262 <- Samsung SSD 970 EVO 500GB
  Serial:               S466NX0K726339N
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              2B2QEXE7
  Icon:                 drive-harddisk
  Created:              2019-04-23
```

superm1, 2019-04-23:
I think this might be due to the USB class GUIDs and the device de-duplication logic.
Both devices have these GUIDs:
```
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa <- USB\CLASS_03&SUBCLASS_00&PROT_00
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2 <- USB\CLASS_03&SUBCLASS_00
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194 <- USB\CLASS_03
```

So when `fu_device_list_add` runs to find matching devices they get passed in to `fu_device_list_get_by_guids` which will check those.

My opinion here is that we should stop including GUIDs that are so "generic", just class, subclass, and prot in fwupd.  In practice you will never have a piece of firmware that can apply to all "Class 3" USB devices, that's just plain dangerous.  The VID and VID/PID GUIDs should really be the only ones that USB devices should use.

superm1, 2019-04-23:
@pcm720 please try that PR ^ so we can confirm that was the root cause and then discuss whether this is the right thing to do or not.

hughsie, 2019-04-23:
Don't we use the super generic class/subclass for a quirk fixup? I forget the details.

superm1, 2019-04-23:
> Don't we use the super generic class/subclass for a quirk fixup? I forget the details.

I didn't see `CLASS` used in any .quirk files and looking through `fu-usb-device` code nothing stood out to me at least.  It looks like it was introduced in 74f976a821 to match the the MS behavior.

pcm720, 2019-04-23:
> @pcm720 please try that PR ^ so we can confirm that was the root cause and then discuss whether this is the right thing to do or not.

Okay, with 2e23b1c67a7c1d6214fcaa19dc0ad074e43ec6b8 and both devices plugged in I get this:
```
8BitDo M30 gamepad
  DeviceId:             672c087de09848d9e7ee32aa1dea2fbeb8b81e6b
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Guid:                 cc93af80-221d-50fe-acb2-de0b278379dd <- USB\VID_2DC8&PID_5006&REV_0001
  Guid:                 49c74d94-86d7-5587-b48f-aa36decb1568 <- USB\VID_2DC8&PID_5006
  Guid:                 5bc60e62-f461-58bd-91b6-7de3392f8bd6 <- USB\VID_2DC8
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|supported|needs-bootloader|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x2DC8
  Version:              1.10
  Icon:                 input-gaming
  InstallDuration:      120
  Created:              2019-04-23

WDC WD30EZRZ-00GXCB0
  DeviceId:             602b0a6cc821d155208724f0e22f8d111542b74c
  Guid:                 aaebb49f-5ebd-5492-bdec-f3e386466879 <- IDE\WDC_WD30EZRZ-00GXCB0____________________80.00A80
  Guid:                 ef46686e-82df-582a-a948-e919a81541d4 <- IDE\0WDC_WD30EZRZ-00GXCB0____________________
  Guid:                 c217d516-e636-5757-8387-072817ae30a9 <- WDC WD30EZRZ-00GXCB0
  Serial:               WD-WCC7K5ADK37N
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              80.00A80
  Icon:                 drive-harddisk
  Created:              2019-04-23

Samsung SSD 970 EVO 500GB
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 47335265-a509-51f7-841e-1c94911af66b <- NVME\VEN_144D&DEV_A808
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf <- NVME\VEN_144D
  Guid:                 5fc1d3d7-f6cf-5f1c-ba7e-18a56f00f262 <- Samsung SSD 970 EVO 500GB
  Serial:               S466NX0K726339N
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              2B2QEXE7
  Icon:                 drive-harddisk
  Created:              2019-04-23
```
M30 is detected by fwupd, but iFi doesn't show up.
There are no errors in the output of `# /usr/lib/fwupd/fwupd --verbose`, but I'm not sure if this is correct behaviour.

superm1, 2019-04-23:
oh... dfu.quirk does use `CLASS_FE&SUBCLASS_01`.  So this isn't the right solution then.

superm1, 2019-04-23:
And it will also affect fastboot.quirk.  I guess I missed those in my first grep.

superm1, 2019-04-23:
OK try this one instead ^

pcm720, 2019-04-23:
> OK try this one instead ^

Okay, with ae315d891811523004e81e83ee8bc3e7e2073178, both devices show up in fwupdmgr:
```
iFi (by AMR) HD USB Audio
  DeviceId:             51a5f0521bddd3e9d93388f831b2b5ffdf225d9a
  Guid:                 21985dfd-1213-5d9d-a54a-71e2422e917a <- USB\VID_20B1&PID_3008&REV_0530
  Guid:                 4bd15d9f-4710-5711-a415-07e4a980747a <- USB\VID_20B1&PID_3008
  Guid:                 be1b4605-827a-587f-a71d-2f22e6f5ebe3 <- USB\VID_20B1
  Guid:                 073430ea-f120-5ba5-ab70-cc4fe2fe1fd9 <- USB\CLASS_FE&SUBCLASS_01&PROT_01
  Guid:                 ab182806-032b-5b7b-994f-b0713031d2b6 <- USB\CLASS_FE&SUBCLASS_01
  Guid:                 c418fefd-0db6-5152-a4da-7a7527183cd1 <- USB\CLASS_FE
  Serial:               00000000
  Plugin:               dfu
  Flags:                updatable|registered
  Vendor:               iFi (by AMR)
  VendorId:             USB:0x20B1
  Version:              5.30
  Icon:                 drive-harddisk-usb
  Created:              2019-04-23

8BitDo M30 gamepad
  DeviceId:             672c087de09848d9e7ee32aa1dea2fbeb8b81e6b
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Guid:                 cc93af80-221d-50fe-acb2-de0b278379dd <- USB\VID_2DC8&PID_5006&REV_0001
  Guid:                 49c74d94-86d7-5587-b48f-aa36decb1568 <- USB\VID_2DC8&PID_5006
  Guid:                 5bc60e62-f461-58bd-91b6-7de3392f8bd6 <- USB\VID_2DC8
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|supported|needs-bootloader|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x2DC8
  Version:              1.10
  Icon:                 input-gaming
  InstallDuration:      120
  Created:              2019-04-23

WDC WD30EZRZ-00GXCB0
  DeviceId:             602b0a6cc821d155208724f0e22f8d111542b74c
  Guid:                 aaebb49f-5ebd-5492-bdec-f3e386466879 <- IDE\WDC_WD30EZRZ-00GXCB0____________________80.00A80
  Guid:                 ef46686e-82df-582a-a948-e919a81541d4 <- IDE\0WDC_WD30EZRZ-00GXCB0____________________
  Guid:                 c217d516-e636-5757-8387-072817ae30a9 <- WDC WD30EZRZ-00GXCB0
  Serial:               WD-WCC7K5ADK37N
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              80.00A80
  Icon:                 drive-harddisk
  Created:              2019-04-23

Samsung SSD 970 EVO 500GB
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 47335265-a509-51f7-841e-1c94911af66b <- NVME\VEN_144D&DEV_A808
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf <- NVME\VEN_144D
  Guid:                 5fc1d3d7-f6cf-5f1c-ba7e-18a56f00f262 <- Samsung SSD 970 EVO 500GB
  Serial:               S466NX0K726339N
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              2B2QEXE7
  Icon:                 drive-harddisk
  Created:              2019-04-23
```
I guess the issue can be considered fixed.
Thanks!

commit ae315d89 (2019-04-23): fu-usb-device: Only create class/subclass GUIDs for 0xFE and 0xFF (Fixes: #1162)

These are dangerous GUIDs that shouldn't generally be used by any device
for distribution of firmware updates.  They'll potentially apply to
devices outside of the vendor they were created for.

Additionally if two USB devices with common interface classes are
on the system, they may lead to one device not being populated by
fwupd.
closed 2019-04-23 by hughsie