fwupd/fwupd #1164: Run twice to update a Logitech unifying receiver : failed to update_detach using unifying: failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7

yaneti, 2019-04-23:
```shell
$ fwupdmgr --version
client version:	1.2.8
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.8
```

```shell
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             b6edc1d09e0377a1c692c5d767ea16260851b302
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4 <- HIDRAW\VEN_046D&DEV_C52B
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836 <- HIDRAW\VEN_046D
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0015
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-04-23
.....
```

```shell
# fwupdmgr get-updates
Unifying Receiver has firmware updates:
Device ID:               b6edc1d09e0377a1c692c5d767ea16260851b302
GUID:                    9d131a0c-a606-580f-8eda-80587250b8d6
GUID:                    279ed287-3607-549e-bacc-f873bb9838c4
GUID:                    21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
ID:                      com.logitech.Unifying.RQR12.firmware
Update Version:          RQR12.08_B0030
Update Name:             Unifying Receiver Device Update Update
Update Summary:          Firmware for the Logitech Unifying receiver
Update Remote ID:        lvfs
Update Duration:         7 seconds
Update Checksum:         SHA1(8dabc1e0fdd536c13d1e94c637d300f3541d3bf1)
Update Location:         https://fwupd.org/downloads/690feedb8fe9cb05d1eaf80f903d66381eec7200-Logitech-Unifying-RQR12.08_B0030.cab
Update Description:      This release addresses an encrypted keystroke injection issue known as Bastille security issue #13. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.
                         
                         A few of Logitech's devices used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature. Affected hardware is:
                         
                          • Wireless Mouse M335
                          • Zone Touch Mouse T400
                          • Wireless Mouse M545
                          • Wireless Mouse M560
                          • Touch Mouse M600
                          • Touch Mouse T620
                          • Wireless Rechargeable Touchpad T650
                         
                         Although Logitech does not recommend it, these features may be re-activated by keeping/downgrading the receiver to an older firmware.
                         
# fwupdmgr update 
Downloading RQR12.08_B0030 for Unifying Receiver...
Fetching firmware https://fwupd.org/downloads/690feedb8fe9cb05d1eaf80f903d66381eec7200-Logitech-Unifying-RQR12.08_B0030.cab
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
failed to update_detach using unifying: failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7
# fwupdmgr update 
Downloading RQR12.08_B0030 for Unifying Receiver...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Unifying Receiver…-                                     ]
Updating Unifying Receiver…**************************************]
Writing…                 [***************************************]
# fwupdmgr get-updates
No upgrades for Unifying Receiver, current is RQR12.08_B0030: RQR12.08_B0030=same, RQR12.07_B0029=older
________________________________________________

Devices that were not updated correctly:

 • Unifying Receiver (RQR12.07_B0029 → RQR12.08_B0030)

Devices that have been updated successfully:

 • Unifying Receiver (RQR12.07_B0029 → RQR12.08_B0030)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]: n

```
```
fwupd-1.2.8-1.fc31.x86_64
```

hughsie, 2019-04-23:
If you unplug, and then replug, does it then work?

yaneti, 2019-04-24:
Well, the receiver is already updated to the new firmware, unplugging and plugging it doesn't seem to make a difference here  Those two fwupdmgr update  runs above were run in sequence the second one apparently succeeded..

lots0logs, 2019-09-07:
I can confirm that the first attempt always fails and leaves you without a working keyboard and mouse until you unplug and plug it back in. Now if I use the following command it completes fine the second time and when its done everything works as it should.

```
sudo fwupdmgr update; sleep 3; sudo fwupdmgr update;
```

Here's the output::
```
Downloading RQR24.06_B0030 for Unifying Receiver...
Fetching firmware https://fwupd.org/downloads/0afa79c32ba48db68ccb80670a24980fbf8d8d76-Logitech-Unifying-RQR24.06_B0030.cab
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
failed to update_detach using unifying: failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7
Downloading RQR24.06_B0030 for Unifying Receiver...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Unifying Receiver…-                                     ]
Updating Unifying Receiver…**************************************] Less than one minute remaining…
Writing…                 [***************************************]
```

FFY00, 2019-09-07:
The timeout waiting for the bootloader to come up is too low, should be fixed by https://github.com/fwupd/fwupd/pull/1337.

hughsie, 2019-09-09:
I've added an LVFS "known issue" about this too.

yatesco, 2019-09-30:
(this is happening on Ubuntu 19.10 on ZFS with boot on ext4 but root on ZFS. https://github.com/fwupd/fwupd/issues/1164#issuecomment-529130782 fixes it)

FFY00, 2019-10-30:
/notstale

lots0logs, 2019-12-05:
/notstale

superm1, 2019-12-05:
Those still affected by this, is if still happening on a current release of fwupd? Ideally fwupd 1.3.5.

letsfindaway, 2019-12-07:
It still happens on 1.3.1, so #1337 seems not to fix it. Have not tested on newer versions of fwupd. I'm using openSUSE and 1.3.1 is the latest available in the repositories.

superm1, 2019-12-09:
The only notable thing later than 1.3.1 is that the client will notify you if a replug is needed on device thats take unsigned firmware:
https://github.com/fwupd/fwupd/commit/6c5e7c1e92ed19cc6069081c8e17cd6c799a0e07#diff-57fa81ad95f400e0fe0b23264b44a366

@FFY00 is this one of those devices like that?  If so, that may help.

@letsfindaway can you please run `/usr/libexec/fwupd/fwupdtool get-devices --verbose` and share the output?  We can check the `fu_logitech_hidpp_peripheral_to_string` output to see what features the device claims to confirm if that would help.

letsfindaway, 2019-12-09:
The output of fwupdtool is quite long. See the attached file. I ran the command as root, and on openSUSE it is located at `/usr/lib/fwupd/fwupdtool`.
[fwupdtool-get-devices-verbose.log](https://github.com/fwupd/fwupd/files/3941347/fwupdtool-get-devices-verbose.log)

FFY00, 2019-12-09:
Yes, `BOT01.02_B0015` is unsigned IIRC.

superm1, 2019-12-09:
That log does confirm it's `SignedFirmware:        false`, but yeah that stuff related to needing to replug is peripheral specific and this is operating on a receiver.

superm1, 2019-12-09:
The error appears to come from here: https://github.com/fwupd/fwupd/blob/1.3.1/src/fu-io-channel.c#L144 but we don't look at `errno` to decode the error any further.

Is there anything else in your kernel log about a USB failure or anything at this time?

letsfindaway, 2019-12-09:
I just ran `fwupdmgr update` again and watched `journalctl -f`. The result:

```
family@gepard:~> fwupdmgr update 
Downloading RQR12.10_B0032 for Unifying Receiver...
Entpacken …              [***************************************]
Authentifizierung …      [***************************************]
failed to update_detach using unifying: failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7
```

and not a single line in the log. After the command, I had to unplug and replug the receiver to get it working again.

superm1, 2019-12-09:
I have to wonder did it actually make it into bootloader mode at that time or crash?  Could you by chance run `fwupdmgr get-devices` while in the failed state after running `fwupdmgr update`?

superm1, 2019-12-09:
and if your keyboard is connected via that unifying dongle then I think you can do this:

`fwupdmgr update || fwupdmgr get-devices`

letsfindaway, 2019-12-09:
Only the mouse is connected, but anyway - this time it worked!

```
family@gepard:~> fwupdmgr update || fwupdmgr get-devices
Downloading RQR12.10_B0032 for Unifying Receiver...
Entpacken …              [***************************************]
Authentifizierung …      [***************************************]
Updating Unifying Receiver…
Bereit …                 [***************************************]
Updating Unifying Receiver…**************************************]
Schreiben …              [***************************************]
family@gepard:~> fwupdmgr get-devices                   
Z68MX-UD2H-B3
│
├─Hitachi HDS5C3020ALA632:
│     Device ID:           0a8c36d4c09c803cd6b5861e443fb7a41a20cbe6
│     Zusammenfassung:     ATA Drive
│     Current version:     ML6OA580
│     Vendor:              ATA
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUID:                57402006-3a42-5fd1-a203-b7a8c10ff52a
│                          36171f9c-7d48-5990-b730-30172c6fb75a
│                          753e53b8-94e5-53bc-92a9-40e9e0a27dab
│   
├─M4-CT128M4SSD2:
│     Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
│     Zusammenfassung:     ATA Drive
│     Current version:     070H
│     Vendor:              ATA
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUID:                989b9851-4a4a-55d0-9c5d-30766cc8e158
│                          e1708dcc-5fd8-5471-bfa9-1210ad9c13b6
│                          b6290d01-856c-536d-965d-41677ad46458
│   
├─Samsung SSD 860 EVO 500GB:
│     Device ID:           6008c780ccbd49ca27ab1c91841a356d9f5f55f0
│     Zusammenfassung:     ATA Drive
│     Current version:     RVT01B6Q
│     Vendor:              ATA
│     Flags:               internal|updatable|require-ac|registered|needs-reboot
│     GUID:                f37354a2-fa11-5539-9039-c5194ac4ea16
│                          3831476a-add3-5bc8-88ff-31739e99ad07
│                          bd25ba7e-8dd0-5aed-90dd-db6a84712c8f
│   
└─Unifying Receiver:
      Device ID:           a28b2c3c7111a94a28de87134cacd4c4265c5892
      Zusammenfassung:     A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0014
      Vendor:              Logitech (USB:0x046D)
      Install Duration:    30 seconds
      Update State:        success
      Flags:               updatable|supported|registered
      GUID:                9d131a0c-a606-580f-8eda-80587250b8d6
                           279ed287-3607-549e-bacc-f873bb9838c4
                           a160f16a-613b-5627-a224-ace42013f9a6
    
________________________________________________

Nicht korrekt aktualisierte Geräte:

 • Unifying Receiver (RQR12.08_B0030 → RQR12.10_B0032)

Erfolgreich aktualisierte Geräte:

 • Unifying Receiver (RQR12.08_B0030 → RQR12.10_B0032)

Das Hochladen von Firmware-Berichten hilft Hardwareherstellern, fehlerhafte und erfolgreiche Aktualisierungen auf realen Geräten schnell zu erkennen.
Bericht jetzt hochladen? (Erfordert Internetverbindung) [Y|n]: Y
Ziel:                    https://fwupd.org/lvfs/firmware/report
[... 46 lines cut ...]
                                 "CpuArchitecture" : "x86_64",
                                 "RuntimeVersion(org.freedesktop.fwupd)" : "1.3.1",
                                 "DistroId" : "opensuse-leap",
                                 "CompileVersion(org.freedesktop.fwupd)" : "1.3.1",
                                 "RuntimeVersion(org.kernel)" : "4.12.14-lp151.28.32-default",
                                 "CompileVersion(com.redhat.fwupdate)" : "12",
                                 "CompileVersion(org.freedesktop.gusb)" : "0.2.11",
                                 "RuntimeVersion(org.freedesktop.appstream-glib)" : "0.7.14",
                                 "DistroVersion" : "15.1",
                                 "CompileVersion(com.redhat.efivar)" : "35",
                                 "RuntimeVersion(com.dell.libsmbios)" : "2.4",
                                 "BootTime" : "1575923485",
                                 "RuntimeVersion(com.redhat.fwupdate)" : "12"
                               }
                             }
                           ]
                         }
Mit dem Hochladen fortfahren? [Y|n]: Y
Der Aktualisierungsfehler ist ein bekanntes Problem, besuchen Sie diese URL für weitere Informationen: https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Logitech-Restart-Failure
family@gepard:~> fwupdmgr get-updates 
No upgrades for Unifying Receiver, current is RQR12.10_B0032: RQR12.10_B0032=same, RQR12.08_B0030=older, RQR12.07_B0029=older
```
When I ran the command half an hour before, the error message came just immediately, less than a second. This time the updating process took about 5 seconds.

Of course my receiver is now updated and I can no longer use it for testing the issue, I thing. Good and bad...

superm1, 2019-12-09:
Dang, that's too bad.

superm1, 2019-12-09:
OK well I'm going to mark this closed for now.  If someone else comes back with the same symptoms, please provide the following and we'll re-open:

0. The fwupd version.  If you're less than 1.3.1, please upgrade to 1.3.1 or later to reproduce it and help.
1. The output of `fwupdmgr update` || `fwupdmgr get-devices` (or the the output of `fwupdmgr get-devices` some time after you've run the `update` command but before rebooting ore unplugging dongle)
2. If you can readily reproduce, please turn on some very verbose daemon logging.  Depending on your fwupd version launch the daemon like this:

**1.3.1 ->1.3.3**
```
sudo /usr/libexec/fwupd/fwupd -v --plugin-verbose=unifying
```

**1.3.4 or later**
```
sudo /usr/libexec/fwupd/fwupd -v --plugin-verbose=logitech_hidpp
```
Then run the frontend `fwupdmgr update` and share the daemon's verbose output.

closed 2019-12-09 by superm1

letsfindaway, 2019-12-29:
I now have another Logitech unifying receiver here and wanted to follow the steps you recommended. However I ran into some problems. What I did:

```
linux-tuxedo:~ # /usr/lib/fwupd/fwupd -v --plugin-verbose=unifying >fwupd.log 2>&1
^Z
[1]+  Stopped                 /usr/lib/fwupd/fwupd -v --plugin-verbose=unifying > fwupd.log 2>&1
linux-tuxedo:~ # bg
[1]+ /usr/lib/fwupd/fwupd -v --plugin-verbose=unifying > fwupd.log 2>&1 &
linux-tuxedo:~ # /usr/lib/fwupd/fwupdtool update
Loading…                 [***************************************]
Decompressing…           [ -                                     ]The input is not of cabinet format
linux-tuxedo:~ # fwupdmgr update
Mismatched daemon and client, use fwupdmgr instead
linux-tuxedo:~ # /usr/lib/fwupd/fwupdtool get-updates
Loading…                 [***************************************]
○
│
└─Unifying Receiver:
  │   Device ID:           b8f11848dfcbf441df375b9f11b0e42d085f688b
  │   Summary:             A miniaturised USB wireless receiver
  │   Current version:     RQR24.06_B0030
  │   Bootloader Version:  BOT03.00_B0006
  │   Vendor:              Logitech (USB:0x046D)
  │   Install Duration:    30 seconds
  │   Flags:               updatable|supported|registered
  │   GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  │                        279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
  │ 
  └─Unifying Receiver (RQR24) Device Update:
        Version:           RQR24.10_B0036
        Remote ID:         lvfs
        Summary:           Firmware for the Logitech Unifying Receiver (RQR24.xx)
        License:           LicenseRef-proprietary
        Size:              74.9 kB
        Vendor:            Logitech
        Duration:          30 seconds
        Flags:             is-upgrade
        Description:       This release addresses an encryption-key extraction vulnerability.
      
      w of Logitech's devices used to send select buttons in an unencrypted way, and in an effort to protect against this vulnerability, Logitech removed the feature. Affected hardware is:
      
       Wireless Mouse M335
       Zone Touch Mouse T400
       Wireless Mouse M545
       Wireless Mouse M560
       Touch Mouse M600
       Touch Mouse T620
       Wireless Rechargeable Touchpad T650
      
      ough Logitech does not recommend it, these features may be re-activated by keeping/downgrading the receiver to an older firmware.
      
linux-tuxedo:~ # /usr/lib/fwupd/fwupdtool update
Loading…                 [***************************************]
Decompressing…           [ -                                     ]The input is not of cabinet format
linux-tuxedo:~ # fwupdmgr update
Mismatched daemon and client, use fwupdmgr instead
linux-tuxedo:~ # 
```

The following things are a little bit strange and blocked me:
* I cannot run `fwupdmgr` after having started the daemon. Running `fwupdtool` instead is however possible.
* Using `fwupdtool` I get the error message: The input is not in cabinet format.
* And there are some typos or missing letters in the beginning of sentences in the messages when I execute `get-updates`

@superm1 : can you give me some advice, so that I can create usable debug output for you?

I have attached the log file created during this session.
[fwupd.log](https://github.com/fwupd/fwupd/files/4008415/fwupd.log)

lots0logs, 2020-01-08:
@superm1 I'm also still seeing this issue.

superm1, 2020-01-08:
At this point the debugging of this issue has been brought into #1183.  I think we should have any further conversations and patch testing there.