fwupd/fwupd #1169: Regression reported from 3c8777f0f6f0fb537b65b0747f81ed7aea640f2d

superm1, 2019-04-25:
As a result of 3c8777f0f6f0fb537b65b0747f81ed7aea640f2d one storage vendor reports that they can no longer install firmware on their disks.

```
Firmware version formats were different, device was '(null)' and release is 'plain'
```

To me this leaves one of two options:
1. Add a heuristic specifically for `nvme` and `ata` plugins that the version format is explicitly set to plain (and if necessary we can add quirks for individual vendors)
2. Add a quirk for each storage vendor to set it plain.

Thoughts?

hughsie, 2019-04-25:
> device was '(null)'

Maybe we just neuter the check if either the release **or** device versions are unknown?

superm1, 2019-04-25:
> Maybe we just neuter the check if either the release or device versions are unknown?

I thought that seemed like a good idea, but it ends up failing later too by 2c40b3734 
```
failed to get release version: version format unset and version 13000001 ambiguous
```

hughsie, 2019-04-25:
> version 13000001 ambiguous

This is a valid failure, is this the actual version used by the storage vendor? In that case I think we want to be very explicit about the version format for the FuDevice.

superm1, 2019-04-25:
Yes this is the actual version being used.  I have a production firmware 13000000 and test firmware 13000001.  Before https://github.com/hughsie/fwupd/commit/3c8777f0f6f0fb537b65b0747f81ed7aea640f2d the device was being implicitly set to plain, and release was explicitly set to plain.

hughsie, 2019-04-25:
> Yes this is the actual version being used

13000001 as a version number isn't very friendly for users :( I think we could default to plain, if we stop the autoconversion to triplet/quad -- which would break the old Dell UEFI updates.

I think it would be fine to default to plain in those two plugins tho...

hughsie, 2019-04-25:
> I think it would be fine to default to plain in those two plugins tho

Actually, change of heart. Maybe a quirk for that vendor and then using guess_format for the others?

superm1, 2019-04-25:
> Actually, change of heart. Maybe a quirk for that vendor and then using guess_format for the others?

Well this isn't the fist vendor that has a collection of numbers for a version.. it's just the one being worked with right now.

hughsie, 2019-04-25:
> a collection of numbers

Could we convince them to use a triplet or quad format both for release and device format? It's much easier for users to understand.

superm1, 2019-04-25:
> Could we convince them to use a triplet or quad format both for release and device format? It's much easier for users to understand.

I discussed that, but it leads to confusion comparing Windows and Linux then.

Example from support.dell.com on a platform that supports a variety of SSDs:

[image]

Those happen to have letters in the current release but don't always.

hughsie, 2019-04-25:
I think we just have to add a VersionFormat quirk for that vendor. General rule then becomes "_anything with a version not in semver format (e.g. 1.0.A.B or 1.2.3.4) has to have a version format set as a quirk_".

superm1, 2019-04-25:
>"anything with a version not in semver format (e.g. 1.0.A.B or 1.2.3.4) has to have a version format set as a quirk".

We probably need to have the following to avoid this rabbit hole each time of debugging:
1. Daemon make noise about this (`g_warning` type thing)
or 
2. Remove updatable flag and set `UpdateError` so it's apparent in frontends.

> I think we just have to add a VersionFormat quirk for that vendor.

Hmm, ATA doesn't have a vendor ID to quirk, so that would mean doing it for every single disk instead and also I realize that I can't guarantee the behavior of the version string across !Dell.  How about this:

```
diff --git a/plugins/ata/fu-ata-device.c b/plugins/ata/fu-ata-device.c
index 626b4672..102d9b10 100644
--- a/plugins/ata/fu-ata-device.c
+++ b/plugins/ata/fu-ata-device.c
@@ -181,6 +181,7 @@ fu_ata_device_parse_id_maybe_dell (FuAtaDevice *self, const guint16 *buf)
        guid_efi = fu_ata_device_get_guid_safe (buf, 129);
        if (guid_efi != NULL)
                fu_device_add_guid (FU_DEVICE (self), guid_efi);
+       fu_device_set_version_format (FU_DEVICE (self), FU_VERSION_FORMAT_PLAIN);
 }

 static gboolean
```

commit 8b222bd5 (2019-04-25): ata/nvme: For devices running Dell firmware, set `VersionFormat` (Fixes: #1169)

Dell firmware is known to use `VersionFormat` `plain`.
commit 462cc1a0 (2019-04-27): Revert "ata/nvme: For devices running Dell firmware, set `VersionFormat` (Fixes: #1169)"

This reverts commit 8b222bd55761dd193bba550ad8e30ecc85f62177.
closed 2019-04-25 by superm1