fwupd/fwupd #1171: SuperIO Update failing with snap

Sean-StarLabs, 2019-04-26:
SuperIO updates fail when using the snap, regardless of firmware version or version of fwupd (stable or edge).

The update is successful from when fwupd is installed from source.

```shell
$ fwupdmgr --version
client version:	1.2.4
checkout info:	1.2.3-16-g4c0df63d
compile-time dependency versions
gusb:	0.3.0
efivar:	34
daemon version:	1.2.8
```

Journalctl shows:
```
Apr 25 16:38:01 sean-LabTop systemd[1]: fwupd-offline-update.service: Succeeded.
Apr 25 16:38:01 sean-LabTop systemd[1]: fwupd-offline-update.service: Main process exited, code=killed, status=15/TERM
```

Device to update:
```
SuperIO IT8987
  DeviceId:             1f664ad149f2151fb750b52751dbf9ba045a7846
  Guid:                 0aa07091-dc02-5e62-a0cd-9d3880721b64
  Summary:              Embedded Controller
  Plugin:               superio
  Flags:                internal|updatable|only-offline|require-ac|supported|registered|needs-reboot
  Version:              03.07
  Icon:                 computer
  Created:              2019-04-26
```

Please answer the following questions:

- Operating system and version:
Ubuntu 18.04/19.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
Snap

hughsie, 2019-04-26:
Can you upload your journal logs, starting from the offline-update service startup and finishing with the end of reboot.service please.

Sean-StarLabs, 2019-04-26:
Log is here:
[shortlog.txt](https://github.com/hughsie/fwupd/files/3120677/shortlog.txt)
A few attempts in here:
[journal.txt](https://github.com/hughsie/fwupd/files/3120678/journal.txt)

hughsie, 2019-04-26:
@superm1 -- help? I think there's some interaction with mounting the snap -- perhaps we need some extra WantedBy in the snap case?

superm1, 2019-04-26:
At least looking through the shortlog you still have the hand install in place, it's going to conflate this issue so please can you do the following as we continue:
1. Make sure that you've removed the packaged version
2. Go into your build directory and run `ninja uninstall` to remove most of the hand install related items.

However that being said, I don't believe the snap knows anything about `fwupd-offline-update.service`.
@Sean-StarLabs I've submitted a PR.  Assuming CI passes, can you please build the snap locally and test it?  I have nothing that I can use to test `fwupdoffline`.

@hughsie 
Does the systemd unit need to be manually enabled or anything?  I wasn't sure...

hughsie, 2019-04-26:
> Does the systemd unit need to be manually enabled or anything?

There's a trigger looking for /system-update that changes the boot target. I don't think you need to enable anything.

hughsie, 2019-04-26:
> I have nothing that I can use to test fwupdoffline.

I'm pretty sure you can install any live firmware with --offline -- e.g. a ColorHug or Unifying receiver.

hughsie, 2019-04-26:
Of course, the other way to do this is to do the upgrade on shutdown (e.g. `After=final.target` but before everything is remounted r/w) and then we don't need to futz around with `/system-upgrade` symlinks. I'm also hitting this problem with a Lite installed with Fedora SilverBlue (i.e. OSTree immutable image): https://lists.freedesktop.org/archives/systemd-devel/2019-April/042434.html

Sean-StarLabs, 2019-04-26:
@superm1 Sorry, those logs were from my laptop - using the ubuntu.sh destroyed the install log and I have got around to manually "tidying".

Just trying to build the snap now, getting:
```
Failed to fetch stage packages: Error downloading packages for part 'gettext': The package 'libunistring0' was not found..
```
Will text on a fresh install once I figure this one out!

superm1, 2019-04-26:
> I'm pretty sure you can install any live firmware with --offline -- e.g. a ColorHug or Unifying receiver.

Oh right.. I guess once I have CI working I can try.

> Of course, the other way to do this is to do the upgrade on shutdown (e.g. After=final.target but before everything is remounted r/o) 

You mean basically combine the activate and offline targets?  I don't have any opposition to that.  We already have activate working now, even with snap.

@Sean-StarLabs 
Try building it in a container like docker or lxc if you're not already.

superm1, 2019-04-26:
Well I did an experiment using #1172 (with #1173 added too) and unfortunately got stuck in a boot loop.  I don't really know what's wrong here though.

superm1, 2019-04-26:
OK in single boot mode I get:
```
internal error, please report: running "fwupd.fwupd.offline" failed: timeout waitin for snap system profiles to get updated
```

So I guess there is some extra dependency needed for that service.

Sean-StarLabs, 2019-04-30:
Logs for local snap with #1172:
```
Apr 30 14:11:43 star-LabTop systemd[1]: fwupd-offline-update.service: Triggering OnFailure= dependencies.
Apr 30 14:11:43 star-LabTop systemd[1]: fwupd-offline-update.service: Failed with result 'exit-code'.
Apr 30 14:11:43 star-LabTop systemd[1]: fwupd-offline-update.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
```

[fulllog.txt](https://github.com/hughsie/fwupd/files/3131360/fulllog.txt)
[log.txt](https://github.com/hughsie/fwupd/files/3131361/log.txt)

superm1, 2019-04-30:
Well looking this over, I think there is a bigger problem.  `fwupdoffline` symlinks `/system-update` to a hardcoded `/var/lib/fwupd`.  So any installation not using that for local state directory will fail.

superm1, 2019-04-30:
The commits I've got on that branch get it working for the non-snap non `/usr` prefix case (which are a pre-requisite to making a snap work).
It seems like the problem for the snap is that snap core is not mounting.

superm1, 2019-05-01:
I don't see an obvious way how to get snap core to automatically mount during `system-update` target.  `RequiresMountsFor` [has issues with symlinks ](https://github.com/systemd/systemd/issues/8907) so it can't just be `RequiresMountsFor=/snap/core/current`.  Different versions of core snap will be distributed independently and change so the symlinks to a version of core can't be put in `system-update.target.wants`.  `snapd` puts them in in `multi-user.target.wants` as part of managing the lifecycle.  I think a "proper" solution might be for snapd to also put them in `system-update.target.wants` too.  I'll have to experiment if that actually fixes the behavior, and if so, probably this needs a feature request filed with snapd.

Sean-StarLabs, 2019-05-01:
Is it just our EC's which use the offline mode? If it is, might be easier to just install it on shutdown.

hughsie, 2019-05-01:
> Is it just our EC's which use the offline mode

Yes -- but it's something another vendor has wanted to use for some time. The advantages from my point of view is that the hardware comes up in a known "fresh" state rather than being poked into various legacy states by bits of userspace. Perhaps the pragmatic choice might to be to do it on shutdown or even live when fwupd is installed as a snap.

Sean-StarLabs, 2019-05-01:
>  live when fwupd is installed as a snap.
Can't see any issue's with that, internally we use live more than the offline. Never had any issues, just "hard to read" progress:
[image]

superm1, 2019-05-01:
>  Perhaps the pragmatic choice might to be to do it on shutdown or even live when fwupd is installed as a snap.

Looking through the tree the only that is marked to only install offline is `superio`.  It sounds like from this description that it doesn't "really" need to be done offline.  Is it only marked as offline for safety purposes?  For example keyboard not working while EC is being flashed?  If so, I would lean on:

1. Let's move that to run at shutdown instead of offline (of course can always manually use `--offline`)
2. On the surface tear out offline functionality when run as snap (you won't see `--offline` in `fwupdmgr`)

>  just "hard to read" progress:

That's using `fwupdtool` or `fwupdmgr`?  I don't believe i've seen snap behave that way, it seems like backspace characters aren't working properly.

Sean-StarLabs, 2019-05-01:
> That's using fwupdtool or fwupdmgr? I don't believe i've seen snap behave that way, it seems like backspace characters aren't working properly.

That's using fwupdmgr. Pretty sure it's an Ubuntu exclusive - I've seen the progress work normally on F30 and ZorinOS 15.

superm1, 2019-05-01:
>Pretty sure it's an Ubuntu exclusive - I've seen the progress work normally on F30 and ZorinOS 15.

Using snap on F30 and ZorinOS or using native packages?

Sean-StarLabs, 2019-05-01:
> Using snap on F30 and ZorinOS or using native packages?

The snap.

hughsie, 2019-05-01:
> it seems like backspace characters aren't working properly

It's because we're updating the keyboard controller; all kind of weird events come out the controller when it's in debug mode.

Sean-StarLabs, 2019-05-24:
Fixed with #1208
Thanks for your help :)

commit 254ec3c5 (2019-04-26): Add support for fwupdoffline (Fixes: #1171)
commit 7eb92afd (2019-04-26): Add support for fwupdoffline (Fixes: #1171)
commit c567ab74 (2019-04-26): Add support for fwupdoffline (Fixes: #1171)
commit 72cde350 (2019-04-26): Add support for fwupdoffline (Fixes: #1171)
closed 2019-05-24 by Sean-StarLabs