fwupd/fwupd #1183: Unable update Unifying Receiver

NTMan, 2019-05-02:
```shell
# fwupdmgr -v update
Downloading RQR12.08_B0030 for Unifying Receiver...
(fwupdmgr:3923): FuCommon-DEBUG: 00:18:21.714: creating path /root/.cache/fwupd
(fwupdmgr:3923): FuMain-DEBUG: 00:18:21.714: skpping download as file already exists
(fwupdmgr:3923): Fwupd-DEBUG: 00:18:21.715: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:3923): Fwupd-DEBUG: 00:18:21.716: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:3923): Fwupd-DEBUG: 00:18:21.785: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:3923): Fwupd-DEBUG: 00:18:21.787: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
failed to update_detach using unifying: failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7
```

```shell
# fwupdmgr --version
client version:	1.2.8
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.8
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
# fwupdmgr get-devices
System Product Name System Firmware
  DeviceId:             4fabf32dccf7e94857e50e589f4ff3bf1f5f6d13
  Guid:                 9382663c-0122-518c-b5e4-e9be73e7236e
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.0.8706
  VersionLowest:        0.0.8706
  Icon:                 computer
  Created:              2019-05-02

INTEL SSDPE21D480GA
  DeviceId:             71b677ca0f1bc2c5b804fa1d59e52064ce589293
  Guid:                 bd38a208-a535-5db5-b5f6-b20195a93991 <- NVME\VEN_8086&DEV_2700
  Guid:                 3f235826-4614-56f6-8fd4-b67b346882c9 <- NVME\VEN_8086
  Guid:                 15b9cef9-9279-5a24-8a5e-1b5ea66f2f03 <- INTEL SSDPE21D480GA
  Serial:               PHM2809000LL480BGN
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  VendorId:             NVME:0x8086
  Version:              E2010420
  Icon:                 drive-harddisk
  Created:              2019-05-02

Unifying Receiver
  DeviceId:             438e3180f9785031aa1c01ff6fd7af6b352915b8
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4 <- HIDRAW\VEN_046D&DEV_C52B
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836 <- HIDRAW\VEN_046D
  Guid:                 a160f16a-613b-5627-a224-ace42013f9a6
  Guid:                 941f4133-39e5-56a0-9df6-902547436c5c
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
[... 268 lines cut ...]
├── loader
│   └── entries
│       ├── 54d67d149c1d4b7286a5db6298e033b4-0-rescue.conf
│       ├── 54d67d149c1d4b7286a5db6298e033b4-5.1.0-0.rc6.git2.1.fc31.x86_64.conf
│       ├── 54d67d149c1d4b7286a5db6298e033b4-5.1.0-0.rc6.git3.1.fc31.x86_64.conf
│       └── 54d67d149c1d4b7286a5db6298e033b4-5.1.0-0.rc7.git1.1.fc31.x86_64.conf
├── memtest86+-5.01
├── System.map-5.1.0-0.rc6.git2.1.fc31.x86_64
├── System.map-5.1.0-0.rc6.git3.1.fc31.x86_64
├── System.map-5.1.0-0.rc7.git1.1.fc31.x86_64
├── vmlinuz-0-rescue-54d67d149c1d4b7286a5db6298e033b4
├── vmlinuz-5.1.0-0.rc6.git2.1.fc31.x86_64
├── vmlinuz-5.1.0-0.rc6.git3.1.fc31.x86_64
└── vmlinuz-5.1.0-0.rc7.git1.1.fc31.x86_64

14 directories, 106 files
```

Please answer the following questions:

- Operating system and version: Fedora 31 (Rawhide)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): from Fedora repo
- Have you tried rebooting?: Yes
- Are you using an NVMe disk?: Yes
- Is secure boot enabled (only for the UEFI plugin)?: Yes

mcpi, 2019-05-15:
I also have this issue with Fedora 30.

ruabmbua, 2019-08-15:
Same issue on archlinux.

I managed to solve it by rerunning the update command, when the bootloader was already active. Took a couple of tries, but the it worked.

FFY00, 2019-09-23:
This should be fixed in the latest release.

NTMan, 2019-10-03:
> This should be fixed in the latest release.

I which version?

```
# fwupdmgr -v update
Downloading RQR12.10_B0032 for Unifying Receiver...
(fwupdmgr:30594): FuCommon-DEBUG: 23:24:12.870: creating path /root/.cache/fwupd
(fwupdmgr:30594): Fwupd-DEBUG: 23:24:12.872: looking for os-release at /etc/os-release
(fwupdmgr:30594): FuMain-DEBUG: 23:24:12.872: downloading https://fwupd.org/downloads/8d83954fcf79453738dbeba9615a095bae9caed9-Logitech-Unifying-RQR12.10_B0032.cab to /root/.cache/fwupd/8d83954fcf79453738dbeba9615a095bae9caed9-Logitech-Unifying-RQR12.10_B0032.cab
Fetching firmware https://fwupd.org/downloads/8d83954fcf79453738dbeba9615a095bae9caed9-Logitech-Unifying-RQR12.10_B0032.cab
(fwupdmgr:30594): GLib-GIO-DEBUG: 23:24:12.879: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.879: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.879: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.879: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.879: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.879: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.879: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.879: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.880: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.880: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:30594): GLib-DEBUG: 23:24:12.880: posix_spawn avoided (fd close requested) 
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.880: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.886: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.886: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.886: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.886: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:30594): dconf-DEBUG: 23:24:12.886: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:30594): GLib-GIO-DEBUG: 23:24:12.889: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:30594): GLib-GIO-DEBUG: 23:24:12.896: _g_io_module_get_default: Found default implementation gnutls (GTlsBackendGnutls) for ‘gio-tls-backend’
(fwupdmgr:30594): FuMain-DEBUG: 23:24:13.540: progress: 11%
Downloading…             [****                                   ](fwupdmgr:30594): FuMain-DEBUG: 23:24:13.540: progress: 25%
Downloading…             [*********                              ](fwupdmgr:30594): FuMain-DEBUG: 23:24:13.540: progress: 38%
Downloading…             [**************                         ](fwupdmgr:30594): FuMain-DEBUG: 23:24:13.540: progress: 51%
Downloading…             [*******************                    ](fwupdmgr:30594): FuMain-DEBUG: 23:24:13.616: progress: 65%
Downloading…             [*************************              ](fwupdmgr:30594): FuMain-DEBUG: 23:24:13.616: progress: 78%
Downloading…             [******************************         ](fwupdmgr:30594): FuMain-DEBUG: 23:24:13.617: progress: 91%
Downloading…             [***********************************    ](fwupdmgr:30594): FuMain-DEBUG: 23:24:13.617: progress: 100%
Downloading…             [***************************************]
(fwupdmgr:30594): Fwupd-DEBUG: 23:24:13.624: Emitting ::status-changed() [decompressing]
Decompressing…           [ -                                     ](fwupdmgr:30594): Fwupd-DEBUG: 23:24:13.625: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:30594): Fwupd-DEBUG: 23:24:13.766: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [  -                                    ](fwupdmgr:30594): Fwupd-DEBUG: 23:24:13.768: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
failed to update_detach using unifying: failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7

# fwupdmgr --version
client version:	1.3.2
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.3.2

```

alfC, 2019-11-23:
This problem persists in Fedora and Ubuntu

mcpi, 2019-12-23:
Still seems to be broken in Fedora 31
```
# fwupdmgr update
Upgrade available for Unifying Receiver from RQR12.07_B0029 to RQR12.10_B0032
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: 
Downloading RQR12.10_B0032 for Unifying Receiver...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7
```
```
# fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
```

DominoTree, 2020-01-03:
Seeing the same behavior on F31 as well

```
[nprice@localhost log]$ fwupdmgr --version
client version:	1.3.5
compile-time dependency versions
	gusb:	0.3.1
	efivar:	37
daemon version:	1.3.5
```

```
[nprice@localhost log]$ fwupdmgr update --verbose
Upgrade available for Unifying Receiver from RQR24.07_B0030 to RQR24.11_B0036
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading RQR24.11_B0036 for Unifying Receiver...
(fwupdmgr:14043): FuCommon-DEBUG: 11:21:27.790: creating path /home/nprice/.cache/fwupd
(fwupdmgr:14043): FuMain-DEBUG: 11:21:27.790: skpping download as file already exists
(fwupdmgr:14043): Fwupd-DEBUG: 11:21:27.792: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:14043): Fwupd-DEBUG: 11:21:27.793: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:14043): Fwupd-DEBUG: 11:21:27.839: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:14043): Fwupd-DEBUG: 11:21:27.857: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7
```

```
[nprice@localhost log]$ sudo /usr/libexec/fwupd/fwupd -v --plugin-verbose=unifying
16:16:22:0653 FuDebug              Verbose debugging enabled (on console 1)
16:16:22:0653 FuDebug              setting FWUPD_UNIFYING_VERBOSE=1
16:16:22:0654 FuConfig             loading config values from /etc/fwupd/daemon.conf
16:16:22:0654 Fwupd                looking for os-release at /etc/os-release
16:16:22:0654 FuConfig             loading /usr/share/fwupd/metainfo
16:16:22:0654 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:16:22:0654 XbSilo               file: 0a92b377-4c26-5a99-b87a-3ef20d65382e, current:0a92b377-4c26-5a99-b87a-3ef20d65382e, cached: (null)
16:16:22:0654 XbSilo               loading silo with file contents
16:16:22:0654 FuConfig             using config path of /etc/fwupd
16:16:22:0654 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
16:16:22:0654 FuConfig             loading config from /etc/fwupd/remotes.d/dell-esrt.conf
16:16:22:0654 FuConfig             loading config from /etc/fwupd/remotes.d/vendor.conf
16:16:22:0654 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
16:16:22:0654 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-directory.conf
16:16:22:0655 FuConfig             ordering lvfs-testing=lvfs+1
16:16:22:0655 FuConfig             ignoring unfound remote fwupd
16:16:22:0655 FuConfig             ignoring unfound remote fwupd
16:16:22:0655 FuConfig             ignoring unfound remote fwupd
16:16:22:0655 FuConfig             ignoring unfound remote fwupd
16:16:22:0659 FuEngine             client certificate exists and working
16:16:22:0659 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
16:16:22:0660 FuHistory            got schema version of 5
16:16:22:0660 FuIdle               setting timeout to 7200s
16:16:22:0660 FuHwids              smbios property Manufacturer=Micro-Star International Co., Ltd
16:16:22:0660 FuHwids              smbios property EnclosureKind=3
16:16:22:0660 FuHwids              smbios property Family=To be filled by O.E.M.
16:16:22:0660 FuHwids              smbios property ProductName=MS-7A32
16:16:22:0660 FuHwids              smbios property ProductSku=To be filled by O.E.M.
16:16:22:0660 FuHwids              smbios property BiosVendor=American Megatrends Inc.
16:16:22:0660 FuHwids              smbios property BiosVersion=1.L0
16:16:22:0660 FuHwids              smbios property BiosMajorRelease=05
16:16:22:0660 FuHwids              smbios property BiosMinorRelease=0d
16:16:22:0660 FuHwids              smbios property BaseboardManufacturer=Micro-Star International Co., Ltd
16:16:22:0660 FuHwids              smbios property BaseboardProduct=X370 GAMING PRO CARBON (MS-7A32)
16:16:22:0662 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
16:16:22:0662 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
16:16:22:0662 XbSilo               file: f610fda7-7d97-5488-a849-437ae047d2c9, current:f610fda7-7d97-5488-a849-437ae047d2c9, cached: (null)
16:16:22:0662 XbSilo               loading silo with file contents
16:16:22:0662 FuEngine             remote lvfs-testing not enabled, so skipping
16:16:22:0662 FuEngine             remote vendor not enabled, so skipping
16:16:22:0662 FuEngine             remote vendor-directory not enabled, so skipping
[... 301 lines cut ...]
16:16:23:0225 FuPlugin             performing fu_plugin_device_registered() on uefi
16:16:23:0225 FuDeviceList         ::added 22060c2e17a6933a1d05d92df5f0118e8d3c5616
16:16:23:0226 XbSilo               0.57ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='c24c9c14-bbca-53eb-9a1d-fb7dce88d8bd']/../..|components/component/provides/firmware[@type='flashed'][text()='b43b9980-8146-54c1-999c-ef0c609ea5ba']/../..` limit=1 -> 0 results
16:16:23:0226 FuHistory            get device
16:16:23:0235 FuDevice             using 9d0e0f79e362744fc8bf70d8c8c04fb798772c42 for PCI_SLOT_NAME=0000:1f:00.0
16:16:23:0235 FuPlugin             emit added from optionrom: 9d0e0f79e362744fc8bf70d8c8c04fb798772c42
[... cut ...]
16:16:23:0268 FuCommon             not checking RQR24.07_B0030 as no version format set
16:16:23:0268 FuCommon             not checking RQR24.07_B0030 as no version format set
16:16:23:0268 FuEngine             Emitting PropertyChanged('Status'='idle')
16:16:23:0268 FuMain               Emitting PropertyChanged('Status'='idle')
16:16:23:0270 FuMain               Started with locale en_US.UTF-8
16:16:23:0270 FuMain               Daemon ready for requests
16:16:23:0271 FuMain               FuMain: acquired name: org.freedesktop.fwupd
```

superm1, 2020-01-06:
Potentially this might be fixed by https://github.com/fwupd/fwupd/pull/1691 if you can have a try.

DominoTree, 2020-01-06:
@superm1 Got this branch built and installed.  First run I got a segfault so I restarted the service in verbose mode but got the same output from fwupdmgr (and no further segfaults) - tried but didn't find a core dump 

```
17:35:50 nprice@localhost.localdomain ~ fwupdmgr update -v
Upgrade available for Unifying Receiver from RQR24.07_B0030 to RQR24.11_B0036
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading RQR24.11_B0036 for Unifying Receiver...
(fwupdmgr:3570): FuCommon-DEBUG: 17:35:55.481: creating path /home/nprice/.cache/fwupd
(fwupdmgr:3570): FuMain-DEBUG: 17:35:55.481: skpping download as file already exists
(fwupdmgr:3570): Fwupd-DEBUG: 17:35:55.482: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:3570): Fwupd-DEBUG: 17:35:55.483: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:3570): Fwupd-DEBUG: 17:35:55.517: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:3570): Fwupd-DEBUG: 17:35:55.531: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
failed to detach to bootloader: failed to send: failed to write 7 bytes to 21: Broken pipe
```
```
17:36:30 nprice@localhost.localdomain ~ dmesg | grep fwupd
[   79.518039] traps: fwupd[2665] general protection fault ip:7f3b59008040 sp:7ffe116dc510 error:0 in libfwupdplugin.so.1.0.0[7f3b59007000+22000]
[   79.798918] gnome-software[2369]: segfault at ffffffffffffffc8 ip 00007f934e4a580a sp 00007fffacf7d798 error 5 in libfwupd.so.2.0.0[7f934e4a4000+17000]
```
```
Jan 06 17:38:50 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0456 FuDebug              Verbose debugging enabled (on console 0)
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0456 FuConfig             loading config values from /etc/fwupd/daemon.conf
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0456 FuRemoteList         loading /usr/local/share/fwupd/metainfo
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0456 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0456 XbSilo               file: 2b214398-3036-5f5d-ac2e-436d2f825596, current:2b214398-3036-5f5d-ac2e-436d2f825596, cached: (null)
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0456 XbSilo               loading silo with file contents
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0456 Fwupd                looking for os-release at /etc/os-release
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0456 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs-testing.conf
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0457 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/dell-esrt.conf
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0457 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor.conf
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0457 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/lvfs.conf
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0457 FuRemoteList         loading remotes from /etc/fwupd/remotes.d/vendor-directory.conf
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0457 FuRemoteList         ordering lvfs-testing=lvfs+1
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0457 FuRemoteList         ignoring unfound remote fwupd
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuEngine             client certificate exists and working
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHistory            got schema version of 5
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuIdle               setting timeout to 7200s
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHwids              smbios property Manufacturer=Micro-Star International Co., Ltd
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHwids              smbios property EnclosureKind=3
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHwids              smbios property Family=To be filled by O.E.M.
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHwids              smbios property ProductName=MS-7A32
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHwids              smbios property ProductSku=To be filled by O.E.M.
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHwids              smbios property BiosVendor=American Megatrends Inc.
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHwids              smbios property BiosVersion=1.L0
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHwids              smbios property BiosMajorRelease=05
Jan 06 17:38:50 localhost.localdomain fwupd[4175]: 22:38:50:0463 FuHwids              smbios property BiosMinorRelease=0d
[... cut ...] ']` limit=1 -> 0 results
Jan 06 17:39:32 localhost.localdomain fwupd[4175]: 22:39:32:0531 XbSilo               0.37ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='6228e8e3-99f2-5a20-b326-54f63abb5346']` limit=1 -> 0 results
Jan 06 17:39:32 localhost.localdomain fwupd[4175]: 22:39:32:0531 FuPlugin             882721aa-31ea-59d5-b93a-e04ae4dc6f7f,0640fb59-0e14-5a32-8481-1a807380c35b,6228e8e3-99f2-5a20-b326-54f63abb5346 has no updates, so ignoring device
```

superm1, 2020-01-07:
> @superm1 Got this branch built and installed. First run I got a segfault so I restarted the service in verbose mode but got the same output from fwupdmgr (and no further segfaults) - tried but didn't find a core dump

Thanks for checking.  that's unfortunate you don't have a coredump to go with this crash.  I don't think it's likely introduced by my patch, it's probably something that was already in the code.  If you do happen to repro again and fetch a coredump or abrt or anything like that, really interested to see the backtrace.

> failed to detach to bootloader: failed to send: failed to write 7 bytes to 21: Broken pipe

The error is slightly different (since it's using the async io path instead) but yes same problem essentially.  So I guess we're still at the drawing board.

DominoTree, 2020-01-07:
@superm1 Anything I can do to help?

superm1, 2020-01-07:
Maybe @FFY00 or @hughsie have ideas.  I'm at a loss here now.

FFY00, 2020-01-07:
Can you try on a different computer? I can't reproduce.

NTMan, 2020-01-07:
1. Path not applicable to 1.3.6 version because 
instead `timeout,` there used `1500,`
https://github.com/fwupd/fwupd/blob/1.3.6/plugins/logitech-hidpp/fu-logitech-hidpp-hidpp.c#L79

2. Even after fixing patch for 1.3.6 version.
Not passed some tests

```
ninja: Entering directory `/builddir/build/BUILD/fwupd-1.3.6/x86_64-redhat-linux-gnu'
ninja: no work to do.
 1/18 fwupd-exported-api                      OK       0.01 s 
 2/18 fwupd-self-test                         OK       0.01 s 
 3/18 fwupdplugin-exported-api                OK       0.00 s 
 4/18 fwupdplugin-self-test                   OK       2.17 s 
 5/18 fu-self-test                            OK       3.57 s 
 6/18 dfu-self-test                           OK       0.02 s 
 7/18 nitrokey-self-test                      OK       0.02 s 
 8/18 synaptics-prometheus-self-test          OK       0.02 s 
 9/18 wacom-usb-self-test                     OK       0.02 s 
10/18 ata-self-test                           OK       0.02 s 
11/18 logitech-hidpp-self-test                OK       0.02 s 
12/18 optionrom-self-test                     OK       0.02 s 
13/18 tpm-eventlog-self-test                  OK       0.02 s 
14/18 nvme-self-test                          OK       0.01 s 
15/18 redfish-self-test                       OK       0.01 s 
16/18 dell-self-test                          OK       0.02 s 
17/18 synaptics-mst-self-test                 FAIL     0.27 s (killed by signal 6 SIGABRT)
18/18 uefi-self-test                          OK       0.02 s 

Ok:                   17
Expected Fail:         0
Fail:                  1
Unexpected Pass:       0
Skipped:               0
Timeout:               0

The output from the failed tests:

17/18 synaptics-mst-self-test                 FAIL     0.27 s (killed by signal 6 SIGABRT)
```

superm1, 2020-01-07:
@NTMan Yes the PR is dependent upon https://github.com/fwupd/fwupd/commit/2d172534a85d8b86a76b68a10e85d416eda52b1c

The self test failures are unrelated, you can safely ignore them (but we should probably make the tests skip in you failure scenario in the future).

NTMan, 2020-01-07:
@superm1 thank, I applied 1691 and 1693 PR and build new fwupd successfully.
But looks it not fixed the original issue.

```
# fwupdmgr -v update
Upgrade available for Unifying Receiver from RQR12.07_B0029 to RQR12.10_B0032
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading RQR12.10_B0032 for Unifying Receiver...
(fwupdmgr:61197): FuCommon-DEBUG: 02:14:29.485: creating path /root/.cache/fwupd
(fwupdmgr:61197): FuMain-DEBUG: 02:14:29.486: skpping download as file already exists
(fwupdmgr:61197): Fwupd-DEBUG: 02:14:29.487: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:61197): Fwupd-DEBUG: 02:14:29.488: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:61197): Fwupd-DEBUG: 02:14:29.531: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:61197): Fwupd-DEBUG: 02:14:29.532: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
failed to get device after detach: failed to wait for detach replug: device f91d66ed2bddb80159d06b4eefbf7488e63f8dd2 did not come back
```

NTMan, 2020-01-07:
@superm1 ups sorry this is known issue.
https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Logitech-Restart-Failure
I know about it when launching `fwupdmgr get-devices`, but this is not obvious.
https://pastebin.com/SXHcHPys

User-friendly case:

- Instead of error `failed to detach to bootloader: failed to send: failed to write 7 bytes to 21: Broken pipe` `fwupdmgr -v update` should answer `please run "fwupdmgr -v update" again for complete update`

- Or much better retry update inside code without disturb user.

superm1, 2020-01-07:
@NTMan Huh, well that's actually quite enlightening.   I think we still have a real problem here.  The timeout was increased in fwupd 1.3.1 as mentioned in that known issue, but the bootloader detach command is "failing".

@FFY00 So am I right to understand that basically as soon as this receiver receives part of the message it's actually rebooting immediately, which is what's severing the pipe?  If that's the case we probably just need to capture this "failure" into debug output and fwupd will do it's wait on the timeout.

I've just pushed another fix onto PR #1691 that will make that happen.  @NTMan I guess you're not able to help much more since you updated now, but anyone else like @DominoTree if you can try to use that other commit and see if it helps now?

NTMan, 2020-01-07:
@superm1  I'm still in the game. I have another identical mouse. :)

```
# fwupdmgr -v update
No upgrades for Unifying Receiver: current version is RQR12.10_B0032: RQR12.10_B0032=same, RQR12.08_B0030=older, RQR12.07_B0029=older
Upgrade available for Unifying Receiver from RQR12.07_B0029 to RQR12.10_B0032
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: 
Downloading RQR12.10_B0032 for Unifying Receiver...
(fwupdmgr:86045): FuCommon-DEBUG: 03:16:57.436: creating path /root/.cache/fwupd
(fwupdmgr:86045): FuMain-DEBUG: 03:16:57.436: skpping download as file already exists
(fwupdmgr:86045): Fwupd-DEBUG: 03:16:57.437: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:86045): Fwupd-DEBUG: 03:16:57.438: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:86045): Fwupd-DEBUG: 03:16:57.478: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:86045): Fwupd-DEBUG: 03:16:57.480: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
(fwupdmgr:86045): Fwupd-DEBUG: 03:17:21.834: Emitting ::device-added(df3a127cacbce803ad9252e02259893adb21efb3)
failed to get device after detach: failed to wait for detach replug: device 9722ecebfd53e00bb40f506cb82635019c206da8 did not come back
```

superm1, 2020-01-07:
OK.. So this other one waited the full 30 seconds but never came back?
Can you check `lsusb` output while in that state (failed to get device after detach)?  And `fwupdmgr get-devices` output?

NTMan, 2020-01-07:
> OK.. So this other one waited the full 30 seconds but never came back?

1) When mouse moved during the update the message `failed to get device after detach: failed to wait for detach replug: device 2a739ab896d4497673555a205e9ce282f0529edf did not come back` appears much early (estimated after 10s) the mouse stops working.
Demonstration: https://youtu.be/JtJnJjYYbXw

2) If mouse leave motionless during the update the message `failed to get device after detach: failed to wait for detach replug: device 2a739ab896d4497673555a205e9ce282f0529edf did not come back` appears later  (estimated after 40s) and the mouse still working.
Demonstration: https://youtu.be/RHKyTd1aFD0

> Can you check lsusb output while in that state (failed to get device after detach)? And fwupdmgr get-devices output?

https://pastebin.com/ZErJsxA0

superm1, 2020-01-08:
Thanks for sharing all that.  Your lsusb output, was that from the first or second condition (moving or not moving mouse)?  I'd suspect it's from motionless during update.  I'm wondering if the first one we have a logitech unifying bootloader in usb output.

@FFY00 is there something else we're missing for the reboot sequence on this particular dongle maybe?

NTMan, 2020-01-08:
> Thanks for sharing all that. Your lsusb output, was that from the first or second condition (moving or not moving mouse)?

If the mouse was motionless both `lsusb` are identical:
[image]

If the mouse was moving:
then string:
`Bus 001 Device 042: ID 046d:c52b Logitech, Inc. Unifying Receiver`
after `fwupdmgr -v update` will replaced on string:
`Bus 001 Device 043: ID 046d:aaaa Logitech, Inc. USB2.0 Hub`
[image]

FFY00, 2020-01-08:
`046d:c52b` is the receiver, `046d:aaaa` is the receiver in bootloader mode, which we want in order to update the device.

If you try to update again, it should bring back the receiver. Can you try?

superm1, 2020-01-08:
So then there are two questions:
1. When motionless why doesn't it go to bootloader mode? Something missing to make it happen?
2. When motion why doesn't it pick up the bootloader as the replacement device?

mcpi, 2020-02-09:
This still seems to be a problem with 1.3.7

	# fwupdmgr --version
	client version:	1.3.7
	compile-time dependency versions
		gusb:	0.3.3
		efivar:	37
	daemon version:	1.3.7`

	# fwupdmgr update
	Upgrade available for Unifying Receiver from RQR12.07_B0029 to RQR12.10_B0032
	Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: 

	Downloading RQR12.10_B0032 for Unifying Receiver...
	Decompressing…           [***************************************]
	Authenticating…          [***************************************]
	failed to detach to bootloader: failed to send: failed to write: wrote -1 of 7

closed 2020-02-11 by hughsie

mcpi, 2020-02-15:
I can confirm that 1.3.8 fixes the problem. Thanks!

NTMan, 2020-03-19:
I am also can confirm that's update working fine even if plugged several Unifying Receivers.
Only one little problem at first launch `Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached`

```
[root@localhost ~]# fwupdmgr update
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
[root@localhost ~]# fwupdmgr update
• INTEL SSDPE21D480GA has no available firmware updates
• ST12000NE0007-2GT116 has no available firmware updates
• System Firmware has no available firmware updates
Upgrade available for Unifying Receiver from RQR12.07_B0029 to RQR12.10_B0032
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading RQR12.10_B0032 for Unifying Receiver...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Unifying Receiver…
Idle…                    [***************************************]
Updating Unifying Receiver…**************************************]
Writing…                 [***************************************]
Successfully installed firmware
• Unifying Receiver has the latest available firmware version
[root@localhost ~]# fwupdmgr update
• INTEL SSDPE21D480GA has no available firmware updates
• ST12000NE0007-2GT116 has no available firmware updates
• System Firmware has no available firmware updates
• Unifying Receiver has the latest available firmware version
• Unifying Receiver has the latest available firmware version
[root@localhost ~]# fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

speachy, 2021-03-05:
I'm running into something similar with a Logitech K400 (unifying receiver) and fwupd 1.5.7:
```
[pizza@kalamata ~]$ sudo fwupdmgr update
Devices with no available firmware updates: 
 • CT2000MX500SSD1
 • Samsung SSD 860 EVO 1TB
 • USB3.0 Hub
 • USB3.0 Hub
 • USB3.0 Hub
 • USB3.0 Hub
Devices with the latest available firmware version:
 • Jabra SPEAK 510 USB
 • System Firmware
 • UEFI dbx
Upgrade available for Unifying Receiver from RQR12.08_B0030 to RQR12.10_B0032
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: 
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
failed to get device after detach: failed to wait for detach replug: device 6ffdb335d0c526616e5e506389937101e5606428 did not come back
[pizza@kalamata ~]$ fwupdmgr --version
client version: 1.5.7
compile-time dependency versions
        gusb:   0.3.5

daemon version: 1.5.7
```
The only thing of note in the system log is this:
```
Mar 04 21:20:29 kalamata.fl.shaftnet.org kernel: usb 3-3.4.4.4.4: USB disconnect, device number 40
Mar 04 21:20:34 kalamata.fl.shaftnet.org fwupd[234264]: 02:20:34:0223 FuDevice             disabling polling: failed to open /dev/hidraw2
```
And the relevant kernel logs:

```
[112440.962992] usb 3-3.4.4.4.3: new full-speed USB device number 43 using xhci_hcd
[112441.054006] usb 3-3.4.4.4.3: New USB device found, idVendor=046d, idProduct=c52b, bcdDevice=12.08
[112441.054013] usb 3-3.4.4.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[112441.054017] usb 3-3.4.4.4.3: Product: USB Receiver
[112441.054020] usb 3-3.4.4.4.3: Manufacturer: Logitech
[112441.067194] logitech-djreceiver 0003:046D:C52B.0033: hiddev96,hidraw2: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:00:14.0-3.4.4.4.3/input2
[112441.180431] input: Logitech K400 as /devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3.4/3-3.4.4/3-3.4.4.4/3-3.4.4.4.3/3-3.4.4.4.3:1.2/0003:046D:C52B.0033/0003:046D:4024.0034/input/input84
[112441.180996] logitech-hidpp-device 0003:046D:4024.0034: input,hidraw4: USB HID v1.11 Keyboard [Logitech K400] on usb-0000:00:14.0-3.4.4.4.3/input2:1
[112534.466584] usb 3-3.4.4.4.3: USB disconnect, device number 43
[112535.939002] usb 3-3.4.4.4.3: new full-speed USB device number 44 using xhci_hcd
[112536.028744] usb 3-3.4.4.4.3: New USB device found, idVendor=046d, idProduct=aaaa, bcdDevice= 1.02
[112536.028752] usb 3-3.4.4.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[112536.028755] usb 3-3.4.4.4.3: Product: USB BootLoader
[112536.028758] usb 3-3.4.4.4.3: Manufacturer: Logitech
[112536.033143] hid-generic 0003:046D:AAAA.0035: hiddev96,hidraw2: USB HID v1.11 Device [Logitech USB BootLoader] on usb-0000:00:14.0-3.4.4.4.3/input0
[112536.084875] hid-generic 0003:046D:AAAA.0036: hiddev96,hidraw2: USB HID v1.11 Device [Logitech USB BootLoader] on usb-0000:00:14.0-3.4.4.4.3/input0
```

hughsie, 2021-03-05:
@speachy would you be able to test fwupd git master? I added a few related unifying fixes a few days ago. Thanks.

dplanella, 2021-03-23:
I had a similar issue and [this workaround](https://askubuntu.com/a/1210342/9781) helped me in the meantime until the fixes land. Essentially it's a script that calls `fwupdmgr` twice (once for downloading and once for writing?), with a delay in between. I simply followed the steps and they worked for me:

```
cat ./uniupdate.sh 
#!/usr/bin/env bash

fwupdmgr update
sleep 5
fwupdmgr update
```

```
$ sudo ./uniupdate.sh 
Upgrade available for Unifying Receiver from RQR24.07_B0030 to RQR24.11_B0036
Unifying Receiver and all connected devices may not be usable while updating. Continue with update? [Y|n]: y
Downloading RQR24.11_B0036 for Unifying Receiver...
Fetching firmware https://fwupd.org/downloads/b30e729cc544c711a6ad947cef9ce6614b394a7b-Logitech-Unifying-RQR24.11_B0036.cab
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
failed to get device after detach: failed to wait for detach replug: device 47a010bb1b3f1f1a91006cf50db58d2189404511 did not come back
Upgrade available for Unifying Receiver from RQR24.00_B0000 to RQR24.11_B0036
Downloading RQR24.11_B0036 for Unifying Receiver...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Unifying Receiver…-                                     ]
Writing…                 [***************************************]
failed to get device after update: failed to wait for detach replug: device 3d86113b510501373fe0f0946e4b1846bc34628c did not come back
```

After the last failure to come back after update, I ran `fwupdmgr` manually, and it reports that the unifying receiver's firmware is now up to date:

```
$ sudo fwupdmgr update
• Unifying Receiver has the latest available firmware version
```

hughsie, 2021-03-23:
@dplanella do you know if a newer version fixes things?

speachy, 2021-04-05:
> @speachy would you be able to test fwupd git master? I added a few related unifying fixes a few days ago. Thanks.

I'm glad to report fwupd 1.5.8 resolved this issue for me

dplanella, 2021-04-30:
> @dplanella do you know if a newer version fixes things?

Sorry, I never got to update to a new version and try. Essentially the workaround fixed the issue for me and I've not experienced it since.

LinuxOnTheDesktop, 2021-08-04:
I too see a 'broken pipe' message and the message seems to have to do with one of those 'receivers'. For, I see the following when using the flatpak version of fwupd.

```
$ sudo -i flatpak run org.freedesktop.fwupd verify-update
Loading…                 [***************************************]
WARNING: This package has not been validated, it may not work properly.
Choose a device:
0.	Cancel
1.	5792b48846ce271fab11c4a545f7a3df0d36e00a ((null))
2.	4bde70ba4e39b28f9eab1628f9dd6e6244c03027 ((R) Core™ i7-8550U CPU @ 1.80GHz)
3.	2292ae5236790b47884e37cf162dcf23bfcd1c60 (Embedded Controller)
4.	58bd405f31c48e6eca290b425f530a94c91e955c (Event Log)
5.	349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
6.	04e17fcf7d3de91da49a163ffe4907855c3648be (LENSE20512GMSP34MEAT2TA)
7.	a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
8.	c6a80ac3a22083423992a3cb15018989f37834d6 (TPM)
9.	9e4ed2585a7c7ad2de1e4593bb8b631301584cea (Thunderbolt host controller)
10.	f95c9218acd12697af946874bfe4239587209232 (UEFI Device Firmware)
11.	362301da643102b9f38477387e2193e57abaa590 (UEFI dbx)
12.	fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027 (Unifying Receiver)
12
device fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027 does not support verification

** (process:16188): WARNING **: 01:48:18.067: Error writing credentials to socket: Error sending message: Broken pipe
```

hughsie, 2021-08-04:
> ** (process:16188): WARNING **: 01:48:18.067: Error writing credentials to socket: Error sending message: Broken pipe

I don't think that message is from fwupd. From PolicyKit perhaps? Although that would be odd as fwupdtool doesn't use the polkit agent. Maybe you can backtrace on that message and debug it a little? Thanks.

LinuxOnTheDesktop, 2021-08-04:
Please explain what you mean by 'backtrace'.

cconvey, 2023-06-11:
I still get this with Pop!_OS 22.04:
``` bash
$ apt show fwupd
Package: fwupd
Version: 1.8.12-2~1683052290~22.04~48fed3d
...
```

Here's my experience, using a known workaround:
```
$ sudo fwupdmgr update; sleep 3; sudo fwupdmgr update;
Devices with no available firmware updates: 
 • SKHynix HFS001TDE9X084N
 • SSD 980 1TB
 • System Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • USB2.1 Hub
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Unifying Receiver from RQR24.07_B0030 to RQR24.11_B0036?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This release addresses an unspecified encryption-key extraction              ║
║ vulnerability.                                                               ║
║                                                                              ║
║ Unifying Receiver and all connected devices may not be usable while          ║
║ updating.                                                                    ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Restarting device…       [                                       ]
failed to detach using logitech_hidpp: failed to detach to bootloader: failed to send: failed to write: Protocol error
Devices with no available firmware updates: 
 • SKHynix HFS001TDE9X084N
 • SSD 980 1TB
 • System Firmware
 • UEFI Device Firmware
 • UEFI dbx
 • USB2.1 Hub
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade Unifying Receiver from RQR24.00_B0000 to RQR24.11_B0036?             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This release addresses an unspecified encryption-key extraction              ║
║ vulnerability.                                                               ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y
Waiting…                 [***************************************] Less than one minute remaining…
Successfully installed firmware
```

iterepa, 2023-09-07:
>$ sudo fwupdmgr update; sleep 3; sudo fwupdmgr update;
Thank you, it helped.

commit ff86b2a1 (2020-01-07): logitech_hidpp: Ignore detach failures (Fixes: #1183)

Some logitech devices seem to reboot immediately and the failures then
look like a broken pipe, but are actually the device rebooting.

If the device really did fail to detach after the timeout is done we'll
see a message that the device failed to come back instead.
commit ddfc72db (2020-02-11): logitech_hidpp: Ignore detach failures (Fixes: #1183)

Some logitech devices seem to reboot immediately and the failures then
look like a broken pipe, but are actually the device rebooting.

If the device really did fail to detach after the timeout is done we'll
see a message that the device failed to come back instead.