fwupd/fwupd #1184: Thinkpad T460s Management Engine Update fails to update

atrueba, 2019-05-04:
The machine is a 2017 Thinkpad P40 Yoga (20GQ). A fresh installation of fwupdate and gnome-software shown an available update for "Thinkpad T460s Management Engine", which was in its factory version (170.xxx or 171.xxx, I don't recall). The latest available firmware was v184.60.3561

The update seemed to work: download OK, reboot, BIOS messages about finding and installing a new capsule and reboot again into the OS, but a new run of gnome-software and package refresh still showed the same situation as before: installed version 17x and new available version v184. I repeated the process a couple of times during a few days, but the firmware was never updated. Tried also with the command line (fwupmgr refresh & install), with the same results

I noticed there was also an older update available (v184.55.3510) and figured maybe the updates had to be accumulative, so I downloaded the cab and installed it manually with "fwupdmagr install xxx.cab". This time the process seemed to work and gnome-software now reports that  version as the currently installed one and wants to update to the latest one, but this time neither the automated install nor the manual download/install seem to work: as at the begining, everything looks like it's updating but after the last reboot the firmware is still at v184.55.3510.

I'm guessing the problem here resides in Lenovo's firmware rather that in fwupd. Notice that there are two variants of this firmware: Consumer (that's the one I'm having trouble with) and Corporate.

```
$ fwupdmgr --version
client version:	1.2.6
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.6
```

```
$ fwupdmgr get-devices
20GQCTO1WW System Firmware
  DeviceId:             8be4cbd5f33a57bd5ac69de9004c4e5ecc91325f
  Guid:                 b8d73eb5-feba-42e1-a106-04540b6ae7d6
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Checksum:             SHA1(de773864c70add85c32c36c8ee79e985c47c4837)
  Version:              0.1.55
  VersionLowest:        0.1.55
  Icon:                 computer
  Created:              2019-05-04

UEFI Device Firmware
  DeviceId:             aff24b50c0a8c292c6c2a81d3827c36d7a86f62a
  Guid:                 03d3297b-3851-4379-95ad-12e21a96c80a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.55.3510
  VersionLowest:        184.55.3510
  Icon:                 audio-card
  Created:              2019-05-04
  UpdateState:          success

SanDisk SD8TB8U256G1001
  DeviceId:             5dbeb140337f610d54913c0f43d15fdaa0eafb51
  Guid:                 9bbc7ebe-9b6f-5f43-a56e-e5037c19ec56
  Guid:                 818c75b7-74ff-5960-85e0-cdce1cb7b55a
  Guid:                 9a64ac8d-a549-58b3-afb0-277bdd6ac025
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              X4133101
  Icon:                 drive-harddisk
  Created:              2019-05-04
```

```
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0017,0018,0019,001A,001B,001C,0002
Boot0000  Windows Boot Manager	HD(1,GPT,e15f6a59-b630-4bb2-aef7-7700d5b24ca7,0x800,0x100000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* debian	HD(1,GPT,e15f6a59-b630-4bb2-aef7-7700d5b24ca7,0x800,0x100000)/File(\EFI\debian\grubx64.efi)
Boot0002* Linux-Firmware-Updater	HD(1,GPT,e15f6a59-b630-4bb2-aef7-7700d5b24ca7,0x800,0x100000)/File(\EFI\debian\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001A* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot001E* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot001F* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0020* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```
$ efivar -l | grep fw
$ tree /boot

****** efivar and tree are not installed
****** fwupdx64 resides in /boot/efi/EFI/debian
****** the .cab files prepared to install where available inside  /boot/efi/EFI/debian/fw. At the moment that folder is empty
```

Please answer the following questions:

- Operating system and version: Debian 10.0. A few packages from testing & experimental
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? No

Forbo, 2019-05-30:
I'm running into the same issue. ThinkPad T460s with the Corporate version of the firmware. Attemping to install LVFS 184.60.3561 (details can be found here: [https://fwupd.org/lvfs/device/com.lenovo.ThinkPadN1CRM.firmware] )Repeated installs that appear to succeed only to get the same update notification upon login.

Just a couple notes I wanted to include regarding the issue reporting template: It appears that `efivar -l | grep fw` didn't return all the results due to case sensitivity. Once I changed it to `grep -i` it returned the full results. Secondly, it appears that you'll want to run `tree /boot` using either `sudo` or a root shell, otherwise the contents of `/boot/efi` won't be included. Should these be submitted as a different issue? I'm not familiar with GitHub's issue templates....

Anywho, on to the good stuff.

```$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

```$ fwupdmgr get-devices
UEFI Device Firmware
  DeviceId:             25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f
  Guid:                 63451986-0c42-42db-9a18-342a8455ac68
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              176.12.1008
  VersionLowest:        176.12.1008
  Icon:                 audio-card
  Created:              2019-05-30

20FWCTO1WW System Firmware
  DeviceId:             444a229da2b21bc0d24faf814d1c58c5bff3315f
  Guid:                 42f29620-3d63-4f09-950b-9b7055570f28
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.2.12
  VersionLowest:        0.2.12
  Icon:                 computer
  Created:              2019-05-30
```

```$ efibootmgr -v
BootNext: 0001
BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0000,001C,0017,0018,0019,001A,001B,0022,0001
Boot0000* ubuntu	HD(1,GPT,916c2410-6736-4f2a-aa2e-e99e4e540797,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,916c2410-6736-4f2a-aa2e-e99e4e540797,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017  USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018  USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019  NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A  ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D  PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot001F* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot0020* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0021* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0022* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```$ efivar -l | grep -i fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-FWUPDATE_DEBUG_LOG
0abba7dc-e516-4167-bbf5-4d9d1c739416-FWUPDATE_VERBOSE
[... 24 lines cut ...]
fwupdate.c:607:search_file(): Device Path prepared: HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797)
fwupdate.c:619:search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x17,0x0)/Sata(0x0,0x0,0x0)/HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797)
fwupdate.c:626:search_file(): Comparing: HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797) and PciRoot(0)/Pci(0x17,0x0)/Sata(0x0,0x0,0x0)/HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797)
fwupdate.c:626:search_file(): Comparing: HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797) and Pci(0x17,0x0)/Sata(0x0,0x0,0x0)/HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797)
fwupdate.c:626:search_file(): Comparing: HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797) and Sata(0x0,0x0,0x0)/HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797)
fwupdate.c:626:search_file(): Comparing: HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797) and HD(Part1,Sig916C2410-6736-4F2A-AA2E-E99E4E540797)
fwupdate.c:636:search_file(): Match up! Returning \EFI\ubuntu\fw\fwupdate-4uuXQP.cap
fwupdate.c:647:search_file(): File \EFI\ubuntu\fw\fwupdate-4uuXQP.cap searched
fwupdate.c:708:open_file(): Found device
fwupdate.c:715:open_file(): Found volume
fwupdate.c:723:open_file(): Found file
fwupdate.c:1050:add_capsule(): Read file; 7522928 bytes
fwupdate.c:1051:add_capsule(): updates guid: 63451986-0C42-42DB-9A18-342A8455AC68
fwupdate.c:1052:add_capsule(): File guid: 63451986-0C42-42DB-9A18-342A8455AC68
[... cut ...]
fwupdate.c:1304:efi_main(): n_updates: 1
fwupdate.c:1154:apply_capsules(): QueryCapsuleCapabilities: Success max: 521900032 reset:1
fwupdate.c:1155:apply_capsules(): Capsules: 1
fwupdate.c:1330:efi_main(): fwupdate: Reset System```

- Operating system and version: Ubuntu 18.04.2 LTS
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get (included as part of base install)
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? No.
- Is secure boot enabled (only for the UEFI plugin)? Yes.

PhilippWendler, 2019-06-03:
I have a ThinkPad T460**p** (20FXS1B700), and fwupd also shows the update for the T460**s** Management Engine, and while upgrades seem to work (I get the message during reboot), the result is still having the old firmware. Secure boot is disabled.

```
$ fwupd.fwupdmgr --version
client version: 1.2.9
checkout info:  1.2.9
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.9
```
I am running Ubuntu 18.04 but installed fwupd via snap.

```
$ fwupd.fwupdmgr get-devices
20FXS1B700 System Firmware
  DeviceId:             444a229da2b21bc0d24faf814d1c58c5bff3315f
  Guid:                 42f29620-3d63-4f09-950b-9b7055570f28
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.2.11
  VersionLowest:        0.2.11
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-06-03

UEFI Device Firmware
  DeviceId:             25127f5d4d5dfd6e903cda1a61a67cb092e4bb0f
  Guid:                 63451986-0c42-42db-9a18-342a8455ac68
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              176.12.1008
  VersionLowest:        176.12.1008
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-06-03
  Modified:             2019-06-03
  UpdateState:          needs-reboot
```
For the "System Firmware", `get-updates` shows
```
No upgrades for 20FXS1B700 System Firmware, current is 0.2.11: 0.1.37=older, 0.1.39=older, 0.1.40=older, 0.1.41=older, 0.1.42=older, 0.1.43=older
```
I think this is also weird, the notebook is from 2016 and the firmware was never updated, but still it should be newer than what is present online?

And for the "UEFI Device Firmware" I get the same versions `184.60.3561` and `184.55.3510` that the OP has.

mbevc1, 2019-07-07:
Same happens on Lenovo X260 reporting `Lenovo ThinkPad T460s Embedded Controller Firmware Version 1.14` is available, updates, but after a reboot still reporting old version.

PhilippWendler, 2019-07-08:
Now version `184.65.3590` of the "Lenovo ThinkPad T460s Corporate ME Firmware" is offered for my T460p, but it does not work either.

atrueba, 2019-07-08:
Same here with v3590.

Running "fwupdmgr get-updates" from a terminal shows a couple interesting
things:

- The currently installed version is still v3510
- Despite of that, there's an entry in "devices that have been updated
succesfully" which mentions an upgrade to v3561
- The "updateError" string is "failed to update_prepare using upower:
Cannot install update when not on AC power unless forced". The system was
connected to AC and fully charged during the upgrade attempt, so maybe the
root cause here is a misdetection of the power state.

atrueba, 2019-07-08:
Another followup: after the previous message's situation, running "fwupdmgr
update --force" results in the following:

- v3510 is still the current version
- The mention of v3561 having been installed has dissapeared
- There's no "updateError" string, so no idea why that las attempt may have
failed.

Also tried downloading the v3590 .cab file and "fwupdmgr install xxx.cab",
but the situation is the same as above.

atrueba, 2019-07-30:
Hi,

(un)fortunately I cannot follow this bug anymore. Had to install a dual-boot system to Windows 10, and a full upgrade from there did finally apply v3590. Once applied, Linux recognizes it as current and does not suggest any further action.

closed 2019-09-23 by superm1

PhilippWendler, 2019-09-24:
This is still a problem with the latest 184.65.3590 firmware. Please reopen.