fwupd/fwupd #1190: Thinkpad T480 Could not apply capsules: Unsupported

lamarios, 2019-05-10:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.8
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.8

```

```shell
$ fwupdmgr get-devices
T480 Thunderbolt Controller
  DeviceId:             ca2629c7b703826e7af330f97e61ce0fbac4d5fc
  Guid:                 7d131734-64ed-5865-a46e-a67cfb9b5a46
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  Icon:                 computer
  Created:              2019-05-10

20L5CTO1WW System Firmware
  DeviceId:             48477ba14bd9bd0c97323a07caa54b7766f917c2
  Guid:                 dc7f0308-1ef8-4774-9ba5-89a58c4d731c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.19
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2019-05-10
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             ea0fd60df22477668e549841f6a22a8c31cae587
  Guid:                 9e21f98b-fe98-455c-b388-da5450ab6979
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.60.3561
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-05-10

UEFI Device Firmware
  DeviceId:             c19580db344e418cf2b23297bcf46d75c06e432a
  Guid:                 b87a926d-189c-49a1-b1ff-921099b1de89
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.11
  VersionLowest:        0.1.11
  Icon:                 audio-card
  Created:              2019-05-10

UEFI Device Firmware
  DeviceId:             b1825e2b03a9ebdea247a74f1f3ddee2a21ac565
  Guid:                 24aac206-3921-45c0-a79e-0a607b11aef0
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              1.3.1084
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-05-10

UEFI Device Firmware
  DeviceId:             a885679bdb552409a5a7e6245e7d6911b500f5cc
[... 420 lines cut ...]
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
└── vmlinuz-linux

10 directories, 343 files

```

Please answer the following questions:

- Operating system and version: Arch
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

I'm having the same exact issue with a thinkpad X280 as well, same OS, same logs very similar installs.

hughsie, 2019-05-10:
At what point did you get the "Could not apply capsules: Unsupported" message?

lamarios, 2019-05-10:
After rebooting to apply the upgrade. It's showing on top of the Lenovo logo before the grub menu.

hughsie, 2019-05-10:
Can you do `sudo /usr/libexec/fwupd/fwupdate -d` then schedule the update again, then reboot, then call `sudo /usr/libexec/fwupd/fwupdate -L` and attach the output here please.

lamarios, 2019-05-10:
```
../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c-0../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)/\efi\EFI\arch\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap...../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and PciRoot(0)/Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \efi\EFI\arch\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 44198 bytes../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:564:fwup_add_update_capsule(): Checking GOP for ux capsule../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)/\efi\EFI\arch\fw\fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c.cap...../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and PciRoot(0)/Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \efi\EFI\arch\fw\fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c.cap../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 9417280 bytes../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: DC7F0308-1EF8-4774-9BA5-89A58C4D731C../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: DC7F0308-1EF8-4774-9BA5-89A58C4D731C../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:727:efi_main(): n_updates: 2../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:593:fwup_apply_capsules(): Could not query capsule capabilities: Unsupported../fwupd-1.2.8/plugins/uefi/efi/fwupdate.c:742:efi_main(): Could not apply capsules: Unsupported%  
```

hughsie, 2019-05-10:
> Could not query capsule capabilities

This means that `RT->QueryCapsuleCapabilities()` failed -- one sec, I'll get you a patch to try...

hughsie, 2019-05-10:
@lamarios can you install the wip/hughsie/uefi branch please and try again please? I'll need the `fwupdate` steps as before.

lamarios, 2019-05-10:
I'm trying to compile with 
```
meson build
```

but i'm stuck at 
```
Project specifies a minimum meson_version '>=0.47.0' but uses features which were added in newer versions:
 * 0.50.0: {'install arg in configure_file'}

```

lamarios, 2019-05-10:
trying to install, but the PKGBUILD in the contrib folder doesn't seem to work
```
 logit@gz-t480  ~/dev/fwupd/contrib   wip/hughsie/uefi  makepkg -si
==> Making package: fwupd #VERSION#-1 (Fri 10 May 2019 17:56:50 +08)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
==> Extracting sources...
==> Removing existing $pkgdir/ directory...
==> Starting build()...
/home/logit/dev/fwupd/contrib/PKGBUILD: line 18: cd: fwupd: No such file or directory
==> ERROR: A failure occurred in build().

```

I can install using sudo ninja -C build install after running meson to compile but when I try to run it I get 
```
fwupdmgr --version        
fwupdmgr: error while loading shared libraries: libfwupd.so.2: cannot open shared object file: No such file or directory

```

superm1, 2019-05-10:
Try running the arch script in contrib/CI when you are in root of fwupd tree.

lamarios, 2019-05-10:
alright, got it.

Still no dice but different error though:
```
../fwupd/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule
../fwupd/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c-0
../fwupd/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule
../fwupd/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)/\efi\EFI\arch\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap...
../fwupd/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and PciRoot(0)/Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \efi\EFI\arch\fw\fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
../fwupd/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../fwupd/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../fwupd/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../fwupd/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 57218 bytes
../fwupd/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697
../fwupd/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: 3B8C8162-188C-46A4-AEC9-BE43F1D65697
../fwupd/plugins/uefi/efi/fwupdate.c:564:fwup_add_update_capsule(): Checking GOP for ux capsule
../fwupd/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule
../fwupd/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)/\efi\EFI\arch\fw\fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c.cap...
../fwupd/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and PciRoot(0)/Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Pci(0x1D,0x2)/Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Pci(0x0,0x0)/Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and Path(131,23,0x01000000001B448B4482908F84012C00)/HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B) and HD(Part1,SigFF08F16E-B7D0-4862-BD80-BCEA0A9F539B)
../fwupd/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \efi\EFI\arch\fw\fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c.cap
../fwupd/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../fwupd/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../fwupd/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../fwupd/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 9417280 bytes
../fwupd/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: DC7F0308-1EF8-4774-9BA5-89A58C4D731C
../fwupd/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: DC7F0308-1EF8-4774-9BA5-89A58C4D731C
../fwupd/plugins/uefi/efi/fwupdate.c:727:efi_main(): n_updates: 2
../fwupd/plugins/uefi/efi/fwupdate.c:593:fwup_apply_capsules(): Could not query capsule capabilities: Unsupported
../fwupd/plugins/uefi/efi/fwupdate.c:598:fwup_apply_capsules(): Capsules: 2
../fwupd/plugins/uefi/efi/fwupdate.c:604:fwup_apply_capsules(): Could not apply capsule update: Not Found
../fwupd/plugins/uefi/efi/fwupdate.c:742:efi_main(): Could not apply capsules: Not Found%

```

lamarios, 2019-05-11:
Update: I tried with the branch on my X280, some updates were applied, but one failed at the end :
```

../fwupd/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-508f7539-1ad6-48b9-8680-38377535009d-0
../fwupd/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-6d28cd9f-7bcd-4fb9-9f10-0372e2962fc4-0
../fwupd/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule
../fwupd/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-3f4a527b-6588-45b8-b2d3-dc61189b63cb-0
../fwupd/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule
../fwupd/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)/\efi\EFI\arch\fw\fwupd-508f7539-1ad6-48b9-8680-38377535009d.cap...
../fwupd/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \efi\EFI\arch\fw\fwupd-508f7539-1ad6-48b9-8680-38377535009d.cap
../fwupd/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../fwupd/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../fwupd/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../fwupd/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 9417312 bytes
../fwupd/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: 508F7539-1AD6-48B9-8680-38377535009D
../fwupd/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: 508F7539-1AD6-48B9-8680-38377535009D
../fwupd/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule
../fwupd/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)/\efi\EFI\arch\fw\fwupd-6d28cd9f-7bcd-4fb9-9f10-0372e2962fc4.cap...
../fwupd/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and Path(131,23,0x010000005CD2E42B81A0C71284012C00)/HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889) and HD(Part1,Sig32151EB7-6CAE-4EC3-91D7-21F60D8F1889)
../fwupd/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \efi\EFI\arch\fw\fwupd-6d28cd9f-7bcd-4fb9-9f10-0372e2962fc4.cap
../fwupd/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../fwupd/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../fwupd/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../fwupd/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 2198992 bytes
../fwupd/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: 6D28CD9F-7BCD-4FB9-9F10-0372E2962FC4
[... cut ...]
../fwupd/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: 3F4A527B-6588-45B8-B2D3-DC61189B63CB
../fwupd/plugins/uefi/efi/fwupdate.c:727:efi_main(): n_updates: 4
../fwupd/plugins/uefi/efi/fwupdate.c:593:fwup_apply_capsules(): Could not query capsule capabilities: Unsupported
../fwupd/plugins/uefi/efi/fwupdate.c:598:fwup_apply_capsules(): Capsules: 4
../fwupd/plugins/uefi/efi/fwupdate.c:604:fwup_apply_capsules(): Could not apply capsule update: Not Found
../fwupd/plugins/uefi/efi/fwupdate.c:742:efi_main(): Could not apply capsules: Not Found

```

hughsie, 2019-05-14:
> efi_main(): n_updates: 2
> efi_main(): n_updates: 4

If you try scheduling the updates one at a time, does it still return unsupported? You can just delete all  of the files in efi\EFI\arch\fw\fwupd-*.cap and then do the update one at a time using the deviceID.

lamarios, 2019-05-15:
what's the command to update ? on my x280 i ran sudo fwupdmgr update deviceId but there still was a n_update: 2 instead of the 4. Is the command correct ?

Mau-G, 2019-05-29:
I'm having the same error with Thinkpad T460.
fwupd version 1.2.8

balejos, 2019-06-05:
Same error here T480s
fwupd version 1.2.9

thatdocslady, 2019-06-20:
Same error here Lenovo X1C6, brand new laptop 
Fedora 30 fresh install, no encryption (I plan to reinstall with encryption after the firmware update)
fwupd version 1.2.9

Ran through all the steps suggested here except for installing the patch branch because I saw it didn't work. Also tried to download the cab file directly from Lenovo and installing only the primary system 
update, same problem. Debug log doesn't show anything extraordinary except the "Unsupported" error.

jaylinski, 2019-07-08:
Seems to be related to #1096. I'm having the same issue on Carbon X1 gen6 with Manjaro installed.

```
/usr/lib/fwupd/fwupdate --log
../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-a4b51dca-8f97-4310-8821-3330f83c9135-0../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-9c9d9769-32fa-4841-b550-ea998e754e99-0../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-ux-capsule../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-3babca5f-b2bf-4f4b-a72e-2bdc84eb4019-0../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)/\EFI\manjaro\fw\fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap...../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and Pci(0x0,0x0)/Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \EFI\manjaro\fw\fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 9417312 bytes../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: A4B51DCA-8F97-4310-8821-3330F83C9135../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: A4B51DCA-8F97-4310-8821-3330F83C9135../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)/\EFI\manjaro\fw\fwupd-9c9d9769-32fa-4841-b550-ea998e754e99.cap...../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and Pci(0x0,0x0)/Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and Path(131,23,0x010000000025388881BA609184012C00)/HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212) and HD(Part1,SigE961DD7E-282C-4408-BCE8-90DA118A7212)../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \EFI\manjaro\fw\fwupd-9c9d9769-32fa-4841-b550-ea998e754e99.cap../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 2198992 bytes../fwupd-1.2.9/plugi [... cut ...] pdate.c:546:fwup_add_update_capsule(): updates guid: 3BABCA5F-B2BF-4F4B-A72E-2BDC84EB4019../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: 3BABCA5F-B2BF-4F4B-A72E-2BDC84EB4019../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:727:efi_main(): n_updates: 4../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:593:fwup_apply_capsules(): Could not query capsule capabilities: Unsupported../fwupd-1.2.9/plugins/uefi/efi/fwupdate.c:742:efi_main(): Could not apply capsules: Unsupported
```

jaylinski, 2019-09-03:
@hughsie Is there anything we can do to help you solve this issue?

hughsie, 2019-09-03:
Unsupported is coming from the old version of the firmware, i.e. fwupd is calling a method to update the firmware and it's returning with an error. Some people have had success resetting the system BIOS, but I think in most cases you have to update the firmware version using some other method (DOS, Windows etc) to get an "old" BIOS new enough to be able to use fwupd.

closed 2019-09-23 by superm1

jaylinski, 2019-09-30:
I fixed the issue by creating a bootable USB stick with the Lenovo BIOS updater:
* Download the ISO file from https://pcsupport.lenovo.com/ni/de/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-6th-gen-type-20kh-20kg/downloads/ds502281
* Convert the ISO to an IMG by downloading  `wget https://userpages.uni-koblenz.de/~krienke/ftp/noarch/geteltorito/geteltorito/geteltorito.pl` and converting the ISO: `chmod +x geteltorito.pl`, `./geteltorito -o thinkpadbios.img n23ur18w.iso`
* Copy the IMG on a USB stick: `sudo fdisk -l` (look for your usb device) and `sudo dd if=thinkpadbios.img of=/dev/sdX bs=1M status=progress oflag=direct`
* Restart, interrupt start with ENTER, enter boot menu with F12 and select `Linux Firmware Updater`
* Select 2 (Update system program)

elovin, 2020-05-15:
just in case other Lenovo T580 users come across this issue, I had the same issue as stated above.

ulno, 2020-07-26:
Yep, still the same here in current Manjaro on Thinkpad X1 Carbon 6th Gen - only thing not updateable via iso-image is the Intel Management engine.

So only chance is to update that (and the connected software borked docking station) is installing Windows 10? Any advice for a small installation to a USB drive that keeps my local linux HD working?

lamarios, 2020-07-27:
There is a way by using WinPE but i didn't get it to work on my machine as the installer was looking for drivers in C: drive but the WinPE booted with D: for the OS drive. I didn't try to dig too much though.

I'm still stuck at the same issue.

ulno, 2020-07-27:
Couldn't get it running with WinPE neither, lots of complaints about wrong versions of executables (yes I used a 64 bit WinPE environment). I got a little further using this here - https://4z2.de/2018/02/24/updating-intel-me-firmware-linux-winpe - but I had different versions of the management engine so it finally had trouble with the executable again. However, I think disabling Intel's root-kit aka IME might be actually more important than updating it.

Even if I could get fwupd working I still couldn't update most external components like the docking station or some weird usb-hardware requiring firmware updates once in a while. I assume the only halfway painless way to update drivers on modern Lenovo laptops is a full Windows 10 installation. So, I assume somehow you should prepare an external drive with a windows 10 installation, make sure it doesn't totally mess up your boot menu and uefi partition and then connect that drive if you need to update something - what a mess! When I finally get to it, I will document that process and reference it somewhere around here.