fwupd/fwupd #1191: Upgrading to ubuntu 19.04 killed my BIOS - trying to restore it - fwupdmgr "no support devices found."

Moldy01, 2019-05-10:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.5
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
P52s Thunderbolt Controller
  DeviceId:             35d684b3fd989b26d276b3acd67bc5299881301c
  Guid:                 eb8bf568-00a2-51e9-aa09-894358506c50
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  Icon:                 computer
  Created:              2019-05-10

Unifying Receiver
  DeviceId:             b8f11848dfcbf441df375b9f11b0e42d085f688b
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.07_B0030
  VersionBootloader:    BOT03.02_B0009
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-05-10

SAMSUNG MZVLB1T0HALR-000L7
  DeviceId:             be29aadda98b637d33b6b740ff7b367f61886a31
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Guid:                 08575ca8-c22c-540a-8ac9-e784c04f6e5a
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              3L2QEXA7
  Icon:                 drive-harddisk
  Created:              2019-05-10
```

```shell
$ efibootmgr -v
EFI variables are not supported on this system.
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
locales-launch: Data of en_US locale not found, generating, please wait...
/boot [error opening dir]

0 directories, 0 files
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version:
Ubuntu 19.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
sudo apt install fwupdmgr
- Have you tried rebooting?
several times
- Are you using an NVMe disk?
yes
- Is secure boot enabled (only for the UEFI plugin)?
Not sure - BIOS won't start anymore

superm1, 2019-05-10:
Sounds like #1058 maybe

Moldy01, 2019-05-10:
No, I lost the BIOS loader during an upgrade from Ubuntu 18.10 to 19.04.  This is a ThinkPad P52S, and at this point, when I start up the laptop, I get the "LENOVO" screen with the "...press Enter" but upon pressing that, the choices come up. but every thing I chose from there hangs and NEVER comes back.  I'm only looking to fix up BIOS so that it is useful again.

closed 2019-09-30 by stale[bot]