fwupd/fwupd #1192: 'Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware version 1.12' offered over and over

LinuxOnTheDesktop, 2019-05-12:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             16324df8d782cbc6d3396541199c8fc397182242
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.06_B0030
  VersionBootloader:    BOT03.01_B0008
  Icon:                 preferences-desktop-keyboard
  Created:              2019-05-12

UEFI Device Firmware
  DeviceId:             29749712e93b58730ecd1b8537a5503ff5b7fb27
  Guid:                 9c9d9769-32fa-4841-b550-ea998e754e99
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.60.3561
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-05-12

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Icon:                 audio-card
  Created:              2019-05-12

20KHCTO1WW System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.37
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2019-05-12

X1 Carbon Thunderbolt Controller
  DeviceId:             74a7c8a4eaeb4ebe00e809c0eab60b420a5e452c
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              30.00
  Icon:                 computer
  Created:              2019-05-12

```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0000,0010,0011,0012,0013,0014,0018,0019,001A,001B,001C,001D,001E,001F,0024
Boot0000* ubuntu	HD(1,GPT,d3d8b2bf-16e6-470b-8c07-d628691e14de,0x800,0x106000)/File(\EFI\ubuntu\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0018* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0019* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001A* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)

```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Mint 19.1 Cinnamon
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

The manager reports:

```
fwupdmgr get-updates
cannot handle firmware requirement not-child
No upgrades for UEFI Device Firmware, current is 184.60.3561: 184.55.3510=older, 184.60.3561=same
UEFI Device Firmware has firmware updates:
GUID:                    3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
ID:                      com.lenovo.ThinkPadN23HT.firmware
Update Version:          0.1.12
Update Name:             ThinkPad X1 Carbon 6th Embedded Controller Update
Update Summary:          Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(cc61721aba7ebe2f980eff4a5e3ce13bc5d6c0fa)
Update Location:         https://fwupd.org/downloads/bdd806fb6165c856404494b59d879fa9d99e1340-Lenovo-ThinkPad-X1Carbon6th-ECFirmware-1.12.cab
Update Description:      Lenovo ThinkPad X1 Carbon 6th Embedded Controller Firmware version 1.12
                         
                         The computer will be restarted automatically after updating EC completely.
                         
                         Do NOT turn off your computer or remove the AC adaptor while update is in progress.
                         
                         This stable release fixes the following issues:
                          • Fix wrong wattage report issue when plug power adapter.
No upgrades for 20KHCTO1WW System Firmware, current is 0.1.37: 0.1.27=older, 0.1.28=older, 0.1.30=older, 0.1.31=older, 0.1.34=older, 0.1.36=older, 0.1.37=same
```
But after `sudo fwupdmg update` and reboot, the same update is offered again. And again (even after attempting to reinstall the update). I see no error messages.
(As to the 'cannot handle firmware requirement not-child', there is [nothing I can do about it](https://github.com/hughsie/fwupd/issues/1139). For, my distro has updated your program, snap does not work, flatpack does not work, and I can't compile from souce.)

hughsie, 2019-05-12:
Does the update actually install? i.e. You see a Lenovo logo with a red progress bar?

LinuxOnTheDesktop, 2019-05-12:
I did not see those things (not hear the trademark awful beeping, though perhaps that latter is reserved for BIOS updates).

LinuxOnTheDesktop, 2019-05-12:
I attach 
[t.txt](https://github.com/hughsie/fwupd/files/3170557/t.txt)
some further information - from my system log.

hughsie, 2019-05-14:
Can you please do:

    sudo /usr/libexec/fwupd/fwupdate --set-debug
    fwupdmgr update
    <reboot>
    sudo /usr/libexec/fwupd/fwupdate --log

then attach the log to this issue please.

LinuxOnTheDesktop, 2019-05-14:
It turns out I did not have `fwupdate` installed. I imagine I had removed it when trying to install the flatpak or snap version of your software. It seems to me that if 'fwupdmgr' - or 'fwupd' - needs 'fwupdate' then it should check for it.

After I had installed `fwupdate` - and it installs on my system not to the path you gave but to `/usr/bin/fwupdate` - the update was still not taking.

So:

```
$  sudo /usr/bin/fwupdate --set-debug
Enabled fwupdate debugging
$ fwupdmgr update
 $  fwupdmgr update
cannot handle firmware requirement not-child
No upgrades for UEFI Device Firmware, current is 184.60.3561: 184.55.3510=older, 184.60.3561=same
Downloading 0.1.12 for UEFI Device Firmware...
Updating 0.1.12 on UEFI Device Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
No upgrades for 20KHCTO1WW System Firmware, current is 0.1.37: 0.1.27=older, 0.1.28=older, 0.1.30=older, 0.1.31=older, 0.1.34=older, 0.1.36=older, 0.1.37=same

An update requires a reboot to complete. Restart now? [Y|n]: 

```
I will report back after the reboot.

superm1, 2019-05-14:
Actually that is the older version of `fwupdate` before it was merged into `fwupd`.  The path Richard indicated is what is used on non Debian/Ubuntu distros.  On Debian/Ubuntu `libexecdir` isn't used and it's instead in `/usr/lib/fwupd/fwupdate`.  So exact same commands above just a different path.

Fortunately the debugging variables used by `/usr/bin/fwupdate` and `/usr/lib/fwupd/fwupdate` are the same.

LinuxOnTheDesktop, 2019-05-14:
This is all very confusing.

```
$  sudo /usr/bin/fwupdate --log
No debug log found
$  sudo /usr/lib/fwupd/fwupdate --log
sudo: /usr/lib/fwupd/fwupdate: command not found
```

superm1, 2019-05-14:
< sigh >.  It was added in 1.1.0 (57df956d7b9d99d788eefb5f18f7b4e9c84ffcdf), you're on too old of a version for the built-in tool.

LinuxOnTheDesktop, 2019-05-14:
And as I indicated above . . [I cannot install any newer version](https://github.com/hughsie/fwupd/issues/1139).

This is thoroughly unsatisfactory. For, there is reason to think that the following is the case. All Mint users, and probably users of e.g. Debian, have an old version of your tool, that contains bugs, and those users cannot upgrade until their distos hit the next update cycle (or deign to add the package). Perhaps you should try to fix the flatpaks and snaps, presuming that the problem with those does indeed lie at your end.

hughsie, 2019-05-14:
I think you need to complain to your distro, not us.

LinuxOnTheDesktop, 2019-05-14:
Thanks, but (1) I did thar some weeks ago. I have yet to have a response. Also, (2) I am unsure that the respective problem with the flatpak and the snap are the distro (Mint's) fault.

superm1, 2019-05-15:
Given there are reports of both snap and flatpak working properly across Ubuntu, Fedora and others I'm inclined to believe it's a mint problem or at least an SELinux problem.  Ubuntu doesn't use SELinux, they use Apparmor typically.  You can try to turn off SELinux to see if that helps maybe.

LinuxOnTheDesktop, 2019-05-15:
Well, thank you. I managed to get the flatpak installed (without messing with Apparmor or SELinux). But I see the following. I report this to you, as well as to Mint, in case it is a problem with your software.

```
$  sudo flatpak run org.freedesktop.fwupd/x86_64/stable get-devices
flatpak run: Can't find a11y bus: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.a11y.Bus': timed out (service_start_timeout=120000ms)
Loading…                 [-                                      ]01:02:48:0865 FuPluginUefi         Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf
Loading… [. . .]
```

LinuxOnTheDesktop, 2019-05-18:
I am closing this because I managed to get the snap version installed. However, now I have yet another problem: #1195

closed 2019-05-18 by LinuxOnTheDesktop