fwupd/fwupd #1193: New warning with GCC 9.1.0

marxin, 2019-05-15:
I see following warnings:

```
[   83s] FAILED: plugins/uefi/efi/fwupdate.o 
[   83s] gcc -c ../plugins/uefi/efi/fwupdate.c -o plugins/uefi/efi/fwupdate.o -Og -g3 --param=ssp-buffer-size=4 -fexceptions -Wall -Werror -Wextra -std=gnu11 -fpic -fshort-wchar -ffreestanding -fno-strict-aliasing -fno-stack-protector -fno-stack-check -fno-merge-constants -Wsign-compare -Wno-missing-field-initializers -grecord-gcc-switches '-DDEBUGDIR="/usr/lib/debug"' -isystem /usr/include/efi -isystem /usr/include/efi/ia32
[   83s] In file included from /usr/include/efi/efi.h:41,
[   83s]                  from ../plugins/uefi/efi/fwupdate.c:7:
[   83s] ../plugins/uefi/efi/fwupdate.c: In function ‘find_updates’:
[   83s] ../plugins/uefi/efi/fwupdate.c:523:8: error: taking address of packed member of ‘struct update_info_s’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   83s]   523 |        &update->info->time_attempted,
[   83s]       |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[   83s] ../plugins/uefi/efi/fwupdate.c: In function ‘add_capsule’:
[   83s] ../plugins/uefi/efi/fwupdate.c:1050:40: error: taking address of packed member of ‘struct update_info_s’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   83s]  1050 |  if (!cap_out->Flags && !is_ux_capsule(&update->info->guid)) {
[   83s]       |                                        ^~~~~~~~~~~~~~~~~~~
[   83s] ../plugins/uefi/efi/fwupdate.c:1017:39: note: in definition of macro ‘is_ux_capsule’
[   83s]  1017 | #define is_ux_capsule(guid) (guid_cmp(guid, &ux_capsule_guid) == 0)
[   83s]       |                                       ^~~~
[   83s] ../plugins/uefi/efi/fwupdate.c:1060:20: error: taking address of packed member of ‘struct update_info_s’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   83s]  1060 |  if (is_ux_capsule(&update->info->guid)) {
[   83s]       |                    ^~~~~~~~~~~~~~~~~~~
[   83s] ../plugins/uefi/efi/fwupdate.c:1017:39: note: in definition of macro ‘is_ux_capsule’
[   83s]  1017 | #define is_ux_capsule(guid) (guid_cmp(guid, &ux_capsule_guid) == 0)
[   83s]       |                                       ^~~~
[   83s] ../plugins/uefi/efi/fwupdate.c: In function ‘efi_main’:
[   83s] ../plugins/uefi/efi/fwupdate.c:1237:22: error: taking address of packed member of ‘struct update_info_s’ may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   83s]  1237 |        is_ux_capsule(&updates[i]->info->guid))
[   83s]       |                      ^~~~~~~~~~~~~~~~~~~~~~~
[   83s] ../plugins/uefi/efi/fwupdate.c:1017:39: note: in definition of macro ‘is_ux_capsule’
[   83s]  1017 | #define is_ux_capsule(guid) (guid_cmp(guid, &ux_capsule_guid) == 0)
[   83s]       |                                       ^~~~
[   83s] cc1: all warnings being treated as errors
```

It's similar to https://github.com/rhboot/efivar/issues/123

hughsie, 2019-05-15:
Isn't this fixed by https://github.com/hughsie/fwupd/commit/1e07ec95b678b1c29ffe798325019c288e6b1efe - i.e. no more warning with git master?

marxin, 2019-05-15:
Yes, it's a workaround. I would consider fixing the warning.

hughsie, 2019-05-15:
> I would consider fixing the warning.

Why? What architectures would be broken with unaligned access?

marxin, 2019-05-15:
Because a compiler makes assumptions based on pointer alignment that can eventually lead to a miscompilation.

hughsie, 2019-09-23:
I don't think UEFI runs on anything that can't handle an unaligned access.

closed 2019-09-23 by hughsie