fwupd/fwupd #1195: Firmware - Intel ME, I think - not updatable, running fwupd as a snap

LinuxOnTheDesktop, 2019-05-18:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
$  fwupdmgr --version
client version:	1.2.8
checkout info:	1.2.8
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.8

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
X1 Carbon Thunderbolt Controller
  DeviceId:             eded407096031a81370bc25ae01650697a7c72b6
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              30.00
  Icon:                 computer
  Created:              2019-05-18

20KHCTO1WW System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Plugin:               uefi
  Flags:                internal|require-ac|supported|registered|needs-reboot
  Version:              0.1.37
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2019-05-18
  UpdateError:          Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf

UEFI Device Firmware
  DeviceId:             29749712e93b58730ecd1b8537a5503ff5b7fb27
  Guid:                 9c9d9769-32fa-4841-b550-ea998e754e99
  Plugin:               uefi
  Flags:                internal|require-ac|supported|registered|needs-reboot
  Version:              184.60.3561
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-05-18
  UpdateError:          Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Plugin:               uefi
  Flags:                internal|require-ac|supported|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Icon:                 audio-card
  Created:              2019-05-18
  UpdateError:          Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf

LENSE20512GMSP34MEAT2TA
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
  Guid:                 0696debf-90a2-5c62-8c96-10857c206b91
  Guid:                 ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  VendorId:             NVME:0x17AA
  Version:              2.7.8341
  Icon:                 drive-harddisk
  Created:              2019-05-18

Unifying Receiver
  DeviceId:             fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.06_B0030
  VersionBootloader:    BOT03.01_B0008
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-05-18

```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0000,0010,0011,0012,0013,0014,0018,0019,001A,001B,001C,001D,001E,001F,0024,0001
Boot0000* ubuntu	HD(1,GPT,d3d8b2bf-16e6-470b-8c07-d628691e14de,0x800,0x106000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux-Firmware-Updater \fwupx64.efi	HD(4,GPT,a70ec6da-8a1d-4a2b-9517-a67b28a74a3d,0x106800,0x5000000)/File(boot\efi\EFI\ubuntu\fwupx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0018* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0019* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001A* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-a4b51dca-8f97-4310-8821-3330f83c9135-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-3babca5f-b2bf-4f4b-a72e-2bdc84eb4019-0

```

Please answer the following questions:

- Operating system and version: Mint 19.1 x64 Cinnamon
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No (and I can't enable it without reinstalling my OS, on pain of not being able to boot - for, my BIOS got reset at some point)

The problem:

```
$  sudo fwupdmgr get-updates
ignoring 20KHCTO1WW System Firmware [1c53551e7da69d896138fac1ae131c83ad46d923] as not updatable
ignoring UEFI Device Firmware [29749712e93b58730ecd1b8537a5503ff5b7fb27] as not updatable
ignoring UEFI Device Firmware [9698faabddf0d7b18925cfbbda95f8b0d0dacc53] as not updatable
No upgrades for Unifying Receiver, current is RQR24.06_B0030: RQR24.06_B0030=same, RQR24.05_B0029=older
```

hughsie, 2019-05-18:
> Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf

Where is your ESP located?

LinuxOnTheDesktop, 2019-05-18:
I do not know. I do not even know what 'ESP' means here. Enlighten me and I will answer your question. Thanks.

LinuxOnTheDesktop, 2019-05-20:
Ah, [this](https://wiki.archlinux.org/index.php/EFI_system_partition) is what you mean.

Does the following answer your question?

 ```
$  sudo blkid
/dev/loop0: TYPE="squashfs"
/dev/loop1: TYPE="squashfs"
/dev/nvme0n1: PTUUID="a28bce5a-48da-4d36-bec4-45c0ea49bc08" PTTYPE="gpt"
/dev/nvme0n1p1: UUID="87CE-448D" TYPE="vfat" PARTLABEL="uefi" PARTUUID="d3d8b2bf-16e6-470b-8c07-d628691e14de"
/dev/nvme0n1p2: UUID="18b9ad28-d429-4f42-befd-8619f5fd92bc" TYPE="ext4" PARTLABEL="home" PARTUUID="6ae6d55b-18ad-483b-992f-bfb79a7812c9"
/dev/nvme0n1p3: UUID="63c8c38f-b261-4295-8cbd-f8fd16851626" TYPE="swap" PARTLABEL="swap" PARTUUID="7fe9b858-ee99-4ba0-a396-4d0987b10e53"
/dev/nvme0n1p4: UUID="65538fc8-e344-42ac-94f4-2751b5bf7f76" TYPE="ext4" PARTLABEL="root" PARTUUID="a70ec6da-8a1d-4a2b-9517-a67b28a74a3d"
```

superm1, 2019-05-20:
Specifically mount points (output of mount command)

yuzaipiaofei, 2019-05-20:
> Ah, [this](https://wiki.archlinux.org/index.php/EFI_system_partition) is what you mean.
>
> Does the following answer your question?
>
> ```
> $  sudo blkid
> /dev/loop0: TYPE="squashfs"
> /dev/loop1: [...]
this command will report all the fs mount info.
#mount

LinuxOnTheDesktop, 2019-05-20:
```
#  mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=8120660k,nr_inodes=2030165,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=1631072k,mode=755)
/dev/nvme0n1p4 on / type ext4 (rw,relatime,lazytime,errors=remount-ro,stripe=32734)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
efivarfs on /sys/firmware/efi/efivars type efivarfs (rw,nosuid,nodev,noexec,relatime)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,nosuid,nodev,noexec,relatime,rdma)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/hugetlb type cgroup (rw,nosuid,nodev,noexec,relatime,hugetlb)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=27,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=19005)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
mqueue on /dev/mqueue type mqueue (rw,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime,pagesize=2M)
configfs on /sys/kernel/config type configfs (rw,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
/var/lib/snapd/snaps/fwupd_770.snap on /snap/fwupd/770 type squashfs (ro,nodev,relatime,x-gdu.hide)
/var/lib/snapd/snaps/core_6818.snap on /snap/core/6818 type squashfs (ro,nodev,relatime,x-gdu.hide)
tmpfs on /mnt/ramdisk type tmpfs (rw,noatime,size=5120k)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime,size=1631072k)
Firefox on /ramdisk type tmpfs (rw,nosuid,nodev,noatime,size=308224k)
/dev/nvme0n1p2 on /home type ext4 (rw,relatime,lazytime,stripe=32746)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=1631068k,mode=700,uid=1000,gid=1000)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
```

superm1, 2019-05-20:
It appears to me that your EFI system partition isn't mounted.  Without it being mounted fwupd can't know where to stage the update or install the bootloader.

LinuxOnTheDesktop, 2019-05-20:
OK; so should that system be partition always be mounted? If not, must I mount it whenever I use the fwupd program? Either way, should I mount the partition in any particular way?

superm1, 2019-05-20:
At least on Debian & Ubuntu it's declared in `/etc/fstab` and automatically mounted on every boot.  If you don't configure your system that way, you'll need to manually perform the action before starting fwupd.  fwupd will not mount it for you as it can't determine the location.

LinuxOnTheDesktop, 2019-05-20:
I don't know whether my Mint installation ever had its fstab mount the efi partition, but at present my - customised - fstab does not do that.

Anyhow, I mounted the efi partition (using the `disks` GUI), ran fwupdmgr (via the repo version, because, as I reported, the snap version of the package says the firmware is not updatable), fwupdmgr said it would install the update on reboot and . . it did not. Perhaps I need to mount the partition actually in fstab. I will try that now, which will involve trying to determine what mount options to use. Shouldn't fwupdmgr (or fwupd or whatever part of the program is at issue) *say* that one will need a mount for the efi partition, and/or be setup to notice when a reboot failed to install an update?

superm1, 2019-05-20:
> EDIT: ah, I see my fstab _used_ to contain this:

Well glad to know it was self-inflicted damage at least.

> Shouldn't fwupdmgr (or fwupd or whatever part of the program is at issue) say that one will need a mount for the efi partition, and/or be setup to notice when a reboot failed to install an update?

It did say that it couldn't find the ESP, but it's not possible to know if it's a lack of mount, missing ESP, etc.

```
Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf
```

> I reported, the snap version of the package says the firmware is not updatable), fwupdmgr said it would install the update on reboot and . . it did not

After you've fixed your `/etc/fstab` and mounted the partition, you'll need to either restart `snap.fwupd.fwupd` service or reboot the system for fwupd to see the proper ESP location.  If it keeps not working, it's another issue.  Maybe https://github.com/hughsie/fwupd/issues/1190?

LinuxOnTheDesktop, 2019-05-20:
&nbsp;&nbsp;&nbsp;&nbsp;_Success_: once there was a fstab entry (for details see above) for my efi partition, the apt (i.e. repository) version of fwupd managed - via several reboots - to install an update. Moreover, that same version of fwupd now reports nothing amiss. For it reports as follows.

```
$  fwupdmgr get-updates
No upgrades for 20KHCTO1WW System Firmware, current is 0.1.38: 0.1.27=older, 0.1.28=older, 0.1.30=older, 0.1.31=older, 0.1.34=older, 0.1.36=older, 0.1.37=older, 0.1.38=same
No upgrades for UEFI Device Firmware, current is 184.60.3561: 184.55.3510=older, 184.60.3561=same
No upgrades for UEFI Device Firmware, current is 0.1.12: 0.1.12=same
No upgrades for Unifying Receiver, current is RQR24.06_B0030: RQR24.06_B0030=same, RQR24.05_B0029=older
```

Thanks everyone for your help.

&nbsp;&nbsp;&nbsp;&nbsp;_However_: (1) I repeat my point that I think fwupd should in one way or another check for and/or warn about this mounting problem; (2) it might be good to make clear somewhere what happens when, as is the case with my system now, one has both the apt and the snap versions installed. I see this:

```
$  fwupdmgr --version
client version:	1.0.9
daemon version:	1.2.8
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

This means, I take it that, that any `fwupdmgr` commands I issue are invoking the snap version - yes?

closed 2019-05-20 by LinuxOnTheDesktop

superm1, 2019-05-20:
> Success

Great.

> This means, I take it that, that any fwupdmgr commands I issue are invoking the snap version - yes?

Yes, that means that you are using the client from the repo and the daemon from the snap.
Normally with snap the PATH is set with `/usr/bin/` to have precedence over `/snap/bin`  So if you have both installed `/usr/bin/fwupdmgr` gets used instead `/snap/bin/fwupdmgr`.

You can use the snap version by running either `/snap/bin/fwupdmgr` or `fwupd.fwupdmgr`.

> (EDIT: OK, it does say that - might it be made to say it more clearly, though, i.e. with a mention of fstab?

Well there is no confirmation that the problem is caused by `/etc/fstab` though.  It could be that you decided to mount your ESP at `/fat32` rather than a path that fwupd recognizes (`/boot/` `/efi` or `/boot/efi`) the same error will happen.
Or it could have been that you have two ESP's on the system and only one is mounted read write, the wrong one is being picked.
Or it could have been the wrong filesystem on the ESP.

LinuxOnTheDesktop, 2019-05-20:
&nbsp;&nbsp;&nbsp;&nbsp;On snap: thank you for the information.

&nbsp;&nbsp;&nbsp;&nbsp;**ESP**: could `fwupd` not at least
1. recognise and report that it had tried and failed to install an update
and
2. say that the problem involves (in one way or another) a failure to find the ESP a.k.a. UEFI partition?

Unless I miss something, it seem to me that at present 1 does not happen and that 2 does not quite happen. Let me expand on the latter.

&nbsp;&nbsp;&nbsp;&nbsp;2 does not quite happen in that the error message is somewhat cryptic. For, at present the message given is:

> 'Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf.

 I suggest that the first clause be changed to something like this:

> Unable to determine the location of the EFI (a.k.a. UEFI) partition

As to the second clause, viz., 'override using OverrideESPMountPoint in uefi.conf', it seems ambiguous between (i) telling the user how she can perform an override, (ii) telling the user that the program will, without further intervention, attempt an override.

superm1, 2019-05-20:
It's a difficult balance to strike with an error message and telling someone how to fix it.
@hughsie what do you think of changing the error to:

"Unable to determine EFI system partition location, see $WIKI for details how to fix"

And then we can go into detail about the causes and debugging for people there.

hughsie, 2019-05-20:
> what do you think of changing the error to

I'm a huge proponent in making this kind of thing easier to debug, +1 from me.