fwupd/fwupd #1196: failed to execute prepared statement: no such table: history

mnahkies, 2019-05-18:
```shell
$ fwupdmgr --version
client version:	1.2.8
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.8
```

```shell
$ fwupdmgr get-devices
XPS 9560 Thunderbolt Controller
  DeviceId:             86c5af4c332ad19b9ee0b580c198b99c19beda15
  Guid:                 72533768-6a6c-5c06-994a-367374336810 <- TBT-00d407be
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              21.00
  Icon:                 computer
  Created:              2019-05-18

XPS 15 9560 System Firmware
  DeviceId:             f49999b6e3af5b4bf8ad1f2ed2922b76df457271
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(9696d8058d9554550d6443e05be0bb7fdc463ee7)
  Checksum:             SHA256(5a093bdbf557a92b7c9d4564c6c879b9d79e9871e067eef933c78a78ea02d0ab)
  Version:              0.1.5.0
  VersionLowest:        0.1.5.0
  Icon:                 computer
  Created:              2019-05-18
  Modified:             2019-05-18
  UpdateState:          needs-reboot

XPS 15 9560 TPM 2.0
  DeviceId:             cc2b222929056ab3e9a208df9f86b7d9a74ae514
  Guid:                 bc891321-25aa-54f6-86e7-633d065b8d5d
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|require-ac|supported|registered
  Vendor:               Dell Inc.
  Version:              1.3.2.8
  Icon:                 computer
  Created:              2019-05-18

KXG50ZNV512G NVMe TOSHIBA 512GB
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 4d0aed03-a30c-52c6-99e7-a8977797c3d9 <- NVME\VEN_1179&DEV_0116
  Guid:                 ad9fe8f7-cdc4-52c9-9fea-31b6f4988ffa <- NVME\VEN_1179
  Guid:                 98d34383-a939-5b5a-9806-e05e0b288b1a <- KXG50ZNV512G NVMe TOSHIBA 512GB
  Serial:               Y78F204MFQCS
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              AADA4102
  Icon:                 drive-harddisk
  Created:              2019-05-18

Failed to prepare SQL to get history: no such table: history
```

```shell
$ efibootmgr -v
BootNext: 0003
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0002,0003
Boot0000* Windows Boot Manager	HD(1,GPT,d33ba84a-2f1b-4d58-8b0e-16f76203f7be,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Fedora	HD(1,GPT,d33ba84a-2f1b-4d58-8b0e-16f76203f7be,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)
Boot0002* UEFI: KXG50ZNV512G NVMe TOSHIBA 512GB, Partition 1	HD(1,GPT,d33ba84a-2f1b-4d58-8b0e-16f76203f7be,0x800,0x64000)/File(EFI\boot\bootx64.efi)..BO
Boot0003* Linux Firmware Updater	HD(1,GPT,d33ba84a-2f1b-4d58-8b0e-16f76203f7be,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-34578c72-11dc-4378-bc7f-b643866f598c-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ux-capsule
```

```shell
$ tree /boot
/boot
├── 520f44320ac640c68996d2e055036d31
│   └── 4.13.9-300.fc27.x86_64
├── config-5.0.13-100.fc28.x86_64
├── config-5.0.16-100.fc28.x86_64
├── config-5.0.16-300.fc30.x86_64
├── efi
│   ├── 992290f4455d442b88f74b3b76c2ca3c
│   │   ├── 4.16.5-200.fc27.x86_64
│   │   ├── 4.16.6-202.fc27.x86_64
│   │   └── 4.16.7-200.fc27.x86_64
│   ├── EFI
│   │   ├── BOOT
│   │   │   ├── BOOTIA32.EFI
│   │   │   ├── BOOTX64.EFI
│   │   │   ├── fbia32.efi
│   │   │   └── fbx64.efi
│   │   ├── Dell
│   │   │   └── logs
│   │   │       ├── diags_current.xml
│   │   │       └── diags_previous.xml
│   │   └── fedora
│   │       ├── BOOTIA32.CSV
│   │       ├── BOOTX64.CSV
│   │       ├── fonts
│   │       │   └── unicode.pf2
│   │       ├── fw
│   │       │   ├── fwupd-34578c72-11dc-4378-bc7f-b643866f598c.cap
│   │       │   └── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │       ├── fwupdx64.efi
│   │       ├── fwupia32.efi
│   │       ├── fwupx64.efi
│   │       ├── gcdia32.efi
│   │       ├── gcdx64.efi
│   │       ├── grub.cfg
│   │       ├── grubenv
│   │       ├── grubia32.efi
│   │       ├── grubx64.efi
│   │       ├── mmia32.efi
│   │       ├── mmx64.efi
│   │       ├── shim.efi
│   │       ├── shimia32.efi
│   │       ├── shimia32-fedora.efi
│   │       ├── shimx64.efi
│   │       └── shimx64-fedora.efi
│   ├── mach_kernel
│   └── System
│       └── Library
│           └── CoreServices
│               └── SystemVersion.plist
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2
│   ├── grubenv -> ../efi/EFI/fedora/grubenv
│   └── themes
│       └── system
│           ├── background.png
│           └── fireworks.png
├── initramfs-0-rescue-992290f4455d442b88f74b3b76c2ca3c.img
├── initramfs-5.0.13-100.fc28.x86_64.img
├── initramfs-5.0.16-100.fc28.x86_64.img
├── initramfs-5.0.16-300.fc30.x86_64.img
├── loader
│   └── entries
│       ├── 992290f4455d442b88f74b3b76c2ca3c-0-rescue.conf
│       ├── 992290f4455d442b88f74b3b76c2ca3c-5.0.13-100.fc28.x86_64.conf
│       ├── 992290f4455d442b88f74b3b76c2ca3c-5.0.16-100.fc28.x86_64.conf
│       └── 992290f4455d442b88f74b3b76c2ca3c-5.0.16-300.fc30.x86_64.conf
├── lost+found
├── memtest86+-5.01
├── System.map-5.0.13-100.fc28.x86_64
├── System.map-5.0.16-100.fc28.x86_64
├── System.map-5.0.16-300.fc30.x86_64
├── vmlinuz-0-rescue-992290f4455d442b88f74b3b76c2ca3c
├── vmlinuz-5.0.13-100.fc28.x86_64
├── vmlinuz-5.0.16-100.fc28.x86_64
└── vmlinuz-5.0.16-300.fc30.x86_64
```

Please answer the following questions:

- Operating system and version: Fedora 30
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
- Have you tried rebooting? Not yet
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

Attempting to upgrade firmware using `--force` to work around the version formats were different message (#1156) resulted in:

```
# fwupdmgr update
No upgrades for XPS 9560 Thunderbolt Controller, current is 21.00: 21.00=same
Downloading 0.1.12.1 for XPS 15 9560 System Firmware...
Fetching firmware https://fwupd.org/downloads/0476f7fb507f407f5e1664476e6eff28a0529728-firmware_XPS9560_Precision5520_1.12.1.cab
Downloading…             [***************************************]
Decompressing…           [***************************************]
Firmware version formats were different, device was 'quad' and release is 'plain'

# fwupdmgr update --force 
No upgrades for XPS 9560 Thunderbolt Controller, current is 21.00: 21.00=same
Downloading 0.1.12.1 for XPS 15 9560 System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating XPS 15 9560 System Firmware…                            ]
Scheduling…              [***************************************]
failed to execute prepared statement: no such table: history
```

mnahkies, 2019-05-18:
After rebooting the updates were successfully applied. The message "Failed to prepare SQL to get history: no such table: history" is no longer present when running `# fwupdmgr get-devices`.

Interestingly though, running get-history returns nothing:
```
# fwupdmgr get-history
No history
```

This machine started life as Fedora 27, then was on 28 until today when I upgraded to Fedora 30.

MosheSheena, 2019-06-30:
I am also seeing this.
Fedora 29 (5.1.11-200.fc29.x86_64).
➜  ~ fwupdmgr update     
Downloading 0.1.31 for 20L8S2N80H System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Failed to prepare SQL to delete history: no such table: history

hughsie, 2019-07-01:
@MosheSheena was your machine upgraded from a previous Fedora version too?

jam01, 2019-08-02:
Seeing the same SQL table error on a ThinkPad T480 running Fedora 30. I've been upgrading the same install since F26.

```
$ fwupdmgr --version
client version:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9
```

```
$ fwupdmgr update
Downloading 0.1.25 for 20L5CTO1WW System Firmware...
Fetching firmware https://fwupd.org/downloads/23fa19dbeda090e0ba3976783d9518b5dc0e07ed-Lenovo-ThinkPad-T480-SystemFirmware-1.25.cab
Downloading…             [***************************************] Less than one minute remaining…
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Failed to prepare SQL to delete history: no such table: history
```

Operating system and version: Fedora 30
How did you install fwupd (ex: from source, pacman, apt-get, etc): came with F29 (I believe)
Have you tried rebooting? yes
Are you using an NVMe disk? no
Is secure boot enabled (only for the UEFI plugin)? no

scr4bble, 2019-08-04:
Same issue here.
```
Downloading 0.1.25 for 20L5000BMC System Firmware...
Fetching firmware https://fwupd.org/downloads/23fa19dbeda090e0ba3976783d9518b5dc0e07ed-Lenovo-ThinkPad-T480-SystemFirmware-1.25.cab
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Failed to prepare SQL to delete history: no such table: history
```

F29, using NVMe disk, reboot not tried, secore boot disabled

hughsie, 2019-08-06:
> Failed to prepare SQL to delete history: no such table: history

Weird. Can you run `sudo /usr/libexec/fwupd/fwupd --verbose` please, then in another tab do the `fwupdmgr update` then attach the entire log. Thanks!

scr4bble, 2019-08-06:
It has updated the firmware after all (after system reset), but the history is empty, that's probably wrong.

```
$ fwupdmgr get-history
No history
```

When running fwupdmgr update in another tab while running the command you posted, I receive:
```
$ fwupdmgr update
Mismatched daemon and client, use fwupdmgr instead
```

MosheSheena, 2019-08-06:
> @MosheSheena was your machine upgraded from a previous Fedora version too?

@hughsie Yes. From FC28

hughsie, 2019-08-06:
> It has updated the firmware after all (after system reset), but the history is empty, that's probably wrong.

Can you get the verbose trace of startup please, and also attach your /var/lib/fwupd/pending.db file pls.

scr4bble, 2019-08-06:
[fwupd.log](https://github.com/hughsie/fwupd/files/3472600/fwupd.log)
pending.db sent via email to richard@hughsie.com

The log contains this line:
`
failed to migrate /var/lib/fwupd/pending.db database: Failed to prepare SQL for creating tables: table schema already exists
`

hughsie, 2019-08-06:
For my future self, the contents of pending was:

    BEGIN TRANSACTION;
    CREATE TABLE schema (created timestamp TIMESTAMP DEFAULT CURRENT_TIMESTAMP,version INTEGER DEFAULT 0);
    INSERT INTO schema VALUES('2019-08-06 13:53:56',0);
    CREATE TABLE history (device_id TEXT,update_state INTEGER DEFAULT 0,update_error TEXT,filename TEXT,display_name TEXT,plugin TEXT,device_created INTEGER DEFAULT 0,device_modified INTEGER DEFAULT 0,checksum TEXT DEFAULT NULL,flags INTEGER DEFAULT 0,metadata TEXT DEFAULT NULL,guid_default TEXT DEFAULT NULL,version_old TEXT,version_new TEXT,checksum_device TEXT DEFAULT NULL,protocol TEXT DEFAULT NULL);
    CREATE TABLE approved_firmware (checksum TEXT);

hughsie, 2019-08-06:
> Failed to prepare SQL for creating tables: table schema already exists

Okay, I can reproduce with your file, thanks. I'm debugging now.

closed 2019-08-06 by superm1