fwupd/fwupd #1198: Firmware update for Dell Latitude 5480 fails to install

the-spyke, 2019-05-21:
Firmware update for Dell Latitude 5480 fails to install:

```shell
$ fwupdmgr update
No upgrades for Latitude 5480 Thunderbolt Controller, current is 33.00: 23.00=older
Downloading 0.1.14.2 for Latitude 5480 System Firmware...
Decompressing?           [***************************************]
Authenticating?          [***************************************]
Could not check for auth: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: Action org.freedesktop.fwupd.update-internal is not registered
```

```shell
$ fwupdmgr --version
client version:	1.2.9
checkout info:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9
```

```shell
$ fwupdmgr get-devices
Latitude 5480 Thunderbolt Controller
  DeviceId:             5b7c800882d6116ffb07b898aa4dbbed4a819986
  Guid:                 9cdaf37f-584f-5d1f-948b-20fbbbb30231 <- TBT-00d407d0
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              33.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-05-21

Latitude 5480 System Firmware
  DeviceId:             348fa58069868c96025a2416f1a1024667045bdd
  Guid:                 49e03513-7afe-498a-9d03-89e5ba1da865
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.13.0
  VersionLowest:        0.1.13.0
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-05-21

Samsung SSD 960 EVO 1TB
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 5b3df2da-f745-5fd0-81de-5dafd7f0bf8c <- NVME\VEN_144D&DEV_A804
  Guid:                 084d98fd-3d8c-5a4a-a094-6f8925ad62ab <- Samsung SSD 960 EVO 1TB
  Serial:               S3ETNX0J201391T
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  VendorId:             NVME:0x144D
  Version:              3B7QCXE7
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-05-21
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000
Boot0000* Windows Boot Manager	HD(2,GPT,baf5a50e-499e-455d-a013-14e9673dd019,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...,................
Boot0001* ubuntu	HD(2,GPT,baf5a50e-499e-455d-a013-14e9673dd019,0xe1800,0x32000)/File(\EFI\ubuntu\shimx64.efi)
```
[... 289 lines cut ...]
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-5.0.0-13-generic
├── initrd.img-5.0.0-15-generic
├── lost+found [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.0.0-13-generic
├── System.map-5.0.0-15-generic
├── vmlinuz-5.0.0-13-generic
└── vmlinuz-5.0.0-15-generic

6 directories, 288 files
```

Please answer the following questions:

- Operating system and version: Ubuntu 19.04 x64 (5.0.0-15-generic)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

hughsie, 2019-05-21:
> How did you install fwupd: snap

@superm1 -- any ideas?

superm1, 2019-05-21:
Did you install the snap and remove the Deb afterwards?

the-spyke, 2019-05-21:
@superm1 Removed with `apt` first and then installed with `snap`

superm1, 2019-05-21:
Can you try to reboot? I wonder if the PK file from the snap didn't get picked up yet.

the-spyke, 2019-05-21:
@superm1 Yeah, those files where missing. I've reinstalled the snap and it works now. Maybe it was some snappy bug. Thanks!

closed 2019-05-21 by the-spyke

superm1, 2019-05-21:
Well glad to hear that.  I'd like to get to the bottom of that if possible though.
Are you sure you removed the deb w/ apt first?  That behavior would definitely happen if it was the other way around.

the-spyke, 2019-05-21:
@superm1  Ah, I've got it. I was doing this way when I was installing the system, but during upgrade from `18.10` to `19.04` it automatically restored `fwupd` deb package. I just removed it then.

superm1, 2019-05-21:
Ah gotcha, well that makes perfect sense then.
@hughsie what do you think about adding a check when running the snapped `fwupdmgr` that checks that `/usr/share/polkit-1/rules.d/org.freedesktop.fwupd.rules` exists and bailing otherwise to make this clearer in the future?

hughsie, 2019-05-21:
In all honesty, I think i'm at my limit of snap-specific code.

superm1, 2019-05-21:
Well another option is to make it a generic message; when this error shows up:
```
Could not check for auth: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: Action org.freedesktop.fwupd.update-internal is not registered
```
Detect it and say as something like "Policykit rules are missing, please reinstall the package"

hughsie, 2019-05-21:
That works, thanks.