fwupd/fwupd #1206: Error calling StartServiceByName for org.freedesktop.fwupd

faliagas, 2019-05-23:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```
Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 1 seconds
BootOrder: 0000
Boot0000* ubuntu	HD(1,GPT,18ccda63-f99a-49e2-8dd5-3b41fa4f3771,0x800,0x100000)/File(\EFI\UBUNTU\SHIMX64.EFI)
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw

Command 'efivar' not found, but can be installed with:

sudo apt install efivar

a**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
/boot
├── config-5.0.0-13-generic
├── config-5.0.0-15-generic
├── efi [error opening dir]
├── grub
│   ├── fonts
│   │   └── unicode.pf2
│   ├── grub.cfg
│   ├── grubenv
│   ├── locale
│   │   ├── en_AU.mo
│   │   ├── en_CA.mo
│   │   ├── en_GB.mo
│   │   └── en@quot.mo
│   ├── unicode.pf2
│   └── x86_64-efi
│       ├── acpi.mod
│       ├── adler32.mod
│       ├── affs.mod
│       ├── afs.mod
│       ├── ahci.mod
│       ├── all_video.mod
│       ├── aout.mod
│       ├── appleldr.mod
│       ├── archelp.mod
│       ├── ata.mod
│       ├── at_keyboard.mod
│       ├── backtrace.mod
│       ├── bfs.mod
│       ├── bitmap.mod
│       ├── bitmap_scale.mod
│       ├── blocklist.mod
│       ├── boot.mod
│       ├── bsd.mod
│       ├── bswap_test.mod
│       ├── btrfs.mod
│       ├── bufio.mod
│       ├── cat.mod
│       ├── cbfs.mod
│       ├── cbls.mod
│       ├── cbmemc.mod
│       ├── cbtable.mod
│       ├── cbtime.mod
│       ├── chain.mod
│       ├── cmdline_cat_test.mod
│       ├── cmp.mod
│       ├── cmp_test.mod
│       ├── command.lst
│       ├── configfile.mod
│       ├── core.efi
│       ├── cpio_be.mod
│       ├── cpio.mod
│       ├── cpuid.mod
│       ├── crc64.mod
│       ├── cryptodisk.mod
│       ├── crypto.lst
│       ├── crypto.mod
│       ├── cs5536.mod
│       ├── ctz_test.mod
│       ├── datehook.mod
│       ├── date.mod
│       ├── datetime.mod
│       ├── diskfilter.mod
│       ├── disk.mod
│       ├── div.mod
│       ├── div_test.mod
│       ├── dm_nv.mod
│       ├── echo.mod
│       ├── efifwsetup.mod
│       ├── efi_gop.mod
│       ├── efinet.mod
│       ├── efi_uga.mod
│       ├── ehci.mod
│       ├── elf.mod
│       ├── eval.mod
│       ├── exfat.mod
│       ├── exfctest.mod
│       ├── ext2.mod
│       ├── extcmd.mod
│       ├── fat.mod
│       ├── file.mod
│       ├── fixvideo.mod
│       ├── font.mod
│       ├── fshelp.mod
│       ├── fs.lst
│       ├── functional_test.mod
│       ├── gcry_arcfour.mod
│       ├── gcry_blowfish.mod
│       ├── gcry_camellia.mod
│       ├── gcry_cast5.mod
│       ├── gcry_crc.mod
│       ├── gcry_des.mod
│       ├── gcry_dsa.mod
│       ├── gcry_idea.mod
│       ├── gcry_md4.mod
│       ├── gcry_md5.mod
│       ├── gcry_rfc2268.mod
│       ├── gcry_rijndael.mod
│       ├── gcry_rmd160.mod
│       ├── gcry_rsa.mod
│       ├── gcry_seed.mod
│       ├── gcry_serpent.mod
│       ├── gcry_sha1.mod
│       ├── gcry_sha256.mod
│       ├── gcry_sha512.mod
│       ├── gcry_tiger.mod
│       ├── gcry_twofish.mod
│       ├── gcry_whirlpool.mod
│       ├── geli.mod
│       ├── gettext.mod
│       ├── gfxmenu.mod
│       ├── gfxterm_background.mod
│       ├── gfxterm_menu.mod
│       ├── gfxterm.mod
│       ├── gptsync.mod
│       ├── grub.efi
│       ├── gzio.mod
│       ├── halt.mod
│       ├── hashsum.mod
│       ├── hdparm.mod
│       ├── hello.mod
│       ├── help.mod
│       ├── hexdump.mod
│       ├── hfs.mod
│       ├── hfspluscomp.mod
│       ├── hfsplus.mod
│       ├── http.mod
│       ├── iorw.mod
│       ├── iso9660.mod
│       ├── jfs.mod
│       ├── jpeg.mod
│       ├── keylayouts.mod
│       ├── keystatus.mod
│       ├── ldm.mod
│       ├── legacycfg.mod
│       ├── legacy_password_test.mod
│       ├── linux16.mod
│       ├── linuxefi.mod
│       ├── linux.mod
│       ├── loadbios.mod
│       ├── load.cfg
│       ├── loadenv.mod
│       ├── loopback.mod
│       ├── lsacpi.mod
│       ├── lsefimmap.mod
│       ├── lsefi.mod
│       ├── lsefisystab.mod
│       ├── lsmmap.mod
│       ├── ls.mod
│       ├── lspci.mod
│       ├── lssal.mod
│       ├── luks.mod
│       ├── lvm.mod
│       ├── lzopio.mod
│       ├── macbless.mod
│       ├── macho.mod
│       ├── mdraid09_be.mod
│       ├── mdraid09.mod
│       ├── mdraid1x.mod
│       ├── memdisk.mod
│       ├── memrw.mod
│       ├── minicmd.mod
│       ├── minix2_be.mod
│       ├── minix2.mod
│       ├── minix3_be.mod
│       ├── minix3.mod
│       ├── minix_be.mod
│       ├── minix.mod
│       ├── mmap.mod
│       ├── moddep.lst
│       ├── modinfo.sh
│       ├── morse.mod
│       ├── mpi.mod
│       ├── msdospart.mod
│       ├── mul_test.mod
│       ├── multiboot2.mod
│       ├── multiboot.mod
│       ├── nativedisk.mod
│       ├── net.mod
│       ├── newc.mod
│       ├── nilfs2.mod
│       ├── normal.mod
│       ├── ntfscomp.mod
│       ├── ntfs.mod
│       ├── odc.mod
│       ├── offsetio.mod
│       ├── ohci.mod
│       ├── part_acorn.mod
│       ├── part_amiga.mod
│       ├── part_apple.mod
│       ├── part_bsd.mod
│       ├── part_dfly.mod
│       ├── part_dvh.mod
│       ├── part_gpt.mod
│       ├── partmap.lst
│       ├── part_msdos.mod
│       ├── part_plan.mod
│       ├── part_sun.mod
│       ├── part_sunpc.mod
│       ├── parttool.lst
│       ├── parttool.mod
│       ├── password.mod
│       ├── password_pbkdf2.mod
│       ├── pata.mod
│       ├── pbkdf2.mod
│       ├── pbkdf2_test.mod
│       ├── pcidump.mod
│       ├── play.mod
│       ├── png.mod
│       ├── priority_queue.mod
│       ├── probe.mod
│       ├── procfs.mod
│       ├── progress.mod
│       ├── raid5rec.mod
│       ├── raid6rec.mod
│       ├── random.mod
│       ├── read.mod
│       ├── reboot.mod
│       ├── regexp.mod
│       ├── reiserfs.mod
│       ├── relocator.mod
│       ├── romfs.mod
│       ├── scsi.mod
│       ├── search_fs_file.mod
│       ├── search_fs_uuid.mod
│       ├── search_label.mod
│       ├── search.mod
│       ├── serial.mod
│       ├── setjmp.mod
│       ├── setjmp_test.mod
│       ├── setpci.mod
│       ├── sfs.mod
│       ├── shift_test.mod
│       ├── signature_test.mod
│       ├── sleep.mod
│       ├── sleep_test.mod
│       ├── spkmodem.mod
│       ├── squash4.mod
│       ├── syslinuxcfg.mod
│       ├── tar.mod
│       ├── terminal.lst
│       ├── terminal.mod
│       ├── terminfo.mod
│       ├── test_blockarg.mod
│       ├── testload.mod
│       ├── test.mod
│       ├── testspeed.mod
│       ├── tftp.mod
│       ├── tga.mod
│       ├── time.mod
│       ├── trig.mod
│       ├── tr.mod
│       ├── true.mod
│       ├── udf.mod
│       ├── ufs1_be.mod
│       ├── ufs1.mod
│       ├── ufs2.mod
│       ├── uhci.mod
│       ├── usb_keyboard.mod
│       ├── usb.mod
│       ├── usbms.mod
│       ├── usbserial_common.mod
│       ├── usbserial_ftdi.mod
│       ├── usbserial_pl2303.mod
│       ├── usbserial_usbdebug.mod
│       ├── usbtest.mod
│       ├── verify.mod
│       ├── video_bochs.mod
│       ├── video_cirrus.mod
│       ├── video_colors.mod
│       ├── video_fb.mod
│       ├── videoinfo.mod
│       ├── video.lst
│       ├── video.mod
│       ├── videotest_checksum.mod
│       ├── videotest.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-5.0.0-13-generic
├── initrd.img-5.0.0-15-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.0.0-13-generic
├── System.map-5.0.0-15-generic
├── vmlinuz-5.0.0-13-generic
└── vmlinuz-5.0.0-15-generic

5 directories, 287 files
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Ubuntu 19.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): it came with OS
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

hughsie, 2019-05-23:
What does `sudo /usr/lib/fwupd/fwupd --verbose` say?

faliagas, 2019-05-24:
Here is the output of `sudo /usr/lib/fwupd/fwupd --verbose`
04:49:14:0449 FuDebug              Verbose debugging enabled (on console 1)
04:49:14:0449 FuConfig             loading config values from /etc/fwupd/daemon.conf
04:49:14:0450 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
04:49:14:0450 Fwupd                looking for os-release at /etc/os-release
04:49:14:0450 FuConfig             loading /usr/share/fwupd/metainfo
04:49:14:0450 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
04:49:14:0450 XbSilo               file: 44274c8a-9db0-5757-8ea8-3c7ad4e80f35, current:44274c8a-9db0-5757-8ea8-3c7ad4e80f35, cached: (null)
04:49:14:0450 XbSilo               loading silo with file contents
04:49:14:0450 FuConfig             using config path of /etc/fwupd
04:49:14:0450 FuConfig             loading config from /etc/fwupd/remotes.d/fwupd.conf
04:49:14:0450 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
04:49:14:0450 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-directory.conf
04:49:14:0450 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
04:49:14:0450 FuConfig             loading config from /etc/fwupd/remotes.d/vendor.conf
04:49:14:0450 FuConfig             using config path of /usr/lib/fwupd
04:49:14:0450 FuConfig             path /usr/lib/fwupd/remotes.d does not exist
04:49:14:0450 FuConfig             ordering lvfs-testing=lvfs+1
04:49:14:0450 FuConfig             ordering lvfs=fwupd+1
04:49:14:0450 FuConfig             ordering lvfs-testing=lvfs+1
04:49:14:0450 FuIdle               setting timeout to 7200s
04:49:14:0450 FuHwids              smbios property Manufacturer=Gigabyte Technology Co., Ltd.
04:49:14:0450 FuHwids              smbios property EnclosureKind=3
04:49:14:0450 FuHwids              smbios property Family=Default string
04:49:14:0450 FuHwids              smbios property ProductName=Z370 HD3P
04:49:14:0450 FuHwids              smbios property ProductSku=Default string
04:49:14:0450 FuHwids              smbios property BiosVendor=American Megatrends Inc.
04:49:14:0450 FuHwids              smbios property BiosVersion=F4
04:49:14:0450 FuHwids              smbios property BiosMajorRelease=05
04:49:14:0450 FuHwids              smbios property BiosMinorRelease=0c
04:49:14:0450 FuHwids              smbios property BaseboardManufacturer=Gigabyte Technology Co., Ltd.
04:49:14:0450 FuHwids              smbios property BaseboardProduct=Z370 HD3P-CF
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/altos.quirk
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ata.quirk
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/colorhug.quirk
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/csr-aiaiai.quirk
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell-dock.quirk
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell.quirk
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dfu.quirk
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ebitdo.quirk
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/fastboot.quirk
04:49:14:0451 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/flashrom.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nitrokey.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nvme.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hid.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hub.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/steelseries.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/superio.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/synapticsmst.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/uefi.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/unifying.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-raw.quirk
04:49:14:0452 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-usb.quirk
04:49:14:0452 FuQuirks             now 183 quirk entries
04:49:14:0452 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
04:49:14:0452 FuEngine             remote lvfs-testing not enabled, so skipping
04:49:14:0452 FuEngine             remote vendor not enabled, so skipping
04:49:14:0452 FuEngine             remote vendor-directory not enabled, so skipping
04:49:14:0452 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
04:49:14:0452 XbSilo               0.01ms      parse blob
04:49:14:0452 XbSilo               0.05ms      loaded file
04:49:14:0452 XbSilo               file: eeba95ef-e449-5463-aeb5-be5b85fec3b0, current:eeba95ef-e449-5463-aeb5-be5b85fec3b0, cached: (null)
04:49:14:0452 XbSilo               loading silo with file contents
04:49:14:0452 XbSilo               0.00ms      parse blob
[... cut ...]
04:49:14:0755 FuPluginAta          ATA_12 status=0x0, host_status=0x0, driver_status=0x0
04:49:14:0755 FuPluginAta          SB:             00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
04:49:14:0755 FuPluginAta          ATA_12 stat=00 err=00 nsect=00 lbal=00 lbam=00 lbah=00 dev=00
04:49:14:0755 FuPluginAta          invalid component ID, skipping
04:49:14:0755 FuDevice             2048e0e404611a839d4463e0db69133fa41fc116 device overwriting same name value: (null)
Segmentation fault

hughsie, 2019-05-24:
I can't immediately see why it's crashing, but ^^ might fix this bug. If you can install debug symbols for fwupd and get a backtrace that would help a lot to debug this. Thanks!

closed 2019-05-24 by hughsie

Ali537860, 2019-08-27:
i dont know why it says "sorry something went wrong: error calling start service by name for org.freedesktop.fwupd
Please help me fix this

Ali537860, 2019-08-27:
12:44:09:0635 FuDebug              Verbose debugging enabled (on console 1)
12:44:09:0636 FuConfig             loading config values from /etc/fwupd/daemon.conf
12:44:09:0649 Fwupd                looking for os-release at /etc/os-release
12:44:09:0649 FuConfig             loading /usr/share/fwupd/metainfo
12:44:09:0649 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
12:44:09:0649 XbSilo               file: 85fa9e2b-58ae-54b7-8157-5aeaa1c5c1c5, current:85fa9e2b-58ae-54b7-8157-5aeaa1c5c1c5, cached: (null)
12:44:09:0649 XbSilo               loading silo with file contents
12:44:09:0649 FuConfig             using config path of /etc/fwupd
12:44:09:0650 FuConfig             loading config from /etc/fwupd/remotes.d/vendor.conf
12:44:09:0650 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
12:44:09:0650 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-directory.conf
12:44:09:0650 FuConfig             loading config from /etc/fwupd/remotes.d/fwupd.conf
12:44:09:0650 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
12:44:09:0650 FuConfig             using config path of /usr/lib/fwupd
12:44:09:0650 FuConfig             path /usr/lib/fwupd/remotes.d does not exist
12:44:09:0650 FuConfig             ordering lvfs=fwupd+1
12:44:09:0651 FuConfig             ordering lvfs-testing=lvfs+1
12:44:09:0651 FuIdle               setting timeout to 7200s
12:44:09:0651 FuHwids              smbios property Manufacturer=HP
12:44:09:0651 FuHwids              smbios property EnclosureKind=a
12:44:09:0651 FuHwids              smbios property Family=103C_5336AN HP 200
12:44:09:0651 FuHwids              smbios property ProductName=HP 255 G6 Notebook PC
12:44:09:0651 FuHwids              smbios property ProductSku=6BP55ES#ABU
12:44:09:0651 FuHwids              smbios property BiosVendor=Insyde
12:44:09:0651 FuHwids              smbios property BiosVersion=F.31
12:44:09:0651 FuHwids              smbios property BiosMajorRelease=0f
12:44:09:0651 FuHwids              smbios property BiosMinorRelease=1f
12:44:09:0651 FuHwids              smbios property BaseboardManufacturer=HP
12:44:09:0651 FuHwids              smbios property BaseboardProduct=8330
12:44:09:0652 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/altos.quirk
12:44:09:0652 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ata.quirk
12:44:09:0652 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/colorhug.quirk
12:44:09:0652 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/csr-aiaiai.quirk
12:44:09:0652 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell-dock.quirk
12:44:09:0653 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell.quirk
12:44:09:0653 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dfu.quirk
12:44:09:0654 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ebitdo.quirk
12:44:09:0655 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/fastboot.quirk
12:44:09:0655 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/flashrom.quirk
12:44:09:0655 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nitrokey.quirk
12:44:09:0655 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nvme.quirk
12:44:09:0655 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hid.quirk
12:44:09:0655 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hub.quirk
12:44:09:0655 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/steelseries.quirk
12:44:09:0655 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/superio.quirk
12:44:09:0656 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/synapticsmst.quirk
12:44:09:0656 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/uefi.quirk
12:44:09:0656 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/unifying.quirk
12:44:09:0656 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-raw.quirk
12:44:09:0656 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-usb.quirk
12:44:09:0657 FuQuirks             now 183 quirk entries
12:44:09:0657 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
12:44:09:0657 FuEngine             remote lvfs-testing not enabled, so skipping
12:44:09:0657 FuEngine             no /var/lib/fwupd/remotes.d/lvfs/metadata.xml.gz, so skipping
12:44:09:0657 FuEngine             remote vendor not enabled, so skipping
12:44:09:0657 FuEngine             remote vendor-directory not enabled, so skipping
12:44:09:0657 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
12:44:09:0657 XbSilo               0.01ms      parse blob
12:44:09:0657 XbSilo               0.14ms      loaded file
12:44:09:0658 XbSilo               file: 07cfe69d-723d-5f75-b291-0676e7f6ea42, current:07cfe69d-723d-5f75-b291-0676e7f6ea42, cached: (null)
12:44:09:0658 XbSilo               loading silo with file contents
12:44:09:0658 XbSilo               0.03ms      parse blob
12:44:09:0658 XbSilo               0.02ms      query on / with `components/component` limit=0 -> 5 results
12:44:09:0658 FuEngine             5 components now in silo
12:44:09:0658 XbSilo               0.01ms      query on / with `components/component/provides/firmware` limit=0 -> 5 results
[... cut ...]
12:44:09:0695 FuPlugin             performing coldplug() on superio
12:44:09:0696 FuPlugin             performing coldplug() on thunderbolt
12:44:09:0696 FuPlugin             performing coldplug() on thunderbolt_power
12:44:09:0697 FuPluginThunderbolt  Bolt force power support: 0
12:44:09:0697 FuEngine             disabling plugin because: failed to coldplug using thunderbolt_power: No support for force power via kernel or bolt
12:44:09:0698 FuPlugin             performing coldplug() on synapticsmst