fwupd/fwupd #1207: Dell WD15 not being listed

aleixpol, 2019-05-23:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.2.10
checkout info:  1.2.9-5-g568bae29
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.10
```
This is a git master snapshot from this week.

```shell
$ fwupdmgr get-devices
XPS 13 9370 Thunderbolt Controller
  DeviceId:             48e75942786c910eee5d787eb5e162f5fcd949ba
  Guid:                 4eeb9d07-a96c-56d6-92d3-4a23ee7a6e4a
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              33.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-05-23

XPS 13 9370 System Firmware
  DeviceId:             8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
  Guid:                 7ceaf7a8-0611-4480-9e30-64d8de420c7c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.9.0
  VersionLowest:        0.1.9.0
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-05-23

KXG50ZNV512G NVMe TOSHIBA 512GB
  DeviceId:             f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
  Guid:                 4d0aed03-a30c-52c6-99e7-a8977797c3d9
  Guid:                 98d34383-a939-5b5a-9806-e05e0b288b1a
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              AADA4101
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-05-23
```

```shell
$ efibootmgr -v

BootCurrent: 0005
Timeout: 0 seconds
BootOrder: 0001,0002,0003,0004,0005
Boot0000* ubuntu        HD(1,GPT,7f41bf6e-6fc4-463d-946b-65877305ad50,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Diskette Drive        BBS(Floppy,Diskette Drive,0x0)..BO
Boot0002* M.2 PCIe SSD  BBS(HD,P0: KXG50ZNV512G NVMe TOSHIBA 512GB,0x0)..BO
Boot0003* USB Storage Device    BBS(USB,USB Storage Device,0x0)..BO
Boot0004* CD/DVD/CD-RW Drive    BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* linux PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-02-00-2E-3B-EF)/HD(1,GPT,7f41bf6e-6fc4-463d-946b-65877305ad50,0x800,0x12c000)/File(\EFI\grub\grubx64.efi)
Boot0006* Linux Firmware Updater        HD(1,GPT,7f41bf6e-6fc4-463d-946b-65877305ad50,0x800,0x12c000)/File(\EFI\arch\fwupdx64.efi)
```

```shell
$ efivar -l | grep fw
```
^ it's empty

Please answer the following questions:

- Operating system and version: ArchLinux
- How did you install fwupd: from source, installing with pacman had the same results though.
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? don't think so

hughsie, 2019-05-23:
@superm1 ?

superm1, 2019-05-23:
Can you please share a verbose daemon log?

aleixpol, 2019-05-23:
```
12:50:02:0528 FuDebug              Verbose debugging enabled (on console 0)
12:50:02:0529 FuConfig             loading config values from /etc/fwupd/daemon.conf
12:50:02:0529 Fwupd                looking for os-release at /etc/os-release
12:50:02:0529 FuConfig             loading /usr/share/fwupd/metainfo
12:50:02:0529 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
12:50:02:0529 XbSilo               file: e9d160e6-9745-5e9e-a1c0-ace66de66093, current:e9d160e6-9745-5e9e-a1c0-ace66de66093, cached: (null)
12:50:02:0529 XbSilo               loading silo with file contents
12:50:02:0530 FuConfig             using config path of /etc/fwupd
12:50:02:0530 FuConfig             loading config from /etc/fwupd/remotes.d/dell-esrt.conf
12:50:02:0530 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-directory.conf
12:50:02:0530 FuConfig             loading config from /etc/fwupd/remotes.d/vendor.conf
12:50:02:0530 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
12:50:02:0530 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
12:50:02:0531 FuConfig             loading config from /etc/fwupd/remotes.d/fwupd-tests.conf
12:50:02:0531 FuConfig             using config path of /usr/lib/fwupd
12:50:02:0531 FuConfig             path /usr/lib/fwupd/remotes.d does not exist
12:50:02:0531 FuConfig             ordering lvfs-testing=lvfs+1
12:50:02:0531 FuConfig             ignoring unfound remote fwupd
12:50:02:0531 FuConfig             ignoring unfound remote fwupd
12:50:02:0531 FuConfig             ignoring unfound remote fwupd
12:50:02:0531 FuConfig             ignoring unfound remote fwupd
12:50:02:0538 FuEngine             client certificate exists and working
12:50:02:0538 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
12:50:02:0539 FuHistory            got schema version of 5
12:50:02:0539 FuIdle               setting timeout to 7200s
12:50:02:0539 FuHwids              smbios property Manufacturer=Dell Inc.
12:50:02:0539 FuHwids              smbios property EnclosureKind=a
12:50:02:0539 FuHwids              smbios property Family=XPS
12:50:02:0539 FuHwids              smbios property ProductName=XPS 13 9370
12:50:02:0539 FuHwids              smbios property ProductSku=07E6
12:50:02:0539 FuHwids              smbios property BiosVendor=Dell Inc.
12:50:02:0539 FuHwids              smbios property BiosVersion=1.9.0
12:50:02:0539 FuHwids              smbios property BiosMajorRelease=01
12:50:02:0539 FuHwids              smbios property BiosMinorRelease=09
12:50:02:0539 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
12:50:02:0540 FuHwids              smbios property BaseboardProduct=0F6P3V
12:50:02:0540 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/altos.quirk
12:50:02:0540 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ata.quirk
12:50:02:0540 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/colorhug.quirk
12:50:02:0540 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/csr-aiaiai.quirk
12:50:02:0540 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell-dock.quirk
12:50:02:0541 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell.quirk
12:50:02:0541 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dfu.quirk
12:50:02:0541 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ebitdo.quirk
12:50:02:0542 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/fastboot.quirk
12:50:02:0542 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/flashrom.quirk
12:50:02:0542 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nitrokey.quirk
12:50:02:0542 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nvme.quirk
12:50:02:0542 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hid.quirk
12:50:02:0542 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hub.quirk
12:50:02:0542 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/steelseries.quirk
12:50:02:0543 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/superio.quirk
12:50:02:0543 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/synaptics-prometheus.quirk
12:50:02:0543 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/synapticsmst.quirk
12:50:02:0543 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/uefi.quirk
12:50:02:0543 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/unifying.quirk
12:50:02:0543 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-raw.quirk
12:50:02:0543 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-usb.quirk
12:50:02:0544 FuQuirks             now 187 quirk entries
12:50:02:0544 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
12:50:02:0544 FuEngine             remote lvfs-testing not enabled, so skipping
12:50:02:0544 FuEngine             remote fwupd-tests not enabled, so skipping
12:50:02:0544 FuEngine             remote vendor not enabled, so skipping
12:50:02:0544 FuEngine             remote vendor-directory not enabled, so skipping
12:50:02:0544 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
12:50:02:0544 XbSilo               0,02ms      parse blob
[... cut ...]
12:50:06:0165 FuCommon             not checking 0.1.5.1 as no version format set
12:50:06:0165 FuCommon             not checking 0.1.6.3 as no version format set
12:50:06:0165 FuEngine             Emitting PropertyChanged('Status'='idle')
12:50:06:0165 FuMain               Emitting PropertyChanged('Status'='idle')
12:50:06:0167 FuMain               Started with locale ca_ES.UTF-8
12:50:06:0167 FuMain               Daemon ready for requests
12:50:06:0167 FuMain               FuMain: acquired name: org.freedesktop.fwupd
```

aleixpol, 2019-05-23:
I guess this one is especially relevant:
`no GUIDs for device Synaptics VMM3332 inside Dell WD15/TB16/TB18 wired Dock [d83854e8c81f5d47b0b99c8a3b42ae5127c75eb8]
`

superm1, 2019-05-23:
Is the Ethernet controller in the dock working?  Could you have possibly disabled it in BIOS setup?

superm1, 2019-05-23:
It looks like potentially separate problems to me, synapticsmst should have made a device and you correctly called out the error for that, but also there should be another device for the embedded controller in the dock.

aleixpol, 2019-05-23:
Yes, ethernet is working fine. As well as VGA.

I do get a slightly weird thing where I can see the usb reconnect repeatedly on `dmesg`.

superm1, 2019-05-23:
There is a PR for fixing the synapticsmst GUID generation ^ 

> I do get a slightly weird thing where I can see the usb reconnect repeatedly on dmesg.

USB reconnecting rapidly but ethernet still works?  That's odd.

aleixpol, 2019-05-23:
With your patch I do get it:
```
Synaptics VMM3332 inside Dell WD15/TB16/TB18 wired Dock
  DeviceId:             d83854e8c81f5d47b0b99c8a3b42ae5127c75eb8
  Guid:                 78c73924-821a-5e6f-af82-86df6f67b182
  Guid:                 02b77ccb-03ab-521f-b98e-ef05395e9fa2
  Guid:                 a6f00ca7-54b7-5d51-9d6e-543d8df85376
  Summary:              Multi-Stream Transport Device
  Plugin:               synapticsmst
  Flags:                updatable|registered
  Vendor:               Synaptics
  Version:              3.10.002
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-05-23
```

aleixpol, 2019-05-23:
If you want, I can open another bug. It doesn't have the `supported` flag. Is that right?

superm1, 2019-05-23:
There is not a firmware posted in stable LVFS channel, so that's expected.  It's only in the 'testing' channel; https://fwupd.org/lvfs/device/com.dell.mst0a52c8c7.firmware

aleixpol, 2019-05-23:
okay, thanks a lot!! :)

aleixpol, 2019-05-23:
managed to update the firmware, worked brilliantly.

commit 6f00fad6 (2019-05-23): synapticsmst: fix GUID generation (Closes: #1207)

When converting to instances we forgot to add the GUIDs to the device
as well.
commit 6f352cb7 (2019-05-23): synapticsmst: fix GUID generation (Closes: #1207)

When converting to instances we forgot to add the GUIDs to the device
as well.
closed 2019-05-23 by superm1