fwupd/fwupd #1211: Unable to install firmware update 0.1.10.1 on Dell Precision 5530

lgp171188, 2019-05-24:
`fwupdmgr` states that there are system firmware updates available for my Dell Precision 5530 laptop.
```shell
$ fwupdmgr get-updates
Precision 5530 System Firmware has firmware updates:
Device ID:               66352e92895dc6d018071ad4ecfae249b0b8af0d
GUID:                    e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0
ID:                      com.dell.uefie4d7bdc4.firmware
Update Version:          0.1.10.1
Update Name:             XPS 15 9570/Precision 5530 System Update Update
Update Summary:          Firmware for the Dell XPS 15 9570/Precision 5530
Update Remote ID:        lvfs
Update Checksum:         SHA1(09a5c812a6eae3c436e36a4b5ea76016a32bc1f1)
Update Location:         https://fwupd.org/downloads/dc3d2832d3689fcf8294e03c45c728daa087b1e5-XPS_9570_BIOS_1.10.1_firmware.cab
Update Description:      Some new functionality has also been added:
                         
                          • Reduces Deferred Procedure Call (DPC) latency for real-time task performance enhancement, such as audio playback.
                          • Supported Dell Dock Station WD19.
```
However, I am unable to install the update because of an error.
```
$ fwupdmgr update
Downloading 0.1.10.1 for Precision 5530 System Firmware...
Decompressing…           [***************************************]
Firmware version formats were different, device was 'quad' and release is 'plain'
```
Though it has not happened before, the firmware update file explicitly mentions Dell XPS 9570 which is a computer similar to my Dell Precision 5530 and probably use the same BIOS.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.8
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Precision 15 M5530  Thunderbolt Controller
  DeviceId:             3edbc5e81978be9b0aabe51ca0329fc548cb2a8a
  Guid:                 a35a21e3-7eb7-56d5-89c8-6e8b28d33ac2
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              33.00
  Icon:                 computer
  Created:              2019-05-24

Precision 5530 System Firmware
  DeviceId:             66352e92895dc6d018071ad4ecfae249b0b8af0d
  Guid:                 e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.8.1
  VersionLowest:        0.1.8.1
  Icon:                 computer
  Created:              2019-05-24
  UpdateState:          success

KXG50ZNV512G NVMe TOSHIBA 512GB
  DeviceId:             f2759da7fe8e0388c5f3601cb072f837b1070b03
  Guid:                 4d0aed03-a30c-52c6-99e7-a8977797c3d9
[... 343 lines cut ...]
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-4.19-x86_64-fallback.img
├── initramfs-4.19-x86_64.img
├── intel-ucode.img
├── linux419-x86_64.kver
├── memtest86+
│   └── memtest.bin
└── vmlinuz-4.19-x86_64

11 directories, 318 files
```

Please answer the following questions:

- Operating system and version: Manjaro Linux 18.0.4
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

hughsie, 2019-05-24:
The recently-released 1.2.9 should fix this for you -- can you try that and confirm it fixes the bug please?

lgp171188, 2019-05-24:
@hughsie, I upgraded to the 1.2.9 release and this issue doesn't happen any more. However, I had to re-enable the LVFS remote as it had gotten disabled due to some reason (maybe some issue with my distro's packaging?) and wasn't showing any updates.

hughsie, 2019-05-24:
Does Manjaro disable the LVFS by default perhaps?

lgp171188, 2019-05-24:
@hughsie, I did not set up anything at all and it before till I upgraded to 1.2.9 from Manjaro testing. It might have been due to that or due to something I did (I didn't run any commands related to remotes). But since the issue is fixed, I think this is good to be closed. :+1: Thanks for the help!

closed 2019-05-24 by lgp171188