fwupd/fwupd #1219: test suite fails if USB context creation fails

fabled, 2019-06-09:
fwupd 1.2.9

Test suite fails with:
  Failed to get USB context: failed to init libusb: Other error [-99]

if ran on virtual environment without USB controller.

Would be nice to make things work even without USB so building and test suite run works on virtualized, minimal build environment.

hughsie, 2019-06-09:
I'd be happy to accept a patch for this. It would also help if I could reproduce the problem here, and then it could be added to the CI tests for instance.

fabled, 2019-06-10:
Seems the usb context presence triggers the loading of plugins at:
https://github.com/hughsie/fwupd/blob/master/src/fu-engine.c#L4111

Wondering how to fix that. Otherwise it seems we can just make the usb context creation failure a soft error and make the other parts needing it conditional.

hughsie, 2019-06-10:
> and make the other parts needing it conditional

This is really the kind of thing we need in CI to avoid regressing, if we want to support it.

fabled, 2019-06-10:
So since `bubblewrap` is already used in the tests, that might be the best tool to reproduce. I can reproduce this with as simple as:
``` sh
bwrap --ro-bind / / --bind /tmp /tmp --bind $PWD $PWD --proc /proc --dev /dev  /bin/sh
```

and I get:
```
ninja: Entering directory `output'
[0/1] Running all tests.
 1/14 fwupd-exported-api                      OK       0.00 s 
 2/14 fwupd-self-test                         OK       0.01 s 
 3/14 fu-self-test                            FAIL     0.42 s (killed by signal 6 SIGABRT)
 4/14 ata-self-test                           OK       0.01 s 
 5/14 dfu-self-test                           OK       0.03 s 
 6/14 nitrokey-self-test                      OK       0.01 s 
 7/14 synaptics-prometheus-self-test          OK       0.01 s 
 8/14 udev-self-test                          OK       0.01 s 
 9/14 unifying-self-test                      OK       0.01 s 
10/14 wacom-usb-self-test                     SKIP     0.01 s 
11/14 nvme-self-test                          OK       0.02 s 
12/14 dell-self-test                          OK       0.18 s 
13/14 synapticsmst-self-test                  OK       0.02 s 
14/14 uefi-self-test                          OK       0.04 s 

Ok:                   12
Expected Fail:         0
Fail:                  1
Unexpected Pass:       0
Skipped:               1
Timeout:               0

The output from the failed tests:

 3/14 fu-self-test                            FAIL     0.42 s (killed by signal 6 SIGABRT)

[[ snip ]]

(/data/aports/testing/fwupd/src/fwupd-1.2.9/output/src/fu-self-test:21908): XbSilo-DEBUG: 11:21:28.775: attempting to load /tmp/fwupd-self-test/var/cache/fwupd/metadata.xmlb
(/data/aports/testing/fwupd/src/fwupd-1.2.9/output/src/fu-self-test:21908): XbSilo-DEBUG: 11:21:28.775: failed to load silo: Failed to open file */tmp/fwupd-self-test/var/cache/fwupd/metadata.xmlb*: open() failed: No such file or directory
(/data/aports/testing/fwupd/src/fwupd-1.2.9/output/src/fu-self-test:21908): XbSilo-DEBUG: 11:21:28.775: ignoring index: silo has no data
(/data/aports/testing/fwupd/src/fwupd-1.2.9/output/src/fu-self-test:21908): XbSilo-DEBUG: 11:21:28.775: ignoring index: silo has no data
--- stderr ---
**
ERROR:../src/fu-self-test.c:724:fu_engine_device_unlock_func: assertion failed (error == NULL): Failed to get USB context: failed to init libusb: Other error [-99] (g-usb-context-error-quark, 0)
-------

Full log written to /data/aports/testing/fwupd/src/fwupd-1.2.9/output/meson-logs/testlog.txt
FAILED: meson-test 
/usr/bin/meson test --no-rebuild --print-errorlogs
ninja: build stopped: subcommand failed.
```

superm1, 2019-07-15:
I tried to use your reproducer, but it doesn't seem to fail for me.

```
# bwrap --ro-bind / / --bind /tmp /tmp --bind $PWD $PWD --proc /proc --dev /dev  ninja -C output test

ninja: Entering directory `output'
[0/1] Running all tests.
 1/15 fwupd-exported-api                      OK       0.00 s 
 2/15 fwupd-self-test                         OK       0.02 s 
 3/15 fu-self-test                            OK       6.75 s 
 4/15 ata-self-test                           OK       0.02 s 
 5/15 dfu-self-test                           OK       0.03 s 
 6/15 nitrokey-self-test                      OK       0.01 s 
 7/15 synaptics-prometheus-self-test          OK       0.02 s 
 8/15 udev-self-test                          OK       0.02 s 
 9/15 unifying-self-test                      OK       0.01 s 
10/15 wacom-usb-self-test                     SKIP     0.02 s 
11/15 nvme-self-test                          OK       0.02 s 
12/15 redfish-self-test                       OK       0.02 s 
13/15 dell-self-test                          OK       0.27 s 
14/15 synapticsmst-self-test                  OK       0.02 s 
15/15 uefi-self-test                          OK       0.02 s 

Ok:                   14
Expected Fail:         0
Fail:                  0
Unexpected Pass:       0
Skipped:               1
Timeout:               0
```

hughsie, 2019-07-15:
> but it doesn't seem to fail for me

Me either.

superm1, 2019-07-15:
Alright, I'll close. If there is more info on how to get this to happen, please re-open.

closed 2019-07-15 by superm1

fabled, 2019-07-16:
Please reopen. This does happen and is reproducible on modern systems. The difference you experience is probably related to kernel version, or how your /dev is setup.

libusb search for `/dev/bus/usb` and `/proc/bus/usb` and if neither exists, it will fail. This is common scenario in virtual servers and containers.  Maybe trying to make them empty explicitly helps, would this reproduce it for you:
```
bwrap --ro-bind / / --bind /tmp /tmp --bind $PWD $PWD --proc /proc --dev /dev --tmpfs /dev/bus/usb --tmpfs /proc/bus ninja -C output test
```

If not, what perhaps you can post output of:
```
bwrap --ro-bind / / --bind /tmp /tmp --bind $PWD $PWD --proc /proc --dev /dev --tmpfs /dev/bus/usb --tmpfs /proc/bus strace ./output/src/fu-self-test
```
and let me know which kernel and libusb you are using?

This is reproducible on Alpine Linux 4.19.58-0-vanilla, with libusb 1.0.22.

hughsie, 2019-07-16:
This works:

    bwrap --ro-bind / / --bind /tmp /tmp --bind $PWD $PWD --proc /proc --dev /dev --tmpfs /dev/bus/usb --tmpfs /proc/bus ninja -C build test

> If not, what perhaps you can post output of:

attached.

> and let me know which kernel and libusb you are using?

Fedora 30, 5.1.16-300.fc30.x86_64, 
[log.zip](https://github.com/hughsie/fwupd/files/3395931/log.zip)
libusbx-1.0.22-2.fc30.x86_64

fabled, 2019-07-16:
Thanks. Seems libusb is configured differently. We have `--disable-udev` in Alpine. And that seems to make a big difference on how libusb behaves. They have code in like:
```
/* On udev based systems without any usb-devices /dev/bus/usb will not
 * exist. So if we've not found anything and we're using udev for hotplug
 * simply assume /dev/bus/usb rather then making libusb_init fail. */
#if defined(USE_UDEV)
        if (ret == NULL)
                ret = "/dev/bus/usb";
#endif
```

This does imply that `libusb_init` is not expected to fail even if usb devices are absent. So perhaps I should file this as a bug against libusb.

hughsie, 2019-07-16:
> This does imply that libusb_init is not expected to fail even if usb devices are absent

I think that sounds correct, I'd expect it to succeed even if there were no USB devices.

mcuee, 2021-07-14:
https://github.com/libusb/libusb/issues/595
>libusb_init may fail if no USB bus exists. This is not a libusb bug.
>When the container does not support libudev (--disable-udev) and Netlink, libusb_init() will fail.

hughsie, 2021-07-14:
@mcuee what fix do you propose?

mcuee, 2021-07-14:
I am just saying that the assumption that libusb_init() will succeed is not correct. In that case, fwupd can just fail that case and not test the libusb related case.

hughsie, 2021-07-14:
At the moment we have in GUsb:

	rc = libusb_init (&ctx);
	if (rc < 0) {
		g_set_error (error,
			     G_USB_CONTEXT_ERROR,
			     G_USB_CONTEXT_ERROR_INTERNAL,
			     "failed to init libusb: %s [%i]",
			     g_usb_strerror (rc), rc);
		return FALSE;
	}

Do we know what the return code failures libusb_init() could possibly return? If you can compile a list I can fix up GUsb to do something more sane.

mcuee, 2021-07-14:
https://libusb.sourceforge.io/api-1.0/group__libusb__lib.html#ga9517c37281bba0b51cc62eba728be48b
https://github.com/libusb/libusb/blob/master/libusb/core.c#L2273
Right now it is not well documented. Please use https://github.com/libusb/libusb/issues/595 to continue the discussion. The error codes can be dependent on the backend.

mcuee, 2021-07-14:
BTW, I have no idea with the comment here. That can not be true right? If it is not a USB device, why libusb_init() failure is a show stopper.
https://github.com/libusb/libusb/issues/595#issuecomment-879374859
>I have non-USB devices requiring firmware updates to work correctly and libusb is a showstopper for me.

hughsie, 2021-07-14:
> why libusb_init() failure is a show stopper

Because it's never been known to fail before, so we didn't have the code to handle the failure. If you can compile a list I can fix up GUsb to do something more sane.

mcuee, 2021-07-14:
IMHO you do not really need for the detail codes in reality, just use the following. But I understand you want to help the user to know more about the error codes, for that you will have to wait.
https://libusb.sourceforge.io/api-1.0/group__libusb__lib.html#ga9517c37281bba0b51cc62eba728be48b

0 on success, or a LIBUSB_ERROR code on failure

mcuee, 2021-07-14:
I mentioned the documentation improvement on the error code in https://github.com/libusb/libusb/issues/595

hughsie, 2021-07-14:
Right, but `LIBUSB_ERROR_OTHER` is useful to nobody. It should be something like LIBUSB_ERROR_NOT_SUPPORTED surely?

mcuee, 2021-07-14:
Yes, that is a good point. Pull request is welcome.