fwupd/fwupd #1222: dfu-tool hex to dfu bytes too large to fill

microwavesafe, 2019-06-12:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
fwupd 1.0.9
```

Please answer the following questions:

- Operating system and version:
Ubuntu 18.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
sudo apt install fwupd

I use dfu-tool to convert a hex file into a dfu file for programming into an ST microcontroller. This has been working well, but I'm currently trying to get the option bytes into the hex file, which I've done, but this is causing a fault in the hex to dfu conversion.

These are the final lines of the hex file, you can see the address jumps to the option bytes address, which jumps over the entire memory address range.

```
:10678C0000000000000000000000000000000000FD
:10679C0000000000000000000000000000000000ED
:020000041FFFDC
:04F8000000FF55AA06
:0400000508001E4F82
:00000001FF
```
This causes the hex file to fall into this condition

https://github.com/hughsie/fwupd/blob/5d7f5a2a87162407c6680dcf8681dcf65321ccd0/plugins/dfu/dfu-format-ihex.c#L219

and the conversion fails. Is there a work around to this situation? I don't have to have the option bytes in the resultant dfu file, but on the other hand it doesn't hurt to keep them programmed.

hughsie, 2019-06-12:
> This causes the hex file to fall into this condition

What would be the right thing to do in this situation?

microwavesafe, 2019-06-12:
I'm not sure as I'm not familiar with the dfu format. Previously I have just used the tool to convert and not had to dig into it.

Is it possible to have address jumps like the hex file? So we don't have to fill large gaps.

microwavesafe, 2019-06-12:
In my particular instance, as a fallback, I would prefer to drop the data after the large gap. But this doesn't feel like a good solution.

hughsie, 2019-06-12:
A raw file has to be "flat" as in you can't have partitions or holes like you can with an ihex or dfuse file. The DFU format is really just a raw file with a few-hundred byte footer. Could you perhaps just manually drop the option bytes from the ihex file prior to using dfu-tool to convert it to a dfu file?

microwavesafe, 2019-06-12:
I generate the file like this
```
dfu-tool convert dfuse "file.hex" "file.dfu"
```
Does this generate a raw dfu file, not a dfuse file?

hughsie, 2019-06-12:
Ahh, that does indeed generate a `DfuSe` file, which now makes sense as you're talking about deploying to ST rather than something like PIC... I suppose we could use two images in the DfuSe format file, although I don't know what the ST tools would do with the extra partition.

microwavesafe, 2019-06-12:
That I don't know. Currently I use the built in DFU bootloader of the STM32F parts, along with
```
dfu-util -a 0 -D file.dfu
````
with the generated file from the above command. This all works well at the moment, but I'm not sure if the built in DFU bootloader or dfu-util are both compatible with splitting the image up?

hughsie, 2019-06-12:
Did you try flashing with `dfu-tool`? We can add the required logic to ignore the option bytes there.

microwavesafe, 2019-06-12:
I haven't, I'll be honest I didn't know it could. I will try this and get back to you, it may take me a while to try it out. Thanks for your help by the way.

microwavesafe, 2019-06-12:
Do you think it would be possible to actually write the option bytes, instead of ignore them? I can see a use case for being able to program fresh devices, with code protection, by third parties that don't have programmers that can connect over SWD pins.

hughsie, 2019-06-12:
Yes, I think writing them is the best thing to do.

microwavesafe, 2019-06-12:
Looking into it, I would be keen to get dfu-utils to output a file with multiple images, the reason is some users will be using Windows and unless I'm mistaken dfu-tool isn't designed to run on Windows?

Would it be a question of creating a new image section if the gap size condition is triggered, then adding multiple images at the end of the ```dfu_firmware_from_ihex``` function? Or is this same function used to generate the raw dfu files as well?

microwavesafe, 2019-06-13:
Having looked into it further this method wont work. I presumed you could just write to the option bytes, but you have to perform an unlocking procedure on another register before you can. So I can't simply add the option bytes to the hex file.

closed 2019-06-13 by microwavesafe