fwupd/fwupd #1225: Not compatible with bootloader version?

lovesegfault, 2019-06-17:
Whenever I run `fwupdmgr get-updates` I get an annoying "warning" regarding a device with no releases, like the following:
```shell
$ fwupdmgr get-updates
No upgrades for 20MD0028US System Firmware, current is 0.1.22: 0.1.15=older, 0.1.17=older, 0.1.18=older, 0.1.21=older, 0.1.19=older
No upgrades for UEFI Device Firmware, current is 192.35.1427: 192.6.1120=older
No releases found for device: Not compatible with bootloader version: failed predicate [BOT01.0[0-3]_* regex BOT01.04_B0016]
```
Where does this come from? How can I fix it?

Debug info:
```shell
$ fwupdmgr --version
client version:	1.2.9
compile-time dependency versions
	gusb:	0.2.11
	efivar:	37
daemon version:	1.2.9
```
```shell
$ fwupdmgr get-devices
ThinkPad P1 Thunderbolt Controller
  DeviceId:             82426a9dfceb97f00ac3124aa9d69f6b44b8a542
  Guid:                 138f4141-7a3f-57a3-8e38-f10c038f2d91
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              36.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-06-17

ThinkPad Thunderbolt 3 Dock
  DeviceId:             dfa4929b492257b32e7dcdc18def55d97a97fb58
  Guid:                 75039175-c354-5df9-a798-5da41461c007
  Plugin:               thunderbolt
  Flags:                updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0108
  Version:              36.01
  VersionFormat:        pair
  Icon:                 audio-card
  Created:              2019-06-17

20MD0028US System Firmware
  DeviceId:             71519311a643244a2eaf392fbf17c968ec135dd3
  Guid:                 a95d69e7-2fa0-4849-9159-75645bc6e153
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.22
  VersionLowest:        0.1.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-06-17

UEFI Device Firmware
  DeviceId:             aca1c6cf5901b23b637fc8a1fdafeac85700799d
  Guid:                 4226a48a-8c1e-4f66-8f54-7e5dc33dcba9
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              192.35.1427
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-06-17

UEFI Device Firmware
  DeviceId:             6b3fcb33b92ea5cdd94954b276ad0b63ed14f9bb
  Guid:                 74997a6b-1adf-4b12-b994-401f06ea8c72
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.10
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-06-17

Unifying Receiver
  DeviceId:             0805ff7fe49b5b5a5c1d0e181d4931c585c1b70d
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.09_B0030
  VersionBootloader:    BOT01.04_B0016
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-06-17

SAMSUNG MZVLB1T0HALR-000L7
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 08575ca8-c22c-540a-8ac9-e784c04f6e5a
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              5L2QEXA7
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-06-17

SAMSUNG MZVLB1T0HALR-000L7
  DeviceId:             c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 08575ca8-c22c-540a-8ac9-e784c04f6e5a
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              5L2QEXA7
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-06-17
```
```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0001,0018,0019,001A,001B,001C,001D,001E,001F,0020,0021
Boot0000* gentoo	HD(1,GPT,4d062d52-585e-4da5-bb90-97d2a4df8644,0x800,0x100000)/File(\EFI\gentoo\grubx64.efi)
Boot0001* Windows Boot Manager	HD(1,GPT,4d062d52-585e-4da5-bb90-97d2a4df8644,0x800,0x100000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...a................
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Regulatory Information	FvFile(478c92a0-2622-42b7-a65d-5894169e4d24)
Boot0015  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0016  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0017  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0018* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0019* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot001A* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001C* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001D* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001E* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001F* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0020  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot0021  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0022* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,1)
Boot0023* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,0)
Boot0024* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0025* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0026* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```
```shell
$ efivar -l | grep fw
NONE
```
```shell
$ tree /boot
Permissions  Size User Date Modified Name
drwxr-xr-x@     - root 31 Dec  1969  /boot/
.rwxr-xr-x@ 132Ki root 13 Jun 17:31  ├── config-5.1.9-gentoo*
.rwxr-xr-x@ 132Ki root 16 Jun 17:32  ├── config-5.1.10-gentoo*
drwxr-xr-x@     - root 24 Apr 22:42  ├── EFI/
drwxr-xr-x@     - root  2 Apr 17:01  │  ├── Boot/
.rwxr-xr-x@  59Ki root 12 Feb  1:47  │  ├── fwupdx64.efi*
drwxr-xr-x@     - root 14 May 15:23  │  ├── gentoo/
drwxr-xr-x@     - root 19 May 20:18  │  └── Microsoft/
drwxr-xr-x@     - root 16 Jun 17:40  ├── grub/
drwxr-xr-x@     - root 15 Feb  4:15  │  ├── fonts/
.rwxr-xr-x@ 7.5Ki root 16 Jun 17:40  │  ├── grub.cfg*
.rwxr-xr-x@ 1.0Ki root  2 Feb  5:33  │  ├── grubenv*
drwxr-xr-x@     - root  2 Apr 17:59  │  ├── locale/
drwxr-xr-x@     - root  2 Feb  5:33  │  ├── themes/
drwxr-xr-x@     - root 14 May 15:44  │  └── x86_64-efi/
.rwxr-xr-x@  87Mi root 13 Jun 17:34  ├── initramfs-5.1.9-gentoo.img*
.rwxr-xr-x@  87Mi root 16 Jun 17:40  ├── initramfs-5.1.10-gentoo.img*
.rwxr-xr-x@ 482Ki root 10 Jun 18:28  ├── intel-uc.img*
.rwxr-xr-x@ 3.8Mi root 13 Jun 17:31  ├── System.map-5.1.9-gentoo*
.rwxr-xr-x@ 3.8Mi root 16 Jun 17:32  ├── System.map-5.1.10-gentoo*
.rwxr-xr-x@ 9.8Mi root 13 Jun 17:31  ├── vmlinuz-5.1.9-gentoo*
.rwxr-xr-x@ 9.8Mi root 16 Jun 17:32  └── vmlinuz-5.1.10-gentoo*
```
- Operating system and version: Gentoo, ~amd64 keyworded
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Emerge (From source)
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

lovesegfault, 2019-06-17:
Additional info:
```shell
$ systemctl status fwupd
● fwupd.service - Firmware update daemon
   Loaded: loaded (/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: active (running) since Sun 2019-06-16 19:34:35 PDT; 11min ago
     Docs: https://fwupd.org/
 Main PID: 16715 (fwupd)
    Tasks: 5 (limit: 4915)
   Memory: 21.9M
   CGroup: /system.slice/fwupd.service
           └─16715 /usr/libexec/fwupd/fwupd

Jun 16 19:34:22 bergman systemd[1]: Starting Firmware update daemon...
Jun 16 19:34:22 bergman fwupd[16715]: 02:34:22:0516 FuPluginUefi         Failed to get PCR0s: missing executable tpm2_pcrlist in PATH
Jun 16 19:34:35 bergman systemd[1]: Started Firmware update daemon.
```

francescor, 2019-07-05:
I had the same issue (Thinkpad t480s), and I solved it by changing these options in the BIOS setup:
* Boot Order Lock: from Enabled to **Disabled**
* Option Key Display: from Disabled to **Enabled**
* Boot Device List F12 Option: from Disabled to **Enabled**
(but I guess the issue was related to the first one)

eworm-de, 2019-07-08:
The last message is caused by Logitech Unifying Receiver.

kennylevinsen, 2019-07-12:
See https://github.com/hughsie/fwupd/issues/1244.

closed 2019-07-15 by hughsie