fwupd/fwupd #1230: Firmware version incompatible to run program

daya, 2019-06-25:
Issue #830 #893 were closed without validating the assumption. 

Not everyone has MS Windows handy therefore the workaround described in #830 is not going to work. We need a workaround that works for linux machines stuck on version 0.1.25

JohnAZoidberg, 2019-07-02:
It seems that it's not the fault of `fwupd` but that Lenovo hasn't uploaded the necessary updates to take the firmware from specific old versions.
The solution is to hope that vendors upload all FW updates to LVFS.

jevonearth, 2019-07-17:
Is there a contact, or someone to alert at Lenovo? 
My wifi/bluetooth card is not working, and Lenovo support says I need to update my firmware. Installing Windows to do this is not an option for me.

duritong, 2019-08-28:
You can download the bootable update iso and then put it on a usb stick.

E.g. 
```
sudo geteltorito -o /dev/sdb /tmp/n23ur17w.iso 
```

Booting then from that usb stiick allows to update.

joakim-tjernlund, 2019-08-30:
Would be great I one could make bootable USB stick for Dock updates too.
Currently one must run a Windows installer for that.