fwupd/fwupd #1234: Trying to update thinkpad docking station while not connected

pschichtel, 2019-07-08:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
T480 Thunderbolt Controller
  DeviceId:             2540a65b99cd5854a4a1a6c27e8a7d6626e5a798
  Guid:                 fe328c2d-5300-597b-a21d-991116db8ab6
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-07-08

20L5CTO1WW System Firmware
  DeviceId:             48477ba14bd9bd0c97323a07caa54b7766f917c2
  Guid:                 dc7f0308-1ef8-4774-9ba5-89a58c4d731c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.24
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-07-08

UEFI Device Firmware
  DeviceId:             5decfff65e3391259bca4003a090e59bf50e3bea
  Guid:                 dd31d983-cf3b-4c84-acc9-70caa94e827b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.65.3590
  VersionLowest:        184.65.3590
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-07-08

UEFI Device Firmware
  DeviceId:             c19580db344e418cf2b23297bcf46d75c06e432a
  Guid:                 b87a926d-189c-49a1-b1ff-921099b1de89
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.14
  VersionLowest:        0.1.14
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-07-08

UEFI Device Firmware
  DeviceId:             27314a74b0935338fbf788ab05043e745a318147
  Guid:                 2296f4c0-13a0-48f0-b5cb-83e7a1b868a2
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              2.1.9
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-07-08

UEFI Device Firmware
  DeviceId:             2ea1b362db45581c8be7dc1094a973e2a0b45fec
  Guid:                 815e2465-da45-4b49-93c3-1517e649e1e6
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              1.4.1084
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-07-08

WDC PC SN720 SDAQNTW-512G-1001
  DeviceId:             f2759da7fe8e0388c5f3601cb072f837b1070b03
  Guid:                 ff2112dc-038c-596d-90ca-d43c5077c6ec
  Guid:                 63af1441-fb24-5368-b450-886fbc98610b
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Sandisk Corp
  VendorId:             NVME:0x15B7
  Version:              10160101
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-07-08
```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,001D,001E,001F,0024,0001
Boot0000* Linux EFISTUB	HD(2,GPT,278d0f35-ebb7-4916-86d8-a8bd9363882f,0x3b9b000,0x1dc800)/File(\vmlinuz-linux)c.r.y.p.t.d.e.v.i.c.e.=.U.U.I.D.=.".a.7.3.f.5.0.b.2.-.4.5.9.c.-.4.4.b.0.-.a.7.f.3.-.c.e.e.6.d.6.0.3.9.5.f.c.".:.c.r.y.p.t.r.o.o.t. .r.o.o.t.=./.d.e.v./.m.a.p.p.e.r./.c.r.y.p.t.r.o.o.t. .r.w. .i.n.i.t.r.d.=.\.i.n.t.e.l.-.u.c.o.d.e...i.m.g. .i.n.i.t.r.d.=.\.i.n.i.t.r.a.m.f.s.-.l.i.n.u.x...i.m.g.
Boot0001* Linux-Firmware-Updater	HD(2,GPT,278d0f35-ebb7-4916-86d8-a8bd9363882f,0x3b9b000,0x1dc800)/File(\EFI\arch\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001A* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
```
(empty output)

```shell
$ tree /boot
/boot
├── EFI
│   └── arch
│       ├── fw
│       └── fwupdx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── install_entry.sh
├── intel-ucode.img
└── vmlinuz-linux
```

Please answer the following questions:

- Operating system and version: Archlinux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

# The problem
All devices updated without problems, even while not connected to power, but one device (guid: **b87a926d-189c-49a1-b1ff-921099b1de89**) kept getting updates each time I checked for updates, always from 0.1.13 to 0.1.14 even though the laptop (**Thinkpad T480**) rebooted and installed the update seemingly successfully. I didn't know what the device until a friend told me in a different context that thinkpad docking stations get updated during the laptop's firmware update process, if the laptop is connected to the dock. So I tried running `fwupd update` today while the laptop was in the dock and the update actually got installed. 

So I see three problems here:
1. The device should not have been updatable while not connected
2. The device is marked is internal which it isn't
3. There was no proper feedback, that the installation did not succeed.

joakim-tjernlund, 2019-08-30:
As far as I know Lenovo does not update Docks 
I got an T480s and I can a FW named:
```
UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.16
  VersionLowest:        0.1.16
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-08-30
```
Is that for the dock? I don't think so.
Docks can only be updated from Windows since their installer is just an exe

pschichtel, 2019-08-30:
Possible, but why did it only successfully install when connected to the dock? Seems at least related to me.

hughsie, 2019-08-30:
I don't think that's a dock...

superm1, 2019-08-30:
> I don't think that's a dock...

Just a thought; stuff like this is it a common vendor ESRT GUID across lots of systems that take the same payload?  If so, maybe it's worth adding in the `uefi.quirks` file a more proper name than "UEFI Device Firmware"

hughsie, 2019-08-30:
100% agree. The tricky bit is finding a list of guids..

superm1, 2019-08-30:
> supported

Maybe check in the CAB file that matches this device and pull them from that?  Or ask vendor?