fwupd/fwupd #1237: Dependencies missing...

kitingChris, 2019-07-11:
```
|Compiler for C supports link arguments -Wl,-z,relro: YES
|Compiler for C supports link arguments -Wl,-z,now: YES
|Dependency gio-2.0 found: YES (cached)
|Native dependency uuid found: YES 2.31.1
|Checking for function "uuid_generate_sha1": YES
|Configuring config.h using configuration
|Configuring xb-version.h using configuration
Couldn't use fallback subproject in subprojects/libxmlb for the dependency xmlb 
Reason: gobject-introspection dependency was not found, gir cannot be generated.

meson.build:158:0: ERROR:  Native dependency 'xmlb' not found
```

Please update the docs what dependencies are required to run a build.

hughsie, 2019-07-11:
The error message is telling you what you need to install.

closed 2019-07-11 by superm1

kitingChris, 2019-07-11:
I posted it in frustration after about 20 or more tries all having to search for the package and installing it manually. 
Dependencies were not automatically installed and had to be installed one after another. 
Quite annoying.
I quit it and installed the snap instead although I totally dislike snaps. 
System is Ubuntu 18.10

superm1, 2019-07-11:
At least on Ubuntu the easiest way to solve dependencies is:

```sudo apt build-dep fwupd```

kitingChris, 2019-07-11:
> At least on Ubuntu the easiest way to solve dependencies is:
> 
> `sudo apt build-dep fwupd`

mh didn't yet recognized this apt option. Gotta try this tomorrow.

But perhaps it might be a good point to include it in the documentation? ;-)

superm1, 2019-07-11:
Sure, I just added it to the compilation wiki

AndrewBarfield, 2019-08-07:
drew@theater-nuc:~/Projects/fwupd$ sudo apt build-dep fwupd
Reading package lists... Done
E: You must put some 'source' URIs in your sources.list

I have 90-some URIs in my source lists

superm1, 2019-08-07:
Do you have deb-src lines?