fwupd/fwupd #1239: Flashrom plugin is fairly verbose

superm1, 2019-07-11:
With current master I am noticing quite a large number of messages being emitted by the flashrom plugin on a machine which doesn't support flashrom.  I think this is the more common case right now, so these probably should be quieted.

```
# sudo /usr/local/libexec/fwupd/fwupdtool get-devices --show-all-devices 
Loading…                 [-                                      ]13:17:00:0328 FuPluginFlashrom     ========================================================================

13:17:00:0328 FuPluginFlashrom     WARNING! You seem to be running flashrom on an unsupported laptop.

13:17:00:0328 FuPluginFlashrom     Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================

13:17:00:0328 FuPluginFlashrom     Aborting.

Loading…                 [***************************************]
<snip>
```

superm1, 2019-07-11:
CC @artur-rs

superm1, 2019-07-11:
The potential approaches that come to mind to me:
* Lower that output to debug instead of warning
* Configure flashrom to run only on whitelisted machines via quirks
* Configure flashrom to run after `uefi` and only if no real uefi devices were created

hughsie, 2019-07-11:
> Configure flashrom to run only on whitelisted machines via quirks

That's what I thought was being done here... Perhaps everything in fu_plugin_startup() should be moved to the last section in fu_plugin_coldplug and executed only if a device is actually added?

superm1, 2019-07-11:
Here's a more verbose log showing where it comes from:
```
17:37:18:0396 FuPlugin             performing startup() on flashrom
17:37:18:0396 FuPluginFlashrom     Initializing internal programmer

17:37:18:0396 FuPluginFlashrom     /sys/class/mtd/mtd0 does not exist

17:37:18:0397 FuPluginFlashrom     No coreboot table found.

17:37:18:0397 FuPluginFlashrom     Using Internal DMI decoder.

17:37:18:0397 FuPluginFlashrom     DMI string chassis-type: "Notebook"

17:37:18:0397 FuPluginFlashrom     Laptop detected via DMI.

17:37:18:0397 FuPluginFlashrom     DMI string system-manufacturer: "Dell Inc."

17:37:18:0397 FuPluginFlashrom     DMI string system-product-name: "XPS 13 9380"

17:37:18:0397 FuPluginFlashrom     DMI string system-version: "Not Specified"

17:37:18:0397 FuPluginFlashrom     DMI string baseboard-manufacturer: "Dell Inc."

17:37:18:0397 FuPluginFlashrom     DMI string baseboard-product-name: "      "

17:37:18:0397 FuPluginFlashrom     DMI string baseboard-version: "   "

17:37:18:0398 FuPluginFlashrom     ========================================================================

17:37:18:0398 FuPluginFlashrom     WARNING! You seem to be running flashrom on an unsupported laptop.

17:37:18:0398 FuPluginFlashrom     Laptops, notebooks and netbooks are difficult to support and we
recommend to use the vendor flashing utility. The embedded controller
(EC) in these machines often interacts badly with flashing.
See the manpage and https://flashrom.org/Laptops for details.

If flash is shared with the EC, erase is guaranteed to brick your laptop
and write may brick your laptop.
Read and probe may irritate your EC and cause fan failure, backlight
failure and sudden poweroff.
You have been warned.
========================================================================

17:37:18:0398 FuPluginFlashrom     Aborting.

17:37:18:0398 FuEngine             disabling plugin because: failed to startup using flashrom: programmer initialization failed
```

So I guess this is all correct in the error handling; this is a warning that is output in the "Warning" class from `fu_plugin_flashrom_debug_cb`.  I suppose this needs to be downgraded in the flashrom library to debug that means if it's not moved to coldplug.

closed 2019-07-12 by superm1