fwupd/fwupd #1244: U0012 Unifying receiver incorrectly recognized despite no support

kennylevinsen, 2019-07-12:
### Issue

U0012 is recognized by the unifying plugin. However, the plugin (unfortunately) only supports U0009 and U0008.

This leads to a failed predicate on a bootloader version check, which while safe is not what we want.

### Reproduction

1. Plug in U0012 receiver
2. Use fwupdmgr commands such as get-updates.

### Expected results

Either show the device and support updating it, or ignore it/mark it as unsupported.

### Actual results

```
$ fwupdmgr get-releases
Choose a device:
0.	Cancel
1.	[ removed ]
2.	[ removed ]
3.	[ removed ] (Unifying Receiver)
3
No releases found for device: Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.03_B0009]

$ fwupdmgr get-updates
No upgrades for [ removed ]
No upgrades for [ removed ]
No releases found for device: Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.03_B0009]
```

### Conclusion

We need to ensure that the unifying correctly identifies supported devices. Optionally, we should support upgrading U0012, which comes with the MX Anywhere 2S.

### Template stuff

Version:

```shell
$ fwupdmgr --version
client version:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9
```

Please answer the following questions:

- Operating system and version: Arch Linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

### I see this error, and I have a unifying device! Is this the same issue?

Take out your receiver and see if it has anything other than U0007 or U0008 written on it. If yes, then this is the issue. However, if your device is *not* a U0012, please note the device in the comments so we know that that device is also misidentified.

hughsie, 2019-07-15:
> However, the plugin (unfortunately) only supports U0009 and U0008

It should support U0012 too...

hughsie, 2019-07-15:
Okay, this slightly bad error message is actually correct; fwupd is going through the list of firmware that can be applied to this device, and only finds the non signed firmware version which it correctly refuses to install. The "signed" firmware that is designed for these later bootloader device versions isn't "stable" on the LVFS. @xnlcasad -- should that firmware be pushed now?

xnlcasad, 2019-07-15:
I'm not sure why this is an error. If there is no firmware update available for these signed receivers they should not be updated. Maybe the error message is not appropriate.

kennylevinsen, 2019-07-15:
> It should support U0012 too...

In that case the readme for the plugin is out of date, as it specifically says U0007 and U0008. That's an easy fix, though. Any others' that we also support that should be added to the plugin readme?

However, that there is no signed and stable firmware available definitely does not appear to be an error scenario: There just isn't a new applicable firmware available for the device.

However, as a naïve end-user, it seems weird that we try to do anything with an unsigned firmware at all. I would assume that unsigned or unstable firmwares would be ignored entirely.

hughsie, 2019-07-19:
> Maybe the error message is not appropriate

Totally agreed. It should be in the daemon log for sure, but not returned to the user. I'll have a think on how to do this properly.

zagota, 2019-07-20:
Got the problem with Unifying U0008.

```
$ fwupdmgr get-releases
Choose a device:
0.	Cancel
1.	79e57a8e2475161cd68ac9d16df74235b996195a (XPS 13 9370 Thunderbolt Controller)
2.	8a21cacfb0a8d2b30c5ee9290eb71db021619f8b (XPS 13 9370 System Firmware)
3.	5d937fc08eb7457a70a7d6cc6ca526ca43c13200 (Unifying Receiver)
3
No releases found for device: Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]

$ fwupdmgr get-updates
No releases found for device: Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]
No upgrades for XPS 13 9370 Thunderbolt Controller, current is 33.00: 33.00=same, 28.00=older
No upgrades for XPS 13 9370 System Firmware, current is 0.1.10.0: 0.1.10.0=same, 0.1.9.0=older, 0.1.8.1=older, 0.1.6.3=older, 0.1.5.1=older
No releases found for device: Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]

fwupdmgr --version
client version:	1.2.10
checkout info:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```
- Ubuntu 18.04.2 LTS
- Install with snap
- rebooting? Yes
- NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

mame82, 2019-07-26:
Just for info:

I managed to install RQR24.07.B0030 (signed) on CU0008 with BOT03.02 and CU0012 with BOT03.03.

I used a custom tool (not published yet) but it replicates exact steps of fwupd, when it comes to signed fw flashing (write 256 byte signature provided with firmware, before doing CRC check).

Both dongles run without issues, using this firmware.

Daleang, 2019-09-26:
Still having probelems: 
No releases found for device: Not compatible with org.freedesktop.fwupd version 1.2.10, requires >= 1.3.1, Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.03_B0009]

FFY00, 2019-09-26:
This just means there's no compatible firmware available, we need to fix the message.

superm1, 2019-10-24:
So maybe same thing as https://github.com/fwupd/fwupd/issues/1261

FFY00, 2019-10-24:
Looks like it, yes.

superm1, 2019-10-24:
duplicate of #1261

closed 2019-10-24 by superm1