fwupd/fwupd #1245: Cannot upgrade BIOS on XPS 13 9350

guibar, 2019-07-14:
Hello,
I am running ubuntu 19.04 on a DELL XPS 13 9350. 
When I run:

$fwupdmgr update

I get:

No upgrades for XPS 13 9350 Thunderbolt Controller, current is 16.00: 16.00=same
ignoring Dell UEFI updates [8330a096d9f1af8567c7374cb8403e1ce9cf3163] as not updatable

There is indeed a bios update available as I am running version 1.6.1 and 3 more recent versions exist, the latest being 1.10.1:
https://www.dell.com/support/home/uk/en/ukdhs1/drivers/driversdetails?driverid=rdnw7&oscode=wt64a&productcode=xps-13-9350-laptop

However, I have seen that there was a mess-up from DELL on version 1.6.1 and many people report being stuck on that version and incapable of further upgrade. See details here:
https://www.dell.com/community/XPS/XPS-13-9350-quot-BIOS-update-blocked-due-to-unsupported/td-p/5708012/page/8

I have tried an upgrade via the USB key route and indeed nothing happens, the computer just reboots and the version of the BIOS remains the same.

My question is: is the "not updatable" a reflection of this very problem?  
Or is it something else that could be fixed with a newer version of fwupd which  could then open a way of upgrading out of that 1.6.1 version? 

I am reporting the command outputs required below.

Thanks a lot!

`$ fwupdmgr --version
``client version:	1.2.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.5

$ fwupdmgr get-devices
XPS 13 9350 Thunderbolt Controller
  DeviceId:             b2232548c696538abf92956fcdcfe7831886cf7a
  Guid:                 dbbe0ed9-df91-529c-ac86-8976b78f726d
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              16.00
  Icon:                 computer
  Created:              2019-07-14

Dell UEFI updates
  DeviceId:             8330a096d9f1af8567c7374cb8403e1ce9cf3163
  Guid:                 2d47f29b-83a2-4f31-a2e8-63474f4d4c2e
  Summary:              Enable UEFI Update Functionality
  Plugin:               dell_esrt
  Flags:                locked|supported|registered
  Version:              0
  Icon:                 computer
  Created:              2019-07-14
  
  [Release]
  AppstreamId:          org.fwupd.8330a096d9f1af8567c7374cb8403e1ce9cf3163.device
  RemoteId:             fwupd
  Summary:              Enable UEFI Update Functionality
  Description:          <p>Applying this update will enable the UEFI firmware reporting interface on your hardware.</p><p>You will have to restart your computer after this update is installed to be notified of any pending firmware updates.</p>
  Version:              0.0.1
  TrustFlags:           none

PM951 NVMe SAMSUNG 512GB
  DeviceId:             e11623b2caa18fee292058a5c09ca4e6152f7ecf
  Guid:                 cce2386f-6d68-5f85-9f3f-ebead9288696
  Guid:                 77bb4022-9503-586c-99b7-c8a64502da30
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Guid:                 2b05215d-0ecf-5749-a584-012329fb33e6
[... 297 lines cut ...]
│       ├── video_fb.mod
│       ├── videoinfo.mod
│       ├── video.lst
│       ├── video.mod
│       ├── videotest_checksum.mod
│       ├── videotest.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-5.0.0-19-generic
├── initrd.img-5.0.0-20-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.0.0-19-generic
├── System.map-5.0.0-20-generic
├── vmlinuz-5.0.0-19-generic
└── vmlinuz-5.0.0-20-generic

6 directories, 288 files

superm1, 2019-07-14:
Try to use fwupdmgr unlock on the UEFI device and then reboot the machine.

Admittedly this messaging is very confusing. I think we might want to adjust how this works some.

guibar, 2019-07-14:
I have unlocked the UEFI device, rebooted and fwupdmgr get-updates reveals the firwmare update for version 1.10.1 as well as all the intermediary ones.

However, when I try to upgrade, a reboot takes place, then another one and the computer stays on with a black screen. I have to press the power button to get it out of if. When I reboot, the version of the BIOS hasn't changed.

I guess I am hitting the problem that is known with this version 1.6.1 of the BIOS. If you have any other idea of how to pass this hurdle, I'll be interested to hear about it. But I realize that this might be a problem which has nothing to do with fwupdmgr.

Thanks,

closed 2019-07-15 by superm1

superm1, 2019-07-15:
In that case it's not a fwupd issue; I would recommend raising it to Dell support.  Just glancing through that thread it seems some people had success with a downgrade and then upgrade past this version.

You can download the "Windows" update executables to a USB disk and select them at the BIOS POST screen by pressing F12 and selecting the option to update BIOS.

guibar, 2019-07-15:
Hi Mario,

Yes, I will try the downgrade/upgrade route to see if that works.

Thanks,

hiehoo, 2020-09-10:
> Hi Mario, Yes, I will try the downgrade/upgrade route to see if that works. Thanks,
> […](#)

Hi Guibar, 
I see your problem same as mine right now, have you solve this problem and if yes, how?

guibar, 2020-09-10:
Hi hiehoo,
No, I neved solved the problem. I gave up after a while. But it is annoying ...

hiehoo, 2020-09-10:
thank you, i did stuck with these problem for 2 days full time try to resolve it, it make me depressed, try all the way i can but nothing happened. Stucking with BIOS version 1.7.0, the xps cannot sleep probably, whenever i put the lid down, the xps gonna sleep for awhile and restarting, result to overheating and battery draining. Sad

commit 4bef9a53 (2019-07-14): dell-esrt: Improve the experience for the plugin (Closes: #1245)

* Set UpdateError to indicate how the device needs to be updated
* Set the device as needing reboot
* If admin password is set show the message to frontend via `UpdateError`
commit ac627cd8 (2019-07-15): dell-esrt: Improve the experience for the plugin (Closes: #1245)

* Set UpdateError to indicate how the device needs to be updated
* Set the device as needing reboot
* If admin password is set show the message to frontend via `UpdateError`