fwupd/fwupd #1249: Outside of CI flashrom should still default to false

superm1, 2019-07-15:
In thinking about prereqs for a release to me there are two thing about flashrom that should change before it's a default enabled in a fwupd release.

1. It should depend on a tagged release of flashrom or at least a tag that points to a commit rather than a branch
2. It should get merged into flashrom upstream or at least have a PR that's looking good for this.

Thoughts on this?

hughsie, 2019-07-15:
> It should get merged into flashrom upstream

What's the delta now? I think there was only a few things not upstream already.

> Thoughts on this?

I thought it was disabled by default; I think disable it now, prep a release today and then work out the story for default on for the next release.

superm1, 2019-07-15:
> What's the delta now? I think there was only a few things not upstream already.

I guess that part I don't know what's left.  I just noticed https://github.com/hughsie/fwupd/blob/master/subprojects/flashrom.wrap is pointing to a local branch.

I just thought it should be good policy to not push the plugin widely unless upstream flashrom is happy with the stuff it depends upon.

> I thought it was disabled by default; I think disable it now, prep a release today and then work out the story for default on for the next release.

I was seeing it still on in my local system tests (I was missing some deps which made me realize it):
https://github.com/hughsie/fwupd/blob/master/meson_options.txt#L21

hughsie, 2019-07-15:
I've pushed something that might work in https://github.com/hughsie/fwupd/pull/1250 -- the idea being that we still build it by default on a lot of the CI targets, but not by default if the user is using the defaults.

I have to switch to daddy mode, so if that fails I'd appreciate any help with the Debian bits. When that's merged I'll do a tarball and Fedora build for Andrew and the g-f-u CI.

superm1, 2019-07-15:
Looks like it's in good shape, thanks.

closed 2019-07-16 by superm1