fwupd/fwupd #1253: StarLabs: Enters EFI Shell after firmware applied

Stephen-StarLabs, 2019-07-17:
Issue: BIOS will fail to boot to OS and hang for approx 60 seconds on the splash screen, before entering the EFI Shell.

Cause: This ONLY occurs if there are is an EC firmware update and SSD firmware update applied at the same time. Upon running "fwupd update" this will download the firmware and then schedule for application upon next reboot. At next reboot, the EC and Storage firmware will apply sequentially and then reboot. At this point, the post cycle will hang on the splash screen. After 60 seconds, then enter the EFI Shell.

Resolution steps: A power-down of the hardware is required. First boot will result in entering the EFI Shell once again until 'EXIT' is entered in EFI Shell, then the machine will return to booting into the OS as previous to the firmware update.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.11
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.11
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
SuperIO IT8987
  DeviceId:             1f664ad149f2151fb750b52751dbf9ba045a7846
  Guid:                 0aa07091-dc02-5e62-a0cd-9d3880721b64
  Summary:              Embedded Controller
  Plugin:               superio
  Flags:                internal|updatable|only-offline|require-ac|supported|registered|needs-reboot
  Version:              03.08
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-07-17

LabTop System Firmware
  DeviceId:             5984fe62b2de5dfa0c7980660e230707d696c8dd
  Guid:                 8265d473-a6c2-42b4-897b-bc220faa2d32
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(107a02c13e7e52763ae5639638d4ab492a9beed7)
  Checksum:             SHA256(c5f07bb75975a550fa49a83a47ccb5ea3551b0496665b387a3d5b2e7b554b564)
  Version:              1.0.9
  VersionLowest:        1.0.9
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-07-17

Star Drive PCIe SSD
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 8cb1e2fe-eb01-5508-9fb3-98add4bb7c34
  Guid:                 a44eb54c-5441-56f2-8cc0-5e48964c6457
  Guid:                 12d61493-5ac4-57f7-a2c5-96878d115455
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|supported|registered|needs-shutdown
  Vendor:               Phison Electronics Corporation
  VendorId:             NVME:0x1987
  Version:              ECFM12.3
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-07-17
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Ubuntu 19.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): source
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-07-17:
This is a bit confusing to me.  Doesn't the SSD firmware update for this particular drive only take effect on shutdown?

Stephen-StarLabs, 2019-07-18:
Yes, that is correct.

hughsie, 2019-07-19:
> EC firmware update and SSD firmware update applied at the same time

Huh, confusing. The EC update is being done in userspace so I don't see how that would affect the pre-efi environment. The SSD is also installed at runtime, and then just "activated" at shutdown. I don't see either of them changing BootNext.

If you connect a hardware debugger to the main board, do you get any debugging about why the spash is "hanging"? I suppose it could be something to do with the drive power-off -- but that would happen just if the SSD update is applied rather than being affected by the superio update.

superm1, 2019-07-19:
To me this sounds like the firmware couldn't resolve the boot device and gave up.  Perhaps a timing problem with enumeration after the firmware update?  You need a debug BIOS and to dig through what's happening there and what code path is failing.

Also while you're at it; I would highly recommend you remove any codepaths that will allow direct access to EFI shell from your failure scenarios.  This type of failure fallback can be used to circumvent technologies like Secure Boot.

@hughsie I'm thinking what you mentioned in #1203 is probably the correct solution longer term solution from fwupd side.  It will implicitly decouple updates like this from actually happening together while also solving problems with immutable filesystems.

Stephen-StarLabs, 2019-08-14:
I can confirm that further build updates to fwupd seemed to have resolved the issue we were experiencing and this no longer occurs.

Thank you all for your time on this issue, much appreciated.

closed 2019-08-14 by hughsie

ipaqmaster, 2023-11-17:
I have also experenced this today after updating some firmware with `fwupd 1.9.8` on a physical Dell Optiplex 3080M which booted  straight to IPv6 PXE after the updates completed.

The the NVMe was also not booting when plugged into my PC and booted with QEMU, which goes straight to the EFI shell.

fwupd is doing something shift to the /boot/EFI section preventing the setup from being bootable after completion.

The update covered (On the physical Dell Optiplex 3080M):

`Upgrade PC SN530 NVMe WDC 256GB from 21112012 to 21113012`
`Upgrade System Firmware from 2.1.1 to 2.21.1`
`Upgrade UEFI dbx from 217 to 371`

Deleting `/boot/EFI/Arch/fw*` was not enough.

I ended up running `mv /boot/EFI /boot/EFI.old` then `bootctl install` to re-place the bootloader and from then on the SSD was bootable again without having to manually pick the bootloader from the disk list.