fwupd/fwupd #1254: Gnome Software centre unable to get updates due to fwupd timing out

HemilTheRebel, 2019-07-21:
Gnome software centre is unable to check for updates because of an error in fwupd, This is the dialog:
[image]

```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

```shell
$ fwupdmgr get-devices
Aspire E5-575 System Firmware
  DeviceId:             ac76812570158ef82b81c21880478f9dff0a1ab4
  Guid:                 12b99262-648c-4365-bafd-eb869fb71094
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Checksum:             SHA1(d9d6e93ae28cebf2fcfcee5c3768a58afb531881)
  Version:              85.57.277
  VersionLowest:        85.57.277
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-07-21

TOSHIBA MQ01ABD100
  DeviceId:             5dbeb140337f610d54913c0f43d15fdaa0eafb51
  Guid:                 66d0e772-2346-5c45-93ba-0211a64f5a57
  Guid:                 17628727-4832-5b2c-af2b-f408e0d99ff7
  Guid:                 539bd28f-704f-5906-81b1-08eef8cdd97f
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              AX1P5J
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-07-21
```
I don't know whats a UEFI plugin. But my laptop is running UEFI. So 
```shell
$ efibootmgr -v
Skipping unreadable variable "Boot0022": Interrupted system call
Skipping unreadable variable "Boot0023": Interrupted system call
Skipping unreadable variable "Boot0024": Interrupted system call
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0002,2001,2002,2003
Boot0000* Windows Boot Manager	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...,................
Boot0001* fedora	PciRoot(0x0)/Pci(0x17,0x0)/Sata(0,0,0)/HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\grubx64.efi)A01 ..
Boot0002* HDD0: TOSHIBA MQ01ABD100	PciRoot(0x0)/Pci(0x17,0x0)/Sata(0,0,0)/HD(5,GPT,a9e8f73f-8a76-461c-8653-ebc690adc131,0x18ce0800,0x64000)RC
Boot0003* Unknown Device: 	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shim.efi)RC
Boot0005* Unknown Device: 	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shim.efi)RC
Boot0006* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0007* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0008* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0009* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot000A* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot000B* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot000C* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot000D* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot000E* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot000F* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0010* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0011* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0012* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0013* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0014* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0015* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
Boot0016* Fedora	HD(2,GPT,9f00dcd3-41c2-480b-a0d3-c117c6098cb5,0xe1800,0x32000)/File(\EFI\fedora\shimx64.efi)
[... 102 lines cut ...]
├── initramfs-5.1.16-300.fc30.x86_64.img
├── initramfs-5.1.17-300.fc30.x86_64.img
├── initramfs-5.1.18-300.fc30.x86_64.img
├── loader
│   └── entries [error opening dir]
├── lost+found [error opening dir]
├── memtest86+-5.01
├── System.map-5.1.16-300.fc30.x86_64
├── System.map-5.1.17-300.fc30.x86_64
├── System.map-5.1.18-300.fc30.x86_64
├── vmlinuz-0-rescue-8b31c7688fdf4b9bbe00b7ee532e5a93
├── vmlinuz-5.1.16-300.fc30.x86_64
├── vmlinuz-5.1.17-300.fc30.x86_64
└── vmlinuz-5.1.18-300.fc30.x86_64
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Fedora 30 running kernel version 5.1.18-300
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): I didn't. I think it came with fedora 
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? No.
- Is secure boot enabled (only for the UEFI plugin)? Its not. But I have the option of turning it on.

hughsie, 2019-07-26:
What does `fwupdmgr get-updates` say? Is there any backtrace in the journal or in abrt?

HemilTheRebel, 2019-07-26:
`fwupdmgr get-updates ` returns nothing.
`systemctl status fwupd` returns this

```
● fwupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: active (running) since Fri 2019-07-26 19:49:20 IST; 7min ago
     Docs: https://fwupd.org/
 Main PID: 3060 (fwupd)
    Tasks: 5 (limit: 4543)
   Memory: 15.4M
   CGroup: /system.slice/fwupd.service
           └─3060 /usr/libexec/fwupd/fwupd

Jul 26 19:48:53 localhost.localdomain systemd[1]: Starting Firmware update daemon
Jul 26 19:49:20 localhost.localdomain systemd[1]: Started Firmware update daemon
lines 1-12/12 (END)
```

`journalctl -u fwupd -n 50 --no-pager` shows:
```
-- Logs begin at Sun 2019-03-17 20:48:03 IST, end at Fri 2019-07-26 20:05:54 IST. --
Jul 21 18:29:46 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 21 18:30:04 localhost.localdomain systemd[1]: Started Firmware update daemon.
Jul 21 18:56:01 localhost.localdomain fwupd[2827]: 13:26:01:0814 GLib                 g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Jul 21 18:56:01 localhost.localdomain systemd[1]: Stopping Firmware update daemon...
Jul 21 18:56:01 localhost.localdomain systemd[1]: fwupd.service: Succeeded.
Jul 21 18:56:01 localhost.localdomain systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Jul 21 19:02:50 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 21 19:03:13 localhost.localdomain systemd[1]: Started Firmware update daemon.
Jul 21 19:25:21 localhost.localdomain fwupd[2827]: 13:55:21:0069 GLib                 g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Jul 21 19:25:21 localhost.localdomain systemd[1]: Stopping Firmware update daemon...
Jul 21 19:25:21 localhost.localdomain systemd[1]: fwupd.service: Succeeded.
Jul 21 19:25:21 localhost.localdomain systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Jul 21 22:13:28 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 21 22:14:12 localhost.localdomain systemd[1]: Started Firmware update daemon.
Jul 21 22:42:05 localhost.localdomain systemd[1]: Stopping Firmware update daemon...
Jul 21 22:42:05 localhost.localdomain fwupd[2795]: 17:12:05:0759 GLib                 g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Jul 21 22:42:07 localhost.localdomain systemd[1]: fwupd.service: Succeeded.
Jul 21 22:42:07 localhost.localdomain systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Jul 22 06:55:46 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 22 06:56:06 localhost.localdomain systemd[1]: Started Firmware update daemon.
Jul 22 07:13:01 localhost.localdomain fwupd[2843]: 01:43:01:0439 GLib                 g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Jul 22 07:13:01 localhost.localdomain systemd[1]: Stopping Firmware update daemon...
Jul 22 07:13:01 localhost.localdomain systemd[1]: fwupd.service: Succeeded.
Jul 22 07:13:01 localhost.localdomain systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Jul 22 08:11:06 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 22 08:11:31 localhost.localdomain systemd[1]: Started Firmware update daemon.
Jul 22 09:39:06 localhost.localdomain fwupd[2903]: 04:09:05:0998 GLib                 g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Jul 22 09:39:05 localhost.localdomain systemd[1]: Stopping Firmware update daemon...
Jul 22 09:39:06 localhost.localdomain systemd[1]: fwupd.service: Succeeded.
Jul 22 09:39:06 localhost.localdomain systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Jul 23 07:09:05 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 23 07:09:31 localhost.localdomain systemd[1]: Started Firmware update daemon.
Jul 23 07:15:56 localhost.localdomain systemd[1]: Stopping Firmware update daemon...
Jul 23 07:15:58 localhost.localdomain systemd[1]: fwupd.service: Succeeded.
Jul 23 07:15:58 localhost.localdomain systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Jul 24 06:43:03 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 24 06:43:31 localhost.localdomain systemd[1]: Started Firmware update daemon.
Jul 24 08:40:57 localhost.localdomain systemd[1]: Stopping Firmware update daemon...
Jul 24 08:41:14 localhost.localdomain fwupd[2889]: 03:11:14:0200 GLib                 g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Jul 24 08:41:23 localhost.localdomain systemd[1]: fwupd.service: Succeeded.
Jul 24 08:41:23 localhost.localdomain systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Jul 26 07:27:57 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 26 07:28:15 localhost.localdomain fwupd[2859]: 01:58:15:0447 FuPluginUefi         Failed to get PCR0s: Operation was cancelled
Jul 26 07:28:15 localhost.localdomain systemd[1]: Started Firmware update daemon.
Jul 26 07:47:07 localhost.localdomain systemd[1]: Stopping Firmware update daemon...
Jul 26 07:47:07 localhost.localdomain fwupd[2859]: 02:17:07:0345 GLib                 g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Jul 26 07:47:09 localhost.localdomain systemd[1]: fwupd.service: Succeeded.
Jul 26 07:47:09 localhost.localdomain systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Jul 26 19:48:53 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Jul 26 19:49:20 localhost.localdomain systemd[1]: Started Firmware update daemon.
```

hughsie, 2019-07-28:
Is there a backtrace?

HemilTheRebel, 2019-07-29:
how to get one? @hughsie

hughsie, 2019-07-29:
> how to get one?

On Fedora, it should already be captured by abrt.

HemilTheRebel, 2019-08-01:
The software centre suddenly starts to work. Maybe it was a bug on their side

HemilTheRebel, 2019-08-01:
I get the error again! I don't know how to use abrt but `abrt-cli list` has no mention of fwupd failing

HemilTheRebel, 2019-08-01:
I should add, before when I got this error in software centre, it closed. Now, when I got this error, it resumed its normal operation and started looking for updates. 

So yeah, its probably a bug in the GNOME software centre i think

superm1, 2019-08-01:
Can you show output of 

```
systemctl status fwupd.service
```

HemilTheRebel, 2019-08-01:
```
● fwupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: active (running) since Thu 2019-08-01 08:22:49 IST; 11min ago
     Docs: https://fwupd.org/
 Main PID: 4903 (fwupd)
    Tasks: 5 (limit: 4540)
   Memory: 19.0M
   CGroup: /system.slice/fwupd.service
           └─4903 /usr/libexec/fwupd/fwupd

Aug 01 08:22:20 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Aug 01 08:22:49 localhost.localdomain systemd[1]: Started Firmware update daemon.

```

superm1, 2019-08-01:
Well the systemd status looks good. Next time this occurs can you please repeat that again and see if there is anything different?

HemilTheRebel, 2019-08-01:
This happened again when I turned on my computer. Exactly same output.

```
● fwupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: active (running) since Thu 2019-08-01 20:21:59 IST; 1min 54s ago
     Docs: https://fwupd.org/
 Main PID: 2781 (fwupd)
    Tasks: 5 (limit: 4540)
   Memory: 15.4M
   CGroup: /system.slice/fwupd.service
           └─2781 /usr/libexec/fwupd/fwupd

Aug 01 20:21:25 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Aug 01 20:21:59 localhost.localdomain systemd[1]: Started Firmware update daemon.
```

superm1, 2019-08-01:
I wonder if it's fwupd starting up super slowly on your system causing it then?

HemilTheRebel, 2019-08-01:
Maybe. Is there a way to measure that?

HemilTheRebel, 2019-08-05:
fwupd seems to be partially broken on my device. When I enter `fwupdmgr --show-all-devices` it shows command not found. Even `fwupdmgr -v` shows command not found. And then it lists the options which include them

HemilTheRebel, 2019-08-25:
Any updates? Software Centre is still not working though I am able to get updates through dnf

hughsie, 2019-09-24:
> it shows command not found

I think something is broken on your install if that command can't be found. Try reinstalling fwupd.

HemilTheRebel, 2019-09-24:
I did a fresh install and the bug is gone.

closed 2019-09-24 by superm1