fwupd/fwupd #1255: snap: 'Failed unmounting Mount unit' on shutdown

LinuxOnTheDesktop, 2019-07-27:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.10
checkout info:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

```shell
$ fwupdmgr get-devices
X1 Carbon Thunderbolt Controller
  DeviceId:             eded407096031a81370bc25ae01650697a7c72b6
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              30.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-07-27

20KHCTO1WW System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.38
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-07-27

UEFI Device Firmware
  DeviceId:             29749712e93b58730ecd1b8537a5503ff5b7fb27
  Guid:                 9c9d9769-32fa-4841-b550-ea998e754e99
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.65.3590
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-07-27

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.13
  VersionLowest:        0.1.13
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-07-27

LENSE20512GMSP34MEAT2TA
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
  Guid:                 ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  VendorId:             NVME:0x17AA
  Version:              2.8.8341
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-07-27

Unifying Receiver
  DeviceId:             fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.06_B0030
  VersionBootloader:    BOT03.01_B0008
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-07-27
```

Please answer the following questions:

- Operating system and version: Mint 19.1 Cinnamon
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

On shutdown of the PC, I see (always, I think) this:

`Failed unmounting Mount unit for fwupd`

That problem does not seem to slow shutdown but I would rather not see the message.

```

Here are two lots of further information.

 $  snap list
Name   Version    Rev   Tracking  Publisher       Notes
core   16-2.39.3  7270  stable    canonical✓      core
fwupd  1.2.10     880   stable    richard-hughes  classic

```

```
$  neofetch --off [edited]
OS: Linux Mint 19.1 Tessa x86_64 
Host: 20KHCTO1WW ThinkPad X1 Carbon 6th 
Kernel: 5.2.3-050203-generic 
Uptime: 6 mins 
Packages: 3158 (dpkg), 6 (flatpak), 3 (snap) 
Shell: bash 4.4.20 
Resolution: 2560x1440 @ 60.01Hz 
DE: Cinnamon 4.2.3 
WM: Mutter (Muffin) 
WM Theme: Mint-Y-Dark-trans (Mint-Y-Dark-BB_hack) 
Theme: Mint-Y-Dark [GTK2/3] 
Icons: Paper [GTK2/3] 
Terminal: xfce4-terminal 
Terminal Font: InconsolataGo 12 
CPU: Intel i7-8550U (8) @ 4.000GHz 
GPU: Intel UHD Graphics 620 
Memory: 2320MiB / 15927MiB 
Disk (/dev/nvme0n1p4): 17G / 40G (44%) 
Disk (/dev/nvme0n1p2): 185G / 405G (48%) 
Disk (udev): 0 / 7.8G (0%) 

```

hughsie, 2019-07-28:
@superm1 any ideas?

superm1, 2019-07-29:
I'm not really sure.  Would you mind bringing this to the snap forums to ask?

LinuxOnTheDesktop, 2019-07-29:
@superm1 

Sorry, do you mean me? If you do, what (/where) are these forums, please?

superm1, 2019-07-29:
@LinuxOnTheDesktop here: https://forum.snapcraft.io/

LinuxOnTheDesktop, 2019-07-30:
[Done](https://forum.snapcraft.io/t/fireware-update-manager-snap-misbehaving-wont-unmount-at-shutdown/12518). If I get any useful response, I'll mention it (or link it) here.

LinuxOnTheDesktop, 2019-08-01:
This problem [seems to be snap-wide](https://github.com/systemd/systemd/issues/6350). So, closing.

LinuxOnTheDesktop, 2019-09-13:
Perhaps the following points are worth adding.

1. Installing a new kernel seems to cause the error.
2. I cease to see the error message after reinstalling snap (`sudo apt instal snapd --reinstall`).
3. After reinstalling snap, these error messages are generated (only once?):

[image]

LinuxOnTheDesktop, 2019-09-14:
I think my problems with snaps unmounting, and the error messages that appeared after reinstalling snap, owed to a messed up grub / uefi / secure boot situation on my machine.

LinuxOnTheDesktop, 2019-11-22:
However, I do see the attached, at shutdown, fairly frequently.
[image]

LinuxOnTheDesktop, 2019-12-11:
I see the above behaviour, now, on most shutdowns/reboots. Should I close this issue and open a new one - since the 'stop job'  problem is not the 'failed unmounting' problem?

superm1, 2019-12-11:
Seems like a different problem to me.  I'd open a new issue, close this one and start with looking at the `fwupd-activate` journal.

closed 2019-12-11 by superm1