fwupd/fwupd #1256: fwupd 1.2.5-2 faults at startup on Buster

tgunr, 2019-07-27:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
**[pve]/home/root (524): fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37**
```
Hangs for a minute then
```
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
[pve]/home/root (525):
Message from syslogd@pve at Jul 27 08:18:27 ...
 kernel:[152880.426962] watchdog: BUG: soft lockup - CPU#18 stuck for 22s! [fwupd:14875]

Message from syslogd@pve at Jul 27 08:18:55 ...
 kernel:[152908.426961] watchdog: BUG: soft lockup - CPU#18 stuck for 23s! [fwupd:14875]

Message from syslogd@pve at Jul 27 08:19:31 ...
 kernel:[152944.426961] watchdog: BUG: soft lockup - CPU#18 stuck for 22s! [fwupd:14875]

Message from syslogd@pve at Jul 27 08:19:59 ...
 kernel:[152972.426961] watchdog: BUG: soft lockup - CPU#18 stuck for 22s! [fwupd:14875]

Message from syslogd@pve at Jul 27 08:20:35 ...
 kernel:[153008.426961] watchdog: BUG: soft lockup - CPU#18 stuck for 23s! [fwupd:14875]
packet_write_poll: Connection to 10.1.1.1 port 22: Host is down
-bash: cd: /Users/davec/wwiiolng: No such file or directory
```
Then brought down the entire system.

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
**INSERT OUTPUT HERE**
```
Can't execute.

```shell
$ efibootmgr -v
**[pve]/home/root (504): efibootmgr -v
BootCurrent: 000B
Timeout: 1 seconds
BootOrder: 000B,0003,0010,0011,0012,000F
Boot0003* UEFI OS	HD(1,GPT,aceddbb5-a215-46b0-a756-23813e368b16,0x28,0x64000)/File(\EFI\BOOT\BOOTX64.EFI)..BO
Boot000B* debian	HD(2,GPT,cff2f619-4110-4770-ab6b-aee87e59e436,0x800,0x100000)/File(\EFI\DEBIAN\SHIMX64.EFI)
Boot000F* Hard Drive	BBS(HD,,0x0)..GO..NO........o.H.i.t.a.c.h.i. .H.D.T.7.2.1.0.1.0.S.L.A.3.6.0....................A...........................>..Gd-.;.A..MQ..L. . . . . . .T.S.6.F.4.0.R.M.4.2.N.9.P.G........BO..NO........o.H.i.t.a.c.h.i. .H.D.T.7.2.1.0.1.0.S.L.A.3.6.0....................A...........................>..Gd-.;.A..MQ..L. . . . . . .T.S.6.F.4.0.R.M.1.2.2.U.P.K........BO..NO........o.S.T.8.0.0.0.D.M.0.0.4.-.2.C.X.1.8.8....................A...........................>..Gd-.;.A..MQ..L. . . . . . . . . . . . .C.W.0.T.8.R.A.H........BO..NO........o.W.D.C. .W.D.8.0.E.F.A.X.-.6.8.L.H.P.N.0....................A...........................>..Gd-.;.A..MQ..L.S.7.9.G.2.E.C.Y. . . . . . . . . . . . ........BO..NO........o.W.D.C. .W.D.4.0.E.F.R.X.-.6.8.W.T.0.N.0....................A...........................>..Gd-.;.A..MQ..L. . . . .W. .-.D.C.W.4.C.5.E.Y.Z.E.E.6.4........BO..NO........o.S.T.4.0.0.0.D.M.0.0.0.-.1.F.2.1.6.8....................A...........................>..Gd-.;.A..MQ..L. . . . . . . . . . . . .3.Z.2.0.G.H.A.G........BO..NO........o.W.D.C. .W.D.3.0.E.Z.R.X.-.0.0.M.M.M.B.0....................A...........................>..Gd-.;.A..MQ..L. . . . .W. .-.D.C.W.W.A.2.Z.3.5.5.5.6.9........BO..NO........o.W.D.C. .W.D.3.0.E.Z.R.X.-.0.0.D.C.0.B.0....................A...........................>..Gd-.;.A..MQ..L. . . . .W. .-.D.C.W.1.C.0.T.6.4.9.8.5.0........BO..NO........s.I.N.T.E.L. .S.S.D.P.E.K.N.W.5.1.2.G.8....................A.......................................6..Gd-.;.A..MQ..L.B.T.N.H.8.3.6.5.1.9.G.6.5.1.2.A........BO..NO........s.I.N.T.E.L. .S.S.D.P.E.K.N.W.5.1.2.G.8....................A.......................................6..Gd-.;.A..MQ..L.B.T.N.H.8.3.6.4.0.V.Z.A.5.1.2.A........BO..NO..........A.0.S.0. .W.D.C. .W.D.2.0.E.Z.R.X.-.0.0.D.8.P.B.0....................A........................1.N........>.;......H..Gd-.;.A..MQ..L.A.0.S.0. .W.D.C. .W.D.2.0.E.Z.R.X.-.0.0.D.8.P.B.0........BO..NO..........A.0.S.1. .W.D.C. .W.D.3.0.E.Z.R.X.-.0.0.D.8.P.B.0....................A........................1.N........>.;......H..Gd-.;.A..MQ..L.A.0.S.1. .W.D.C. .W.D.3.0.E.Z.R.X.-.0.0.D.8.P.B.0........BO..NO........|.A.0.S.2. .S.T.3.5.0.0.6.3.0.A.S. .P....................A........................1.N........>.;......:..Gd-.;.A..MQ..L.A.0.S.2. .S.T.3.5.0.0.6.3.0.A.S. .P........BO..NO..........A.0.S.3. .H.I.T.A.C.H.I. .H.U.A.7.2.3.0.3.0.A.L.A.6.4.0....................A........................1.N........>.;......N..Gd-.;.A..MQ..L.A.0.S.3. .H.I.T.A.C.H.I. .H.U.A.7.2.3.0.3.0.A.L.A.6.4.0........BO
Boot0010* debian	HD(2,GPT,119728f3-6906-11e9-9726-947a9a5350e3,0x800,0x100000)/File(\EFI\DEBIAN\GRUBX64.EFI)..BO
Boot0011* debian	HD(2,GPT,c15517e8-2d2d-4da3-b481-fb0c058c23fb,0x800,0x100000)/File(\EFI\DEBIAN\GRUBX64.EFI)..BO
Boot0012* debian	HD(2,GPT,cff2f619-4110-4770-ab6b-aee87e59e436,0x800,0x100000)/File(\EFI\DEBIAN\GRUBX64.EFI)..BO**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**nada**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**[pve]/home/root (509): tree /boot
/boot
├── config-4.19.0-5-amd64
[... 599 lines cut ...]
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.19.0-5-amd64
├── initrd.img-4.9.0-9-amd64
├── System.map-4.19.0-5-amd64
├── System.map-4.9.0-9-amd64
├── vmlinuz-4.19.0-5-amd64
├── vmlinuz-4.9.0-9-amd64
├── xen-4.11-amd64.config
├── xen-4.11-amd64.efi
└── xen-4.11-amd64.gz**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Linux pve 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u1 (2019-07-19) x86_64 GNU/Linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt
- Have you tried rebooting? Had no choice, system rebooted itself.
- Are you using an NVMe disk? 2 NVMe disks present but not used or mounted at the time.
- Is secure boot enabled (only for the UEFI plugin)? No

tgunr, 2019-07-27:
After reboot
```
[pve]/home/root (512): fwupdmgr get-devices
To Be Filled By O.E.M. System Firmware
  DeviceId:             c41826fc0db09ea19d1c6d850198c7882c886074
  Guid:                 0acc309e-2709-4ecf-aeec-0c8fbff19a4c
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.0.113
  VersionLowest:        0.0.113
  Icon:                 computer
  Created:              2019-07-27

INTEL SSDPEKNW512G8
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 c6cd9ab0-8f20-512e-9e1f-1af55b8454b9 <- NVME\VEN_8086&DEV_F1A8&REV_03
  Guid:                 82741c78-f5dc-5c23-a152-00de5799edc8 <- NVME\VEN_8086&DEV_F1A8
  Guid:                 3f235826-4614-56f6-8fd4-b67b346882c9 <- NVME\VEN_8086
  Guid:                 2b8c6418-6719-51b3-a700-f6061c86874b <- INTEL SSDPEKNW512G8
  Serial:               BTNH836519G6512A
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  VendorId:             NVME:0x8086
  Version:              002C
  Icon:                 drive-harddisk
  Created:              2019-07-27

INTEL SSDPEKNW512G8
  DeviceId:             0fba4a8bedba53076fc9b147655e59455bc25a87
  Guid:                 c6cd9ab0-8f20-512e-9e1f-1af55b8454b9 <- NVME\VEN_8086&DEV_F1A8&REV_03
  Guid:                 82741c78-f5dc-5c23-a152-00de5799edc8 <- NVME\VEN_8086&DEV_F1A8
  Guid:                 3f235826-4614-56f6-8fd4-b67b346882c9 <- NVME\VEN_8086
  Guid:                 2b8c6418-6719-51b3-a700-f6061c86874b <- INTEL SSDPEKNW512G8
  Serial:               BTNH83640VZA512A
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  VendorId:             NVME:0x8086
  Version:              002C
  Icon:                 drive-harddisk
  Created:              2019-07-27

Hitachi HDT721010SLA360
  DeviceId:             5dbeb140337f610d54913c0f43d15fdaa0eafb51
  Guid:                 7cb5fed0-ec5f-54b0-b8bc-6d0b1fb9623a <- IDE\Hitachi_HDT721010SLA360_________________ST6OA31B
  Guid:                 d9e09882-91ce-5821-8e02-3ad53d1333bd <- IDE\0Hitachi_HDT721010SLA360_________________
  Guid:                 491a22a8-4a16-549d-88a7-fe52703a9dec <- Hitachi HDT721010SLA360
  Serial:               STF604MR249NGP
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              ST6OA31B
  Icon:                 drive-harddisk
  Created:              2019-07-27

Hitachi HDT721010SLA360
  DeviceId:             e421b2fc248391f6fe3e55ddbb3c9043be068bd0
  Guid:                 7cb5fed0-ec5f-54b0-b8bc-6d0b1fb9623a <- IDE\Hitachi_HDT721010SLA360_________________ST6OA31B
  Guid:                 d9e09882-91ce-5821-8e02-3ad53d1333bd <- IDE\0Hitachi_HDT721010SLA360_________________
  Guid:                 491a22a8-4a16-549d-88a7-fe52703a9dec <- Hitachi HDT721010SLA360
  Serial:               STF604MR21U2KP
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              ST6OA31B
  Icon:                 drive-harddisk
  Created:              2019-07-27
[... 107 lines cut ...]
  Guid:                 c5a9ce69-0231-51ec-a1a7-21f59c708956 <- IDE\HITACHI_HUA723030ALA640_________________MKAONS01
  Guid:                 331bbbc6-7cca-5730-ae25-54dd05b3ca38 <- IDE\0HITACHI_HUA723030ALA640_________________
  Guid:                 c8832a10-ef74-5705-9cf9-8362678b338f <- HITACHI HUA723030ALA640
  Serial:               YHHPARBA
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              MKAONS01
  Icon:                 drive-harddisk
  Created:              2019-07-27

WDC WD20EZRX-00D8PB0
  DeviceId:             faf2f69bde48af7e396887d213c87471b8001781
  Guid:                 50c4edf9-063e-556e-9d5b-d8db5cae9941 <- IDE\WDC_WD20EZRX-00D8PB0____________________80.00A80
  Guid:                 0a1cbda3-d8a0-5a9a-b7e7-ea98cf2efaf2 <- IDE\0WDC_WD20EZRX-00D8PB0____________________
  Guid:                 40927c7a-14b5-50b1-bf6b-9ced52e324ce <- WDC WD20EZRX-00D8PB0
  Serial:               WD-WCC4M7SKVJT0
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              80.00A80
  Icon:                 drive-harddisk
  Created:              2019-07-27```

Crash dump: http://pve.polymicro.net/share/pb-Es7wYuf5eb/

superm1, 2019-07-27:
Do you by chance have an AMD GPU in your system?

[EXTERNAL EMAIL]

tgunr, 2019-07-27:
Yes, RX-580

superm1, 2019-07-27:
Please try to blacklist the synapticsmst plugin in fwupd conf file (within /etc/fwupd/).

I believe you're exercising a kernel bug:  https://bugs.freedesktop.org/show_bug.cgi?id=110604

[EXTERNAL EMAIL]

Yes, RX-580

-

tgunr, 2019-07-27:
Will try that,

superm1, 2019-08-13:
Did you have any updates if this helped?

closed 2019-09-30 by stale[bot]