fwupd/fwupd #1257: [1.2.10] 'failed to get device after update' during update of Dell XPS13 9360 Thunderbolt Firmware

Klaas-, 2019-07-28:
There is an error message during thunderbolt updates. The updates are successful though, after reboot the new version is active.

From get-history:
```
XPS13 9360 Thunderbolt Controller
  DeviceId:             5cc3455f972eefe1884b359cb89d4161f37f906f
  Guid:                 dfd51125-338f-56ff-b721-fa3bea8e534e
  Plugin:               thunderbolt
  Flags:                internal|require-ac|supported|reported|wait-for-replug
  Version:              21.00
  Created:              2019-07-25
  UpdateState:          failed
  UpdateError:          failed to get device after update: failed to wait for detach replug: device 5cc3455f972eefe1884b359cb89d4161f37f906f did not come back
  
  [Release]
  RemoteId:             lvfs
  Version:              26.01
  Checksum:             SHA1(2e6bff2434b4311010e1d02266da02afaf327d8b)
  Flags:                none
  DistroVariant:        workstation
  RuntimeVersion(com.redhat.fwupdate): 12
  CompileVersion(org.freedesktop.gusb): 0.3.0
  CpuArchitecture:      x86_64
  RuntimeVersion(org.freedesktop.fwupd): 1.2.10
  DistroVersion:        30
  DistroId:             fedora
  ThunderboltNative:    False
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.14
  CompileVersion(org.freedesktop.fwupd): 1.2.10
  BootTime:             1564080275
  RuntimeVersion(com.dell.libsmbios): 2.4
  CompileVersion(com.redhat.efivar): 37
  KernelVersion:        5.1.18-300.fc30.x86_64
  CompileVersion(com.redhat.fwupdate): 12
```

Also happened last time I updated thunderbolt firmware:
```
XPS13 9360 Thunderbolt Controller
  DeviceId:             5cc3455f972eefe1884b359cb89d4161f37f906f
  Guid:                 dfd51125-338f-56ff-b721-fa3bea8e534e
  Plugin:               thunderbolt
  Flags:                internal|require-ac|supported|reported|wait-for-replug
  Version:              18.5
  Created:              2018-02-17
  Modified:             2018-02-18
  UpdateState:          failed
  UpdateError:          failed to get device after update: failed to wait for detach replug: device 5cc3455f972eefe1884b359cb89d4161f37f906f did not come back
  
  [Release]
  RemoteId:             lvfs
  Version:              21.00
  Checksum:             SHA1(d621932738d56a024cbcc516d9810e4c6ffe477a)
  Flags:                none
  FwupdVersion:         1.0.4
  GUsbVersion:          0.2.11
  BootTime:             1518868190
  KernelVersion:        4.15.3-300.fc27.x86_64
  AppstreamGlibVersion: 0.7.4
  CpuArchitecture:      x86_64

```

```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

```shell
$ fwupdmgr get-devices
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10

```

```shell
$ efibootmgr -v
BootOrder: 0002,0000,0001
Boot0000* ubuntu	HD(1,GPT,f34fc1b5-75bf-4667-ab65-7417648f2977,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux Firmware Updater	HD(1,GPT,f34fc1b5-75bf-4667-ab65-7417648f2977,0x800,0x12c000)/File(\EFI\fedora\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0002* Fedora	HD(1,GPT,f34fc1b5-75bf-4667-ab65-7417648f2977,0x800,0x12c000)/File(\EFI\fedora\shim.efi)
Boot0003* USB Storage Device	BBS(USB,USB Storage Device,0x0)..BO
Boot0004* CD/DVD/CD-RW Drive	BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* USB NIC	BBS(128,Realtek PXE B00 D14,0x0)..BO
Boot0006* M.2 PCIe SSD	BBS(HD,P0: THNSN5512GPUK NVMe TOSHIBA 512G,0x0)..BO
Boot0007* UEFI: THNSN5512GPUK NVMe TOSHIBA 512GB, Partition 1	HD(1,GPT,f34fc1b5-75bf-4667-ab65-7417648f2977,0x800,0x12c000)/File(EFI\boot\bootx64.efi)..BO
```

```shell
$ efivar -l | grep fw
[empty}
```

```shell
$ find /boot -iname "*cap" -or -iname "fwupd*"
/boot/efi/EFI/fedora/fwupdx64.efi
```

Please answer the following questions:

- Operating system and version:
Fedora 30
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
dnf
- Have you tried rebooting?
Update was successful, issue is just about the error message
- Are you using an NVMe disk?
yes
- Is secure boot enabled (only for the UEFI plugin)?
No

superm1, 2019-07-29:
This is surprised to see happening in 1.2.10, we thought we had sorted out the timing issues w/ Thunderbolt updates.

Do you by chance have a long chain of Thunderbolt devices connected?

Can you share the associated system log if you still have it?

Klaas-, 2019-07-29:
@superm1 I don't have any external devices connected, so I guess none is the answer to your first questoin.

System Logs: Yes I do have them, anything in particular you're looking for?

journalctl -u fwupd.service
```
-- Reboot --
Jul 25 20:45:47 notebook systemd[1]: Starting Firmware update daemon...
Jul 25 20:45:48 notebook systemd[1]: Started Firmware update daemon.
Jul 26 08:42:20 notebook systemd[1]: Stopping Firmware update daemon...
Jul 26 08:42:20 notebook fwupd[2864]: 06:42:20:0370 GLib                 g_atomic_ref_count_dec: assertion 'g_atomic_int_get (arc) > 0' failed
Jul 26 08:42:20 notebook systemd[1]: fwupd.service: Succeeded.
Jul 26 08:42:20 notebook systemd[1]: Stopped Firmware update daemon.
-- Reboot --
Jul 26 08:43:04 notebook systemd[1]: Starting Firmware update daemon...
Jul 26 08:43:06 notebook systemd[1]: Started Firmware update daemon.
```

superm1, 2019-07-29:
Hmm, nothing connected means it could have been something wrong with `bolt` or the `intel-wmi-thunderbolt` not bringing the device back up.  `fwupd` uses `bolt` to do this.  `bolt` uses `intel-wmi-thunderbolt` from the kernel to bring it back up.

Can you check syslog specifically?  I want to see what the kernel was showing at the time of the update.  Might need to coordinate bolt logs as well.  If nothing is obvious might need to set both `fwupd` and `bolt` to run in verbose mode and then try to rerun the update.

Klaas-, 2019-07-29:
bolt:
```
-- Reboot --
Jul 25 20:45:47 notebook systemd[1]: Starting Thunderbolt system service...
Jul 25 20:45:47 notebook boltd[2882]: bolt 0.8 starting up.
Jul 25 20:45:47 notebook boltd[2882]: store: located at: /var/lib/boltd
Jul 25 20:45:47 notebook boltd[2882]: config: loading user config
Jul 25 20:45:47 notebook boltd[2882]: bouncer: initializing polkit
Jul 25 20:45:47 notebook boltd[2882]: udev: initializing udev
Jul 25 20:45:47 notebook boltd[2882]: store: loading domains
Jul 25 20:45:47 notebook boltd[2882]: [d9010000-0072-domain?                    ] store: loading domain
Jul 25 20:45:47 notebook boltd[2882]: journal: opened for 'd9010000-0072'; size: 0 bytes
Jul 25 20:45:47 notebook boltd[2882]: [d9010000-0072-domain?                    ] domain: registered (bootacl: 0/0)
Jul 25 20:45:47 notebook boltd[2882]: store: loading devices
Jul 25 20:45:47 notebook boltd[2882]: power: state located at: /run/boltd/power
Jul 25 20:45:47 notebook boltd[2882]: power: force power support: yes
Jul 25 20:45:47 notebook boltd[2882]: power: setting force_power to ON
Jul 25 20:45:47 notebook boltd[2882]: power: state changed: supported/on
Jul 25 20:45:47 notebook boltd[2882]: power: guard '1' for 'boltd' active
Jul 25 20:45:47 notebook boltd[2882]: udev: found 1 domain
Jul 25 20:45:47 notebook boltd[2882]: manager: acquired power guard '1'
Jul 25 20:45:47 notebook boltd[2882]: udev: enumerating devices
Jul 25 20:45:47 notebook boltd[2882]: probing: adding /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0 to roots
Jul 25 20:45:47 notebook boltd[2882]: security level set to 'user'
Jul 25 20:45:47 notebook boltd[2882]: [d9010000-0072-domain0                    ] connected: as domain0 [user] (/sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0)
Jul 25 20:45:47 notebook boltd[2882]: [d9010000-0072-XPS13 9360                 ] device added, status: authorized, at /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0/0-0
Jul 25 20:45:47 notebook boltd[2882]: [d9010000-0072-XPS13 9360                 ] labeling device: Dell XPS13 9360
Jul 25 20:45:47 notebook boltd[2882]: power: guard '1' for 'boltd' deactivated
Jul 25 20:45:47 notebook boltd[2882]: power: shutdown scheduled (T-20,00s)
Jul 25 20:45:47 notebook boltd[2882]: power: state changed: supported/wait
Jul 25 20:45:47 notebook boltd[2882]: dbus: exported domain at /org/freedesktop/bolt/domains/d9010000_0072_7f18_a2c1_b5d7ead11a19
Jul 25 20:45:47 notebook boltd[2882]: [d9010000-0072-XPS13 9360                 ] dbus: exported device at /org/freedesktop/bolt/devices/d9010000_0072...
Jul 25 20:45:47 notebook systemd[1]: Started Thunderbolt system service.
Jul 25 20:45:47 notebook boltd[2882]: probing: started [1000]
Jul 25 20:45:47 notebook boltd[2882]: probing: adding /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0 to roots
Jul 25 20:45:47 notebook boltd[2882]: power: resetting timeout (uevent /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0)
Jul 25 20:45:47 notebook boltd[2882]: [d9010000-0072-XPS13 9360                 ] udev: device changed: authorized -> authorized
Jul 25 20:45:47 notebook boltd[2882]: power: resetting timeout (uevent /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0/0-0)
Jul 25 20:45:47 notebook boltd[2882]: power: state changed: supported/on
Jul 25 20:45:47 notebook boltd[2882]: power: guard '2' for 'fwupd' active
Jul 25 20:45:49 notebook boltd[2882]: probing: timeout, done: [2004135] (2000000)
Jul 25 20:46:08 notebook boltd[2882]: power: got event for guard '2' (10)
Jul 25 20:46:08 notebook boltd[2882]: power: guard '2' for 'fwupd' deactivated
Jul 25 20:46:08 notebook boltd[2882]: power: shutdown scheduled (T-20,00s)
Jul 25 20:46:08 notebook boltd[2882]: power: state changed: supported/wait
Jul 25 20:46:28 notebook boltd[2882]: power: setting force_power to OFF
Jul 25 20:46:28 notebook boltd[2882]: power: state changed: supported/off
Jul 25 20:46:28 notebook boltd[2882]: [d9010000-0072-XPS13 9360                 ] removed (/sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0/0-0)
Jul 25 20:46:28 notebook boltd[2882]: [d9010000-0072-XPS13 9360                 ] dbus: unexported
Jul 25 20:46:28 notebook boltd[2882]: [d9010000-0072-domain0                    ] disconnected from /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0
Jul 25 20:46:33 notebook boltd[2882]: probing: removing /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0 from roots
Jul 26 08:40:23 notebook boltd[2882]: power: setting force_power to ON
Jul 26 08:40:23 notebook boltd[2882]: power: state changed: supported/on
Jul 26 08:40:23 notebook boltd[2882]: power: guard '3' for 'fwupd' active
Jul 26 08:40:25 notebook boltd[2882]: probing: started [1000]
Jul 26 08:40:25 notebook boltd[2882]: probing: adding /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0 to roots
Jul 26 08:40:25 notebook boltd[2882]: [d9010000-0072-domain0                    ] connected: as domain0 [user] (/sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0)
Jul 26 08:40:25 notebook boltd[2882]: [d9010000-0072-XPS13 9360                 ] device added, status: authorized, at /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0/0-0
Jul 26 08:40:25 notebook boltd[2882]: [d9010000-0072-XPS13 9360                 ] labeling device: Dell XPS13 9360
[... cut ...]
Jul 26 08:41:03 notebook kernel: pcieport 0000:02:02.0: BAR 15: failed to assign [mem size 0x00200000 64bit pref]
Jul 26 08:41:03 notebook kernel: pcieport 0000:02:02.0: BAR 13: no space for [io  size 0x1000]
Jul 26 08:41:03 notebook kernel: pcieport 0000:02:02.0: BAR 13: failed to assign [io  size 0x1000]
Jul 26 08:41:03 notebook kernel: pcieport 0000:02:01.0: BAR 13: no space for [io  size 0x1000]
Jul 26 08:41:03 notebook kernel: pcieport 0000:02:01.0: BAR 13: failed to assign [io  size 0x1000]
```

superm1, 2019-07-30:
@gicmo you know the bolt logging a bit better - do you see a timing problem here at all with bolt?  The third guard never released, so I don't think it's a _timeout_ problem per-say.

gicmo, 2019-07-31:
From the log it looks like boltd force powered the port until the reboot (with guard 3). Looks like it didn't come up again after the power cycle.

closed 2019-09-30 by stale[bot]