fwupd/fwupd #1261: Bootloader version BOT01.04_B0016 not supported

meliot, 2019-07-31:
fwupdmgr update not working, according to the error message the bootloader is unsupported:
```shell
meliot@ubuntu:~/fwupd$ fwupdmgr update
No releases found for device: Not compatible with bootloader version: failed predicate [BOT01.0[0-3]_* regex BOT01.04_B0016]
```

Thanks for your amazing work.
meliot

---

Requested information:
```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.2.10
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34

```

```shell
$ fwupdmgr get-devices
Unifying Receiver
[..]
  Summary:              A miniaturised USB wireless receiver (bootloader)
  Plugin:               unifying
  Flags:                updatable|supported|registered|is-bootloader
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR00.00_B0000
  VersionBootloader:    BOT01.04_B0016
  Icon:                 preferences-desktop-keyboard
  Created:              2019-07-31
```

Please answer the following questions:

- Operating system and version: PRETTY_NAME="Ubuntu 18.04.2 LTS"
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? yep
- Are you using an NVMe disk?

FFY00, 2019-09-23:
/unstale

FFY00, 2019-09-23:
We need to change the way this is implemented so that fwupdmgr doesn't spit out errors like that.

hughsie, 2019-09-24:
I think the issue here is that we have the same failure mode for "we can't install the component that matches the hardware because of some underlying problem" and "we can't install the component that matches the hardware because it doesn't really match the hardware". i.e. one is a known failure the user needs to be aware of, and the other is a known restriction that's perfectly normal.

meliot, 2019-10-28:
Thanks for the update. In few words this means that the message will be fixed to show "unsupported hardware" and that the hardware will not be supported in the next future? 
Thank you

superm1, 2019-10-28:
The exact specifics of how messaging will be fixed up I'll defer to @FFY00 and @hughsie.  I was just triaging the issues some.

rgpublic, 2019-11-26:
Sorry for the spam, but I have the same question as @meliot . What does it now actually mean for end-users? That you cannot update the firmware of newer Logitech Unifiying Receivers with fwupd and they should be updated with Windows for now, right?

FFY00, 2019-11-26:
You can, the error you got just means there are no updates available.

rgpublic, 2019-11-26:
Ah, thank you so much @FFY00 for the clarification! Hopefully this will also help others when they end up here while googling the error message.

i-need-to-tell-you-something, 2019-12-01:
@rgpublic In some cases the solution for end users is to upgrade fwupd version by switching to [snap](https://github.com/fwupd/fwupd/wiki/fwupd-snap) or flatpak source. (Worked for my Unifying receiver error on Ubuntu 18.04). Or alternatively switch to a distro that ships newer fwupd.

commit 50903fff (2019-12-12): engine: don't expose bootloader version errors to users

These messages are useful for debugging but not to users.  Many issues
get filed because they're cryptic and users can't do anything about them.

Downgrade the actual messages to debugging in the engine, and just send
back a generic message for clients.

Fixes: #1659
Fixes: #1261
commit a712448a (2019-12-13): engine: don't expose bootloader version errors to users

These messages are useful for debugging but not to users.  Many issues
get filed because they're cryptic and users can't do anything about them.

Downgrade the actual messages to debugging in the engine, and just send
back a generic message for clients.

Fixes: #1659
Fixes: #1261
commit c361b675 (2019-12-13): engine: don't expose bootloader version errors to users

These messages are useful for debugging but not to users.  Many issues
get filed because they're cryptic and users can't do anything about them.

Downgrade the actual messages to debugging in the engine, and just send
back a generic message for clients.

Fixes: #1659
Fixes: #1261
commit 1681058a (2019-12-13): fu-engine: don't expose bootloader version errors to users

These messages are useful for debugging but not to users.  Many issues
get filed because they're cryptic and users can't do anything about them.

Downgrade the actual messages to debugging in the engine, and just send
back a generic message for clients.

Fixes: #1659
Fixes: #1261
closed 2019-12-13 by superm1