fwupd/fwupd #1270: provide a network service file

rickysarraf, 2019-08-05:
In its current form, fwupd does not refresh its database on a periodic basis. Please provide a database update service file so that fwupd can periodically update its database.

```
rrs@priyasi:/var/tmp/$ sudo fwupdmgr  update
No upgrades for XPS 13 9370 System Firmware, current is 0.1.5.1: 0.1.5.1=same, 0.1.4.0=older, 0.1.3.3=older, 0.1.2.1=older, 0.1.2.0=older
10:28 ♒♒♒   ☺ 😄    

rrs@priyasi:/var/tmp/$ sudo fwupdmgr get-updates 
Firmware metadata has not been updated for 30 days and may not be up to date.

Update now? (Requires internet connection) [y|N]: y
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************] Less than one minute remaining…
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

XPS 13 9370 System Firmware has firmware updates:
GUID:                    7ceaf7a8-0611-4480-9e30-64d8de420c7c
GUID:                    43ea5588-d9a4-5031-8ad3-308045302d6b
GUID:                    230c8b18-8d9b-53ec-838b-6cfc0383493a
ID:                      com.dell.uefi7ceaf7a8.firmware
Update Version:          0.1.6.3
Update Name:             XPS 13 9370 System Update
Update Summary:          Firmware for the Dell XPS 13 9370
Update Remote ID:        lvfs
Update Checksum:         SHA1(ce74ca63e86e2844a37748fefbdaf2eaaa36345a)
Update Location:         https://fwupd.org/downloads/af59a2314c4881ffc02e3a204327d765d0fd256b-Signed_1152921504627870090.cab
Update Description:      This stable release fixes the following issues:
                          • Fixed the incorrect Windows operating system logon message that is displayed on the lock screen of Windows.
                          • Fixed the issue where the characters fail to display on the screen while typing continuously.
                          • Enabled support for Intel Turbo Boost Technology on I3-8310U processors that was disabled by default.
                          • Optimized the battery diagnostics in Dell Enhanced Pre-boot System Assessment (ePSA).
13:18 ♒♒♒   ☺ 😄    

rrs@priyasi:/var/tmp/$ sudo fwupdmgr update
Downloading 0.1.6.3 for XPS 13 9370 System Firmware...
Fetching firmware https://fwupd.org/downloads/af59a2314c4881ffc02e3a204327d765d0fd256b-Signed_1152921504627870090.cab
Downloading…           [***************************************] Less than one minute remaining…
Decompressing…         [***************************************]
Authenticating…        [***************************************]
failed to run update_prepare() on upower: Cannot install update when not on AC power unless forced
13:27 ♒♒♒    ☹ 😟=> 1  

rrs@priyasi:/var/tmp/$ sudo fwupdmgr update
Downloading 0.1.6.3 for XPS 13 9370 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Updating XPS 13 9370 System Firmware…                          ]
Scheduling…            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: 
```

Please answer the following questions:
- Operating system and version: Debian
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt
- Downstream bug report: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=921820

[fwupd-update.service.txt](https://github.com/hughsie/fwupd/files/3467555/fwupd-update.service.txt)
[fwupd-update.timer.txt](https://github.com/hughsie/fwupd/files/3467556/fwupd-update.timer.txt)

hughsie, 2019-08-05:
@superm1 didn't you have something like this too?

superm1, 2019-08-05:
I started something on https://github.com/hughsie/fwupd/tree/wip/superm1/motd-integration?files=1 but didn't get around to finishing integration

commit 02904bba (2019-08-20): Add support for an optional motd message (Fixes: #1270)

Fixes Debian bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=921820
commit 10e0205b (2019-08-20): Add support to integrate into the motd (Fixes: #1270)

Fixes Debian bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=921820

Introduce a new --log option to fwupdmgr that will log stdout to an argument.
If run under systemd, prefix that argument with $RUNTIME_DIRECTORY.

Add a new systemd unit and associated timer to regularly refresh metadata.
After the metadata refresh is complete, save the output to the motd location.

The timer and service are disabled by default and can be enabled by an admin.
commit 3b0fc30d (2019-08-20): Add support to integrate into the motd (Fixes: #1270)

Fixes Debian bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=921820

Introduce a new --log option to fwupdmgr that will log stdout to an argument.
If run under systemd, prefix that argument with $RUNTIME_DIRECTORY.

Add a new systemd unit and associated timer to regularly refresh metadata.
After the metadata refresh is complete, save the output to the motd location.

The timer and service are disabled by default and can be enabled by an admin.
commit 5cc4efe6 (2019-08-20): Add support to integrate into the motd (Fixes: #1270)

Fixes Debian bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=921820

Introduce a new --log option to fwupdmgr that will log stdout to an argument.
If run under systemd, prefix that argument with $RUNTIME_DIRECTORY.

Add a new systemd unit and associated timer to regularly refresh metadata.
After the metadata refresh is complete, save the output to the motd location.

The timer and service are disabled by default and can be enabled by an admin.
closed 2019-08-22 by hughsie