fwupd/fwupd #1271: ModemManager plugin broken since 1.2.9

aleksander0m, 2019-08-05:
Hey!

It looks like the ModemManager plugin to update the DW5821e got somewhat broken with 1.2.9, especifically with commit 8ef139faba949f563cfb5acb15c43903cf660552. for issue https://github.com/hughsie/fwupd/issues/1162. If I revert that commit on top of git master, it works.

That commit disabled the most generic usb instance ids for matching devices, and I believe the ModemManager plugin was too weak and it really relied on those for matching. I'm not sure how to best fix the plugin to avoid this issue, so help and comments very appreciated.

The problem since that commit is that as soon as the update procedure starts and the modem reboots in fastboot mode, the newly exposed device isn't detected as a match for the modem device:
```
Downloading T77W968.F1.0.0.4.0.VF.007 for DW5821e Snapdragon X20 LTE...
(fwupdmgr:16942): FuCommon-DEBUG: 15:10:24.350: creating path /home/user/.cache/fwupd
(fwupdmgr:16942): FuMain-DEBUG: 15:10:24.569: skpping download as file already exists
(fwupdmgr:16942): Fwupd-DEBUG: 15:10:24.608: Emitting ::status-changed() [decompressing]
Decompressing…           [      \                                ](fwupdmgr:16942): Fwupd-DEBUG: 15:10:24.887: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:16942): Fwupd-DEBUG: 15:10:25.511: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [       \                               ](fwupdmgr:16942): Fwupd-DEBUG: 15:10:25.514: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
(fwupdmgr:16942): Fwupd-DEBUG: 15:10:25.821: Emitting ::changed()
(fwupdmgr:16942): Fwupd-DEBUG: 15:10:29.784: Emitting ::device-added(daa833f08c8b785b5f46be7143f81c8622987f1c)
failed to get device after detach: failed to wait for detach replug: device a33fad1e84e16555dc6ac46bf0e1dcff54ae8377 did not come back
```
The info in fwupd for the modem BEFORE the upgrade attempt is:
```
DW5821e Snapdragon X20 LTE
  DeviceId:             a33fad1e84e16555dc6ac46bf0e1dcff54ae8377
  Guid:                 b595e24b-bebb-531b-abeb-620fa2b44045 <- USB\VID_413C&PID_81D7&REV_0318&CARRIER_VODAFONE
  Guid:                 64da2d58-8d1b-5e5b-b793-f88ba5a25a8f <- USB\VID_413C&PID_81D7&REV_0318
  Guid:                 761d6124-0002-5185-b767-9adf67bf1a5e <- USB\VID_413C&PID_81D7
  Guid:                 795e079d-093b-5503-aa59-35b832480e95 <- USB\VID_413C
  Summary:              Mobile broadband device
  Plugin:               modem_manager
  Flags:                updatable|supported|registered|use-runtime-version
  Vendor:               Dell Inc.
  Version:              T77W968.F1.0.0.4.0.VF.007
  Icon:                 network-modem
  Created:              2019-08-05
```

The info in fwupd for the modem in fastboot mode during the FAILED attempt is:
DW5821e Snapdragon X20 LTE
  DeviceId:             daa833f08c8b785b5f46be7143f81c8622987f1c
  Guid:                 390e8c98-f738-597d-8611-d20839844933 <- USB\VID_413C&PID_81D6&REV_0100
  Guid:                 a5d74db8-71c5-501d-92e2-d41046e8b3f3 <- USB\VID_413C&PID_81D6
  Serial:               5d9e7b44
  Plugin:               fastboot
  Flags:                updatable|registered
  Vendor:               DELL
  VendorId:             USB:0x413C
  Version:              1.0
  Created:              2019-08-05

My assumption is that if fwupd had added a VID-specific GUID for the fastboot device, both devices would have matched because they both would have had "USB\VID_413C". In this case, it didn't match, because the same device boots with a different PID (81d6 in fastboot mode, 81d7 in modem mode), so that GUID wasn't enough to match.

How can we handle this issue?

hughsie, 2019-08-05:
> both devices would have matched because they both would have had "USB\VID_413C".

Right, I think that's the root of this issue. Matching based on the vendor ID alone seems somewhat fragile in hindsight.

> How can we handle this issue?

Can we just define a counterpart GUID in a quirk file? e.g. this is how the ColorHug runtime says "PID USB\VID_273F&PID_1000 is my bootloader" : https://github.com/hughsie/fwupd/blob/master/plugins/colorhug/colorhug.quirk#L17

Richard

aleksander0m, 2019-08-05:
I tried adding a quirks file like this, but it didn't work. What could it be missing?

```
$ cat /usr/share/fwupd/quirks.d/modem-manager.quirk 
# DW5821e
# in fastboot mode, it exposes a different PID
[DeviceInstanceId=USB\VID_413C&PID_81D7]
Summary = Dell DW5821e LTE modem
Plugin = modem_manager
CounterpartGuid = USB\VID_413C&PID_81D6
```

hughsie, 2019-08-05:
> What could it be missing?

Got logs showing the changeover to fastboot mode?

aleksander0m, 2019-08-05:
Here:
```
14:50:54:0458 FuDeviceList         waiting 10000ms for replug
14:50:54:0458 FuPluginMm           removed modem: /org/freedesktop/ModemManager1/Modem/0
14:50:54:0458 FuPlugin             emit removed from modem_manager: a33fad1e84e16555dc6ac46bf0e1dcff54ae8377
14:50:54:0458 FuDeviceList         waiting 10000ms for DW5821e Snapdragon X20 LTE device removal

14:50:58:0370 FuPlugin             performing usb_device_added() on fastboot
14:50:58:0370 FuDevice             using daa833f08c8b785b5f46be7143f81c8622987f1c for usb:00:01:07
14:50:58:0387 FuPluginFastboot     fastboot version=0.5
14:50:58:0417 FuPlugin             emit added from fastboot: daa833f08c8b785b5f46be7143f81c8622987f1c
14:50:58:0417 FuDeviceList         ::added daa833f08c8b785b5f46be7143f81c8622987f1c
14:50:58:0418 XbSilo               1,20ms      query on / with `components/component/provides/firmware[@type='flashed'][text()='390e8c98-f738-597d-8611-d20839844933']/../..|components/component/provides/firmware[@type='flashed'][text()='a5d74db8-71c5-501d-92e2-d41046e8b3f3']/../..` limit=1 -> 0 results
14:50:58:0418 FuHistory            get device
14:51:04:0464 FuDeviceList         device did not replug
14:51:04:0464 FuDeviceList         doing delayed removal
14:51:04:0464 FuDeviceList         ::removed a33fad1e84e16555dc6ac46bf0e1dcff54ae8377
14:51:04:0464 FuPlugin             performing device_removed() on dell_dock
14:51:04:0465 FuPlugin             performing device_removed() on synapticsmst
14:51:04:0465 FuHistory            modifying device DW5821e Snapdragon X20 LTE [a33fad1e84e16555dc6ac46bf0e1dcff54ae8377], version not important
14:51:04:0480 FuPlugin             performing composite_cleanup() on dell_dock
14:51:04:0480 FuPlugin             performing composite_cleanup() on wacom_usb
14:51:04:0480 FuIdle               uninhibiting: performing update
```

aleksander0m, 2019-08-05:
Oh, it worked now, adding the PIDs the other way around (a DeviceInstanceId for 81D6 with CounterpartGuid for 81D7). Does that make sense?

hughsie, 2019-08-05:
I didn't think it mattered, maybe it does, sorry. If it matters that's probably (another) bug that needs fixing :)

aleksander0m, 2019-08-05:
Your colorhug quirk file has both entries (DeviceInstanceId=USB\VID_273F&PID_1000 with counterpart 1001 and DeviceInstanceId=USB\VID_273F&PID_1001 with counterpart 1000). I did the same here and it seems it worked.

hughsie, 2019-08-05:
I guess that does make it explicit, and in the case where multiple runtime versions all share a bootloader VID:PID (e.g. like all the 8bitdo controllers) maybe you don't want to always "go the other way" by default. If you're happy adding it both ways, sounds good to me.

aleksander0m, 2019-08-05:
ok, will do, preparing a merge request already

closed 2019-08-05 by hughsie