fwupd/fwupd #1280: Latest BIOS Update for T480s does not show up

maaattes, 2019-08-12:
Latest System Firmware ([Version 0.1.33](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480s-type-20l7-20l8/downloads/ds502226) released on 02.08.2019) for Lenovo ThinkPad T480s does not show up. It was [uploaded to LVFS](https://fwupd.org/lvfs/device/com.lenovo.ThinkPadN22ET.firmware).

```shell
$ fwupdmgr --version
client version:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

```shell
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             24dd1466bf80723529c067bcfd0d35ab5bea2f38
  Guid:                 3b125a20-5ed2-5c0f-b22d-ab0e0b687bbc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2019-08-12

20L8S02D00 System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Checksum:             SHA1(9bd665a9982ca3b2a114be4a1b6255782c14e6b0)
  Checksum:             SHA256(3e95c31fbf414a6c3764d736c51717ebce05a2ac5af4d40a015e3546b4e98427)
  Version:              0.1.31
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2019-08-12

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.65.3590
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-08-12

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.13
  VersionLowest:        0.1.13
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2019-08-12

INTEL SSDPEKKF512G8L
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 79517f86-8df8-5d6e-a18b-33f0b36a78e9
  Guid:                 68db11e5-b0cf-5bc9-a94e-17e28496e505
  Guid:                 87fd9fec-f447-5ec6-a1cd-d55fac0d3d30
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  VendorId:             NVME:0x8086
  Version:              L08P
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-08-12
```

Please answer the following questions:

- Operating system and version: Fedora 30
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): via dnf install
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

hughsie, 2019-08-12:
what does `fwupdmgr get-updates` say?

maaattes, 2019-08-12:
```
No upgrades for 20L8S02D00 System Firmware, current is 0.1.31: 0.1.31=same, 0.1.30=older, 0.1.29=older, 0.1.27=older, 0.1.26=older
No upgrades for UEFI Device Firmware, current is 184.65.3590: 184.65.3590=same, 184.60.3561=older, 184.55.3510=older
No upgrades for UEFI Device Firmware, current is 0.1.13: 0.1.13=same
________________________________________________

Erfolgreich aktualisierte Geräte:

 • UEFI Device Firmware (0.1.11 → 0.1.13)
 • 20L8S02D00 System Firmware (0.1.30 → 0.1.31)
 • UEFI Device Firmware (184.60.3561 → 184.65.3590)
```

hughsie, 2019-08-12:
Hmm, not in the metadata either. The release date is exactly the same as 0.1.31 (1467946800) which is a bit suspicious. Debugging...

hughsie, 2019-08-12:
Can you wait 30 minutes, then `fwupdmgr refresh` then try again?

maaattes, 2019-08-12:
Sure, I'll keep you posted.

maaattes, 2019-08-12:
Now it seems to work. I will try to update as soon as my productive working day is over. I will report back to you.

joakim-tjernlund, 2019-08-12:
T480s update to 0.1.33 just worked for me

maaattes, 2019-08-12:
I was just able to update successfully. Thanks for your quick feedback, Richard.

closed 2019-08-13 by hughsie