fwupd/fwupd #1281: Snap version fails: '/snap/fwupd/880/bin/fwupdmgr: not found'

LinuxOnTheDesktop, 2019-08-12:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
fwupdmgr --version
/snap/fwupd/880/fwupd-command: 41: exec: /snap/fwupd/880/bin/fwupdmgr: not found
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```
$  dpkg -l fwupd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                 Version         Architecture    Description
+++-====================-===============-===============-=============================================
ii  fwupd                1.0.9-0ubuntu2  amd64           Firmware update daemon
```

```shell
$ fwupdmgr get-devices
/snap/fwupd/880/fwupd-command: 41: exec: /snap/fwupd/880/bin/fwupdmgr: not found
```

Please answer the following questions:

- Operating system and version: Linux Mint 19.2 Cinnamon
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

The snap version of fwupd, which was failing to unmount at shutdown, showed a problem when i ran 'snap list'. So I used 'snap remove' to remove it and then I reinstalled it with `snap install fwupd --classic`. And now when I try to use the program it reports:

`/snap/fwupd/880/fwupd-command: 41: exec: /snap/fwupd/880/bin/fwupdmgr: not found`

There is also this report from systemd:

```
  UNIT                     LOAD   ACTIVE SUB    DESCRIPTION                             
● snap-core-7270.mount     loaded failed failed Mount unit for core, revision 7270      
● snap-core-7396.mount     loaded failed failed Mount unit for core, revision 7396      
● snap.fwupd.fwupd.service loaded failed failed Service for snap application fwupd.fwupd
```

I find that immediately after reinstalling the snap, I can install firmware updates - but, after that, I end up in the present situation.

superm1, 2019-08-12:
If you want to remove your apt based install it should be:
`# apt remove fwupd`

Snaps are self-contained but there are 2-3 files that are shared with the snap and deb regarding policykit policies.  So these can stomp on each other.

```
core 16-2.40 from Canonical✓ installed
 ~  $  fwupdmgr get-updates
Unsupported daemon version 1.0.9, client version is 1.2.10
```

It's using the deb provided client (`/usr/bin/fwupdgmr`) to contact the snap provided server.

```
 ~  $  fwupdmgr get-updates
/snap/fwupd/880/fwupd-command: 36: /snap/fwupd/880/fwupd-command: /snap/fwupd/880/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: not found
/snap/fwupd/880/fwupd-command: 41: exec: /snap/fwupd/880/bin/fwupdmgr: not found
```

Something weird is going on.  can you try to remove `~/snap/fwupd` and `/root/snap/fwupd` directories and try again?

LinuxOnTheDesktop, 2019-08-13:
> If you want to remove your apt based install it should be [. . .]

I should have said: yes, I did that first. I used that `rm` in addition to `apt remove` in order to avoid precisely the sort of problems that seem to have come up anyway, viz., residues from the old apt installation. I imagine that using apt to _fully_ remove the old package - by doing `apt purge fwupd` - would have removed `~/snap/fwupd`; would it have removed also `/root/snap/fwupd'? If not, then perhaps the snap version of the software should be alive to the possibility of the latter stuff existing.

Anyhow, having now run the commands you recommended - and having removed all my snaps and reinstalled them - the snap version of `fwupdmgr` seems fine (though I remain confused about the relation between `fwupdmgr` and `fwupd`).

I will close this bug report if you think there's is nothing to do at your end.

superm1, 2019-08-13:
>  imagine that using apt to fully remove the old package - by doing apt purge fwupd - would have removed ~/snap/fwupd

Removing via apt doesn't touch `/root/snap` or `~/snap`.  those are only used via snap package.

>If not, then perhaps the snap version of the software should be alive to the possibility of the latter stuff existing.

Completely agree, there is some problem that occurred with junk in one of those directories leading to the problematic behavior you encountered.

>Anyhow, having now run the commands you recommended - and having removed all my snaps and reinstalled them - the snap version of fwupdmgr seems fine (though I remain confused about the relation between fwupdmgr and fwupd).

```
/snap/fwupd/880/fwupd-command: 36: /snap/fwupd/880/fwupd-command: /snap/fwupd/880/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: not found
```

Do you have any idea what version of the snap you upgraded from?  It would be good to try to reproduce this to find where the problem actually was.  If you don't have that information then I think we should close that and re-open if someone else encounters and we want to debug further.

LinuxOnTheDesktop, 2019-08-13:
> Do you have any idea what version of the snap you upgraded from? 

I am afraid not. If the snap upgraded itself then it did so automatically (in a way that is meant to just work . . :).

superm1, 2019-08-13:
Yeah it's _supposed_ to "just work".    Something is wrong with this script that's part of the snap I think.
https://github.com/hughsie/fwupd/blob/master/contrib/snap/fwupd-command

superm1, 2019-08-22:
If someone encounters this again in the future, please re-open.  It would be good to examine the state of ~/snap and /root/snap at that time.

closed 2019-08-22 by superm1