fwupd/fwupd #1283: Question: interfacing with Python or JS via gobject

SamBurgos, 2019-08-13:
Maybe this is not an issue per-se, but I would like to ask if there is any way to use the fwupdmgr command with Python or Javascript via gobject, or if it is limited to C/Bash to execute the commands?

Also if only using gnome-software (Gnome Shell) or Discover (KDE Plasma) are the only frontends available for this or in case someone wants to add it to another frontend would that be possible and where can I find the documentation if needed

hughsie, 2019-08-14:
You can use libfwupd using gobject-introspection. If there's something in the command line tools that you'd like exporting in the library, please let us know! Another work-in-progress front-end is https://gitlab.gnome.org/hughsie/gnome-firmware-updater

closed 2019-08-14 by hughsie

SamBurgos, 2019-08-14:
Thanks for the answer, do you have any documentation or example that I can use to test it with Python or Javascript code via gobject-introspection? It would be great to have something to check in case I want to perform additional tests

additionally, does the WIP front-end that you said is compatible with GTK desktops enviroments outside of Gnome Shell (i.e.: Xfce or Cinnamon)? Though it's good to know that Gnome already provides a frontend and a GUI to do this process but I would like to test if it is possible to implement this without depending on Gnome Shell components

If there is any other way to communicate with you or to check even more information or examples of usage, I truly appreciate it.

Kind regards

superm1, 2019-08-14:
There is one in the contrib directory of the tree already.
https://github.com/hughsie/fwupd/blob/master/contrib/simple_client.py

> additionally, does the WIP front-end that you said is compatible with GTK desktops enviroments outside of Gnome Shell (i.e.: Xfce or Cinnamon)? Though it's good to know that Gnome already provides a [...]

I don't believe it has a dependency on gnome _currently_.  Long term it might go into Gnome later though.