fwupd/fwupd #1285: Distribute icons for plugins and vendors?

superm1, 2019-08-14:
In looking at how well https://gitlab.gnome.org/hughsie/gnome-firmware-updater/merge_requests/10 is turning out, I wonder if it would make sense to also provide some more icons distributed with the daemon that frontends can use.

Two things came to mind:
* The `Vendor` block can use vendor logos (I think nearly all of these exist already from LVFS)
* The `Plugin` block can use vendor logos for obvious vendor plugins (synaptics, dell, etc) and consortium logos for things like UEFI and redfish.

Then a frontend can do an icon lookup with the daemon and show the icon instead of more words in gnome-firmware-updater.

Plugin: dell_dock
Version: 1.21
Vendor: Dell Inc.
Vendor ID: USB:0x413C

Can probably collapse down to:

`Vendor: <dell logo>` w/ a tooltip indicating the plugin and Vendor ID.

hughsie, 2019-08-15:
Do we have permission to use the vendor logos not on the LVFS? I think it's more useful to include logos for UEFI and Redfish, but I think those should be shipped in fwupd and installed in /usr/share/fwupd/icons/com.synaptics.mst.png for clients to use offline too.

superm1, 2019-08-15:
I was thinking to only include logos already on LVFS. Clients can show the logo if it exists and placeholder text if not.

hughsie, 2019-08-15:
I think that might take some vendors by surprise. I also don't think we have a way to "tag" firmware with the vendor group_id from the LVFS, e.g. going from ThinkPad -> lenovo_thinkpad.

superm1, 2019-08-15:
Oh I was thinking it's by device not by firmware. So if you have an NVMe device that identified as vendor SK Hynix then you show SK Hynix logo. 

UEFI devices don't have a vendor attribute so they would only show a UEFI logo for the plugin.

closed 2019-08-22 by superm1