fwupd/fwupd #1292: nothing happens after reboot

emilas44, 2019-08-18:
```
fwupdmgr --version
client version:	1.2.9
checkout info:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9
```

```
sudo fwupdmgr install 31c5f61ceb395c6a7379b3481f1560fe7dabe274-Lenovo-ThinkPad-P1X1Extreme-ConsumerMEFirmware-12.0.35.1427.cab
[sudo] password for elvis: 
Decompressing?           [***************************************]
Authenticating?          [***************************************]
Installing on UEFI Device Firmware?                              ]
Scheduling?              [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: 

```

I have a X1 Extreme and fwupdate does nothing after reboot.... if I run `fwupdmgr update` it downloads the update file and it asks if I want to reboot and If I type in Y and Enter it will reboot and comes normally up to login screen and does nothing. I even made a video so you guys can see it. Maybe someone can suggest something what can I try or do to make it work?

https://www.youtube.com/watch?v=awlPrj4YX0w

- Operating system and version: POP-OS 18.04 
- How did you install fwupd: from snap master branch
- Have you tried rebooting? of course
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? secure boot is disabled

emilas44, 2019-08-18:
@hughsie asked
'have you tried clearing all the system firmware settings back to defaults'?

No I haven't, how can I do that?

dkwo, 2019-08-22:
Same exact issue with a thinkpad t460s running arch linux.
Today is the first time it happens for me, after many succesfull updates.