fwupd/fwupd #1295: 'get-updates' and 'update' have no output whatsoever

LinuxOnTheDesktop, 2019-08-20:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.2.10
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

```shell
$ fwupdmgr get-devices
LITEON CV3-CE256
  DeviceId:             e421b2fc248391f6fe3e55ddbb3c9043be068bd0
  Guid:                 58a07ffd-7bba-5190-b630-b638bdb5245c
  Guid:                 95edb012-8c8a-5329-8850-08ae374c55ca
  Guid:                 57281734-0dd5-5931-ab47-8823902931ef
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              TC87905
  Icon:                 drive-harddisk
  Created:              2019-08-20
```

Please answer the following questions:

- Operating system and version: Mint 19.2 Cinnamon. 5.0 kernel.
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No.
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-08-20:
Presumably you've already run `fwupdmgr refresh`, right?

The device in question doesn't have any updates available at LVFS according to that output.
```
  Flags:                internal|updatable|require-ac|registered|needs-reboot
```

There is no _supported_ flag in those flags.

LinuxOnTheDesktop, 2019-08-20:
> Presumably you've already run fwupdmgr refresh, right?

Yes.

> The device in question doesn't have any updates available at LVFS
OK, but blank output does not suggest that - at least to those not in the know. Rather, blank output suggests that the program is malfunctioning. Or so it seems to me.

superm1, 2019-08-20:
> OK, but blank output does not suggest that - at least to those not in the know. Rather, blank output suggests that the program is malfunctioning. Or so it seems to me.

For scripting purposes you don't want output on stdout in the scenario there is nothing to do.

What's the return code from `#fwupdmgr get-updates`?  I'd expect it's 0.

LinuxOnTheDesktop, 2019-08-20:
OK, but for user purposes one wants some output. Also, can't scripts use the return code and/or e.g. `grep`?

The exit-code is 0/true, with or without sudo.

LinuxOnTheDesktop, 2019-08-20:
I deem this behaviour a bug, at least without the top-level, in-program documentation explaining the behaviour.

closed 2019-08-21 by hughsie

LinuxOnTheDesktop, 2019-08-21:
Good stuff. Thank you.

LinuxOnTheDesktop, 2019-08-23:
Here's a thing: showing messaging when PC booted in legacy mode. Let me explain.

I am finding, on a computer on which I had not hitherto used fwupd (namely, a ThinkPad X230) that many operations are failing silently (with the latest snap) perhaps because the machine was booted in legacy mode. (I cannot make this machine boot in EUFI mode, because I lack a BIOS password.) I wonder whether the commit just made will cover this case.

superm1, 2019-08-23:
> $ fwupdmgr --version
client version:	1.0.9
daemon version:	1.2.10

It looks like you weren't using the 1.2.10 client.  The 1.2.10 client does notify you if you're running in legacy mode due to this commit:
https://github.com/fwupd/fwupd/commit/65a4894bba36137419739e41f23e4014232455d2#diff-2f5c39c775e4cac7ebbeabed4cefca6a
So if you remove the deb packaged version, or run `/snap/bin/fwupdmgr` you should see that.

If you want to see the behavior of this latest commit that fixed this reported issue, you can get the snap from the edge channel and try it.

LinuxOnTheDesktop, 2019-08-23:
Well, I've had a bad experience here.

1) I didn't realise that the apt version of `fwupd` was installed already (or whatever the relevant package names are called, the multiplicity of packages - `fwupd`, `fwupdmgr` being another problem). My distro must have installed it by default. And it seems that the apt version was running in preference to the snap version.

2) Uninstalling the apt version was hard, because I did not know quite what packages to install. (See 1.)

3) While I can now run the snap version, I can do so (i) only by using its full path, (ii) `fwupdmgr --version` now produces a 'no such file' error, (iii) `/snap/bin/fwupdmgr get-updates` produces no output.

superm1, 2019-08-23:
> I didn't realise that the apt version of fwupd was installed already (or whatever the relevant package names are called, the multiplicity of packages - fwupd, fwupdmgr being another problem). My [...]

Well according to your output you're running the snapped server, and apt client.
As mentioned in https://github.com/fwupd/fwupd/wiki/fwupd-snap it's recommended to remove the deb or rpm packaged version first to avoid confusion.

>While I can now run the snap version, I can do so (i) only by using its full path, (ii) fwupdmgr --version now produces a 'no such file' error, (iii) /snap/bin/fwupdmgr get-updates produces no output.

Log out and back in (or re-run bash) and it should adjust your PATH.  otherwise you can run `fwupd.fwupdmgr` to launch it.

> (iii) /snap/bin/fwupdmgr get-updates produces no output.
Let's make sure you have correct one installed now and in use.

`/snap/bin/fwupdmgr --version`
or
`fwupd.fwupdmgr --version`

LinuxOnTheDesktop, 2019-08-23:
Thanks. After logging out, I get no error messages.

Both the two version-checking commands that you posted produce this output:

```
client version: 1.2.10
checkout info: 1.2.10
compile-time dependency versions
       gusb: 0.3.0
       efivar: 37
daemon version: 1.2.10
```

I still get no output from `get-updates` or `update`, even though `get-devices` lists two devices as 'upgradeable' and one (the BIOS) as not upgradeable-due-to-legacy-mode. Perhaps here we have the problem that commit ed138eed1c984901a316a6021b107573b1d313ba fixes.

superm1, 2019-08-23:
You're running the latest stable release (1.2.10).  You won't get any output from `get-updates` until you are running the fix included from this issue (d2afb59b4259379b0c6c0172db8d4ced2fc2391d).  You can install the edge channel snap if you want to try it.  You should be able to switch with a command like this:

```
# snap switch fwupd --channel edge
```

commit 48968462 (2019-08-20): trivial: If no devices support updates, show messaging (Closes: #1295)

Normally the flag 'supported' is used to indicate that devices can
update from a remote such as LVFS.

Running `#fwupdmgr get-updates` in this situation will show a message
for each device that is already up to date as well as messages for
devices that have updates available.

If no devices contain the supported flag, `# fwupdmgr get-updates`
will show no output, which is confusing to some people.

Show "No updatable devices" instead for those situations.
commit d2afb59b (2019-08-26): trivial: If no devices support updates, show messaging (Closes: #1295)

Normally the flag 'supported' is used to indicate that devices can
update from a remote such as LVFS.

Running `#fwupdmgr get-updates` in this situation will show a message
for each device that is already up to date as well as messages for
devices that have updates available.

If no devices contain the supported flag, `# fwupdmgr get-updates`
will show no output, which is confusing to some people.

Show "No updatable devices" instead for those situations.