fwupd/fwupd #1298: Checksum in fu-hash.h changing between builds

lcp, 2019-08-21:
The checksum in fu-hash.h is used to detect the 3rd party plugins. However, after enabling LTO, the checksum changed between builds even without any source code change. It's because the extra LTO sections like .gnu.lto_.profile.e56fb3a1c2ab9039 were added to the object files in libfwupdprivate.a, and the static library becomes irreproducible.

I currently work around the issue by creating a never used libfwupdprivate_dyn.so from libfwupdprivate.a and hash it. It's not perfect but at least guarantee a stable hash.

hughsie, 2019-08-21:
Isn't a stable .a a prerequisite for a reproducible build? From memory the LTO build didn't actually work correctly, i.e. the plugin initialisation was failing.

lcp, 2019-08-21:
Yeah, it's funny that .a became irreproducible after enabling LTO, and I'm not sure how to deal with those LTO sections correctly.
Anyway, LTO is enabled by default for openSUSE Tumbleweed now, and the daemon seems loading the plugins without errors, so it's probably fine now.

hughsie, 2019-08-21:
Hmm, LTO does seem to work on Fedora now too. Does the fwupd binary change hash between builds, or just the .a static lib? If not, the `input : libfwupdprivate` could be replaced with something that creates a checksum of the source files rather than the build staticlib.

closed 2019-08-25 by hughsie

lcp, 2019-08-26:
Sorry for my late reply. The fwupd binary is stable. The static library was the only one changed between builds. I was thinking about stripping the LTO section from object files in libfwupdprivate.a, but hashing the source files also works anyway. Thanks for the fix.