fwupd/fwupd #1306: lenovo s1 yoga not recognised

igalic, 2019-08-24:
here's what `sudo -H lshw -json | jq .configuration` has to say:

```json
{
  "administrator_password": "disabled",
  "chassis": "notebook",
  "family": "ThinkPad S1 Yoga",
  "power-on_password": "disabled",
  "sku": "LENOVO_MT_20CD_BU_Think_FM_ThinkPad S1 Yoga",
  "uuid": "81A7D6EA-A053-CB11-A64D-98F361651DC8"
}
```

```shell
$ fwupdmgr --version
client version:	1.2.10
checkout info:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

```shell
$ fwupdmgr get-devices
SAMSUNG MZ7TE256HMHP-000L7
  DeviceId:             602b0a6cc821d155208724f0e22f8d111542b74c
  Guid:                 57fd661f-ff6a-5b36-b33b-6d53e83201d6
  Guid:                 2928f622-3825-5cbe-8880-8b7ba6bca8a6
  Guid:                 bd4c1a66-9ad5-57a3-9195-d3848f47aaaf
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              EXT09L6Q
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-08-24

```

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,001A,0001,0016,0017,0018,0019,001B
Boot0000* ubuntu	HD(1,GPT,15c6b8a7-08b1-4dac-b275-167a619fde1c,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Windows Boot Manager	HD(2,GPT,cddf5dd2-adaa-43ca-8de0-2aeaab809298,0x1f4800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...T................
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0017* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0018* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot0019* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001A* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001B* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
```

```shell
$ tree /boot
/boot
├── config-4.15.0-55-generic
├── config-4.15.0-58-generic
[... 318 lines cut ...]
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.15.0-55-generic
├── initrd.img-4.15.0-58-generic
├── lost+found [error opening dir]
├── System.map-4.15.0-55-generic
├── System.map-4.15.0-58-generic
├── vmlinuz-4.15.0-55-generic
└── vmlinuz-4.15.0-58-generic

6 directories, 328 files
```

Please answer the following questions:

- Operating system and version: elementary OS 5.0 Juno (based on Ubuntu 18.04 LTS)
- How did you install fwupd? — via snap
- Have you tried rebooting? — not yet… will do and report back
- Are you using an NVMe disk? yesno: I have an NVMe disk and SSD in an encrypted LVM
- Is secure boot enabled (only for the UEFI plugin)? — no

hughsie, 2019-08-25:
What does `fwupdmgr get-devices --show-all-devices` say?

igalic, 2019-08-25:
the exact same thing it says without `--show-all-devices`.

igalic, 2019-08-25:
update: secure boot is *not* actually enabled

hughsie, 2019-08-26:
Is UpdateCapsule support turned on in your BIOS? Does it work on Fedora 30? None of us actually test on elementary OS.

igalic, 2019-08-26:
is there anything that can be done about this?

i can download the cab manually, and then not do anything useful with it…

igalic, 2019-08-26:
> Is UpdateCapsule support turned on in your BIOS?

i honestly don't know:
[image]

I'm downloading a Fedora 30 ISO right now…

igalic, 2019-08-27:
i have now tried 6 different methods to put 2 different Fedora 30 ISOs on 3 different USB sticks, and none of them result in something that'll boot (on my machine??)

anyway, have i mentioned, that elementary OS is an Ubuntu LTS?

igalic, 2019-09-26:
i have provided as much info as i can

as mentioned, i was unable to boot a Fedora to try that, not sure what else i can do 🤷‍♀️

igalic, 2019-10-01:
I did a `snap refresh` which pulled in (1.3.2-14-g34c366aa) and i'm finally seeing … something!

```
❰meena❙~❱✔≻ fwupdmgr get-devices --show-all-devices -vv
(fwupdmgr:12819): FuMain-DEBUG: 09:03:50.506: SAMSUNG MZ7TE256HMHP-000L7
  DeviceId:             602b0a6cc821d155208724f0e22f8d111542b74c
  Guid:                 57fd661f-ff6a-5b36-b33b-6d53e83201d6
  Guid:                 2928f622-3825-5cbe-8880-8b7ba6bca8a6
  Guid:                 bd4c1a66-9ad5-57a3-9195-d3848f47aaaf
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  Version:              EXT09L6Q
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-10-01

(fwupdmgr:12819): FuMain-DEBUG: 09:03:50.506: System Firmware
  DeviceId:             123fd4143619569d8ddb6ea47d1d3911eb5ef07a
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|require-ac|registered|needs-reboot
  Vendor:               LENOVO
  Version:              GQET37WW (1.17 )
  VersionFormat:        plain
  Icon:                 computer
  Created:              2019-10-01
  UpdateError:          UEFI Capsule updates not available or enabled

❰meena❙~❱✔≻ 
```

now to figure out how to enable UEFI Update Capsule for my computer.

superm1, 2019-10-18:
Sounds good, so nothing wrong with fwupd here then.

closed 2019-10-18 by superm1

igalic, 2019-10-21:
yeah, i think i over-zealously accepted [the only search result for "lenovo s1 yoga"](https://fwupd.org/lvfs/search?value=lenovo+s1+yoga) as my laptop's firmware… and was very confused when fwupd refused to recognize and update it.

but the critical detail is that that firmware is for "S1 3rdGen", which my laptop is not.

So, yes, nothing wrong with fwupd, only with my hopes and expectations 😹️