fwupd/fwupd #1312: Adding flag for "needs-reboot" in metadata of cab files.

RJ-LeJaLTD, 2019-08-27:
Is there a way for us to include the “needs-reboot” flag in the metadata?
I have encountered the error below several times, especially using 1.2.10.

> UpdateError=failed to get device after update: failed to wait for detach replug: device <_numbers_> did not come back

but when I check it after reboot, the FW was sucessfully updated.

hughsie, 2019-08-27:
@RJ-LeJaLTD thanks for filing this -- can you follow the steps Mario outlined in https://github.com/fwupd/fwupd/issues/1312#issuecomment-525138931 please?

RJ-LeJaLTD, 2019-08-27:
However the `fwupdmgr update` generates a failed result and once reported to LVFS this causes the stable firmware to rollback to embargoed. 
Is there a way for us filter these error?

RJ-LeJaLTD, 2019-08-27:
> Can you reproduce this locally?
I have experienced it before, so i think i can. 
What kind of logs do i need to provide you with?

RJ-LeJaLTD, 2019-08-28:
Please attached files 

[dmesg.txt](https://github.com/fwupd/fwupd/files/3549038/dmesg.txt)

[image]

hughsie, 2019-08-28:
If this is a kernel version issue, perhaps we need a `<require version="4.13.0" compare="ge">kernel<require>` for metadata to use?

ChiWei-Chen, 2020-06-04:
Hi @superm1,
I'm working with @RJ-LeJaLTD. 
I can file a new issue to discuss if you think it's better.

It seems the update error narrowed down to T470/T470s/P50 Thunderbolt Controller firmware
> UpdateError= Error failed to get device after update: failed to wait for detach replug: device <DEV_ID> did not come back

In the Intel KBL platform and lower systems, the native mode can not power up the TBT controller without TBT device connected to the device. Hence generating a fail report regardless if the update is successful or not. it seems to be a design problem. 
We would like to know if we can add “needs-reboot” flag in the metadata to reboot for T470/T470s/P50 system after upgrading TBT controller firmware to bypass the issue. Otherwise, the firmware would be pulled back to testing once the fault-failure report received. The success rate only 5.9% / 3.4% / 14.3%, I cannot say it was a usual phenomenon.

Thanks.

superm1, 2020-06-04:
Just adding needs reboot flag will not bypass the issue. The plugin checks that update was successful by probing device. If device isn't present it's impossible to know whether it was successful.

From your description this might actually be caused by force power plugin in kernel and fwupd not working as intended.  Force power is a hardware feature of thunderbolt controller though that is controlled by BIOS. Since you mentioned native mode I would also wonder if BIOS SMI related to native enumeration doesn't work with force power.

I really don't think this is root caused yet or isolated to BIOS, fwupd, kernel, thunderbolt firmware, or hardware.

ChiWei-Chen, 2020-06-04:
Thanks, @superm1.
Your feedback is very appreciated. I think we need to work with BIOS team to see if they can control the force power attribute.

superm1, 2020-06-04:
Yeah, I think that's the best starting point.  And if everything looks right from their side while in this state then check the controller itself and see if it got wedged some how.  

>native mode can not power up the TBT controller without TBT device connected to the device. 

If it truly turns into a situation not solvable by BIOS, then the options from fwupd side are:

1. Modify all supported versions of fwupd to not allow force power device anymore.  This means that users will no longer be able to update device unless something is plugged in.  You'll need to confirm this issue NEVER happens when something is plugged in for this approach.
2. Add support for fwupd to use `needs-reboot` in the Thunderbolt plugin.  Add quirks for all problematic devices that need a reboot to un-wedge themselves.

Of course having to make fwupd code changes will also take a while to roll out the updates to distributions.  I don't really like either of these, but if it does turn out to not be fixable in firmware there probably aren't any other options.

superm1, 2020-06-04:
@hughsie any thoughts on this?

hughsie, 2020-06-04:
>  This means that users will no longer be able to update device unless something is plugged in.

This is not awesome from a UX point of view.

>  support for fwupd to use needs-reboot in the Thunderbolt plugin

This is something we could do.

superm1, 2020-06-04:
I am starting to suspect this might be an intersection of a device that supports native enumeration but does not support RTD3, so it should be a fixed number of models.  I can't speak for Lenovo, but for Dell I'm not aware of any products with CML or newer that don't support RTD3.

> This is not awesome from a UX point of view.

However it is something that we have done for master (the thunderbolt plugin overhaul dropped force power support).

superm1, 2020-06-29:
That is the exact flow that fwupd uses.  The problem is that the device can't probe after it finished updating.

superm1, 2020-06-30:
@ChiWei-Chen FYI to you, you should review the proposal https://github.com/fwupd/fwupd/pull/2235.  This is removing the force power support in 1.4.x branch.  It will have the following side effects:

- Updates are no longer offered unless a device is plugged in
- This issue should no longer be hit - force power will never be used by fwupd anymore

superm1, 2020-06-30:
That has been pulled into 1.4.x.  If you want to stop failure reports on earlier versions you can set the metadata requirement to 1.4.5 or later.

commit c8208b12 (2019-08-28): thunderbolt: Add support for kernel safety checks (Fixes: #1312)

There are commits to the Thunderbolt kernel driver that make sure
that the upgrade process goes smoothly.  If these commits aren't
present then it will look like a fwupd problem, when it's actually
a kernel problem.

When this issue was reported it appeared that commit
https://github.com/torvalds/linux/commit/e4be8c9b6a512e274cb6bbac4ac869d73880a8b3
was missing from the locally tested kernel, but it's impossible
to determine that from userspace.

Prevent running the thunderbolt plugin on older kernels than that
set in `$sysconfd [... cut]
commit 5ba1d240 (2019-08-28): thunderbolt: Add support for kernel safety checks (Fixes: #1312)

There are commits to the Thunderbolt kernel driver that make sure
that the upgrade process goes smoothly.  If these commits aren't
present then it will look like a fwupd problem, when it's actually
a kernel problem.

When this issue was reported it appeared that commit
https://github.com/torvalds/linux/commit/e4be8c9b6a512e274cb6bbac4ac869d73880a8b3
was missing from the locally tested kernel, but it's impossible
to determine that from userspace.

Prevent running the thunderbolt plugin on older kernels than that
set in `$sysconfd [... cut]
commit 6865dd3b (2019-08-28): thunderbolt: Add support for kernel safety checks (Fixes: #1312)

There are commits to the Thunderbolt kernel driver that make sure
that the upgrade process goes smoothly.  If these commits aren't
present then it will look like a fwupd problem, when it's actually
a kernel problem.

When this issue was reported it appeared that commit
https://github.com/torvalds/linux/commit/e4be8c9b6a512e274cb6bbac4ac869d73880a8b3
was missing from the locally tested kernel, but it's impossible
to determine that from userspace.

Prevent running the thunderbolt plugin on older kernels than that
set in `$sysconfd [... cut]
closed 2020-06-30 by superm1