fwupd/fwupd #1314: Significant ambiguity, to do with 'administrator', in the help text

LinuxOnTheDesktop, 2019-08-27:
```shell
$ fwupdmgr --version
client version:	1.2.10
checkout info:	1.2.10
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.10
```

- Operating system and version: Mint 19.2 Cinnamon
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): snap

**Issue**

`fwupdmgr --help` [or just `fwupdmgr`] produces text that ends thusly

> This tool allows an administrator to query and control the fwupd daemon, allowing them to perform actions such as installing or downgrading firmware.

The text confuses. For, it gives one to wonder whether the following. Does the tool require `sudo` (and if so, for just which actions?)? Or is it instead that 'administrator' has another sense here, one that has nothing, directly, to do with root access?

hughsie, 2019-08-27:
`s/an administrator/a user`?

LinuxOnTheDesktop, 2019-08-27:
Sorry; I don't understand. Is the question what the message would better say? If so, perhaps something like the following would do.

> This tool (fwupdmgr) allows one to query and control the fwupd daemon and thus to perform actions such as installing or downgrading firmware. Some actions require that one have (or authenticate in order to get) 'root' access to the system.

I am unsure about the second sentence. For I do not really know the relevant terminology (though I know enough, I think, to realise that 'administrator', at least in the way the existing text uses that word, is confusing).

closed 2019-10-03 by stale[bot]