fwupd/fwupd #1320: Need a new tarball release

hughsie, 2019-08-28:
I think we've had more than enough changes for a new release. My proposal would be that we branch `1_2_X` just after `1.2.10` and make master into 1.3.1. We'd then backport any of the sane stable stuff into `1_2_X` and then carry on having fun in master until we feel master is stable enough.

hughsie, 2019-08-28:
I've backported a ton of stuff from master onto `1_2_X` now -- can you have a look and see if you think anything is missing, or any new features that you really need in the tarball release that's headed to stable distros?

hughsie, 2019-08-28:
> 4fa965a
> ba3d913

Both backported, thanks.

superm1, 2019-09-19:
I think it's a good idea to do a 1.2.11 now

superm1, 2019-09-23:
I backported a few other things from master that were strictly fixes. a3928370545da930123ba7450ba8dd2c153f6f35
149ef5ec03b1c0c171578d2c30c34bb1945d75af
8f7e1bdf0e60942eba017971a971c3d64e4323a5
f4558a67da607b8efc7d16acb3d04d4f6cfd7f2f

superm1, 2019-09-28:
1.2.11 tagged today

closed 2019-09-28 by superm1