fwupd/fwupd #1324: Ubuntu/Debian build requires libtss2-udev

Sean-StarLabs, 2019-08-29:
1.3.1 requires libtss2-udev to build (for Debian and Ubuntu at least).

Can it please be added to the contrib dependencies?

superm1, 2019-08-29:
It should only need `libtss2-dev`, which is in the contrib dependencies.

Sean-StarLabs, 2019-08-29:
Sorry, long day. I meant libtss2-dev. It doesn't show up in the build-deps, i.e. If I run
```
OS=debian-x86_64 contrib/ci/generate_docker.py
docker run --privileged -t -v `pwd`:/build fwupd-debian-x86_64
```
The control file has the below:
```
Build-Depends:
        bash-completion,
        bubblewrap,
        debhelper (>= 11),
        dh-strip-nondeterminism,
        fontconfig,
        fonts-noto,
        gcab,
        gettext (>= 0.19.8.1),
        gir1.2-pango-1.0,
        gnu-efi [amd64 arm64 armhf i386],
        gnutls-bin,
        gnutls-dev,
        gobject-introspection,
        gtk-doc-tools,
        help2man,
        libarchive-dev,
        libcairo-dev,
        libcairo-gobject2,
        libefiboot-dev [amd64 arm64 armhf i386],
        libefivar-dev [amd64 arm64 armhf i386],
        libelf-dev,
        libfreetype6-dev,
        libgcab-dev,
        libgirepository1.0-dev,
        libglib2.0-dev (>= 2.45.8),
        libgpgme11-dev,
        libgudev-1.0-dev,
        libgusb-dev (>= 0.2.9),
        libjson-glib-dev (>= 1.1.1),
        libmm-glib-dev,
        libpolkit-gobject-1-dev,
        libqmi-glib-dev,
        libsmbios-dev [i386 amd64],
        libsoup2.4-dev,
        libsqlite3-dev,
        libtool-bin,
        libxmlb-dev (>= 0.1.5),
        locales,
        meson,
        pkg-config,
        policykit-1 (>> 0.105-14),
        python3-gi-cairo,
        python3-pil,
        python3-requests,
        shared-mime-info,
        systemd (>= 231),
        udev,
        umockdev,
        valac,
        valgrind [!ia64 !riscv64 !x32 !mips !sparc64 !sh4 !ppc64 !powerpcspe !hppa !alpha !mips64el !armhf !armel !mipsel !m68k],
```

commit 37a3ecf0 (2019-08-29): trivial: add libtss2-dev into debian/control (Closes: #1324)
closed 2019-08-29 by superm1